espressif/esp-idf #14984: s_log_level_get_and_unlock cause system panic

zzzh, 2024-12-06:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v4.4.2

### Espressif SoC revision.

ESP32-C3 (revision 4)

### Operating System used.

Linux

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

None

### Development Kit.

ESP32C3-CORE

### Power Supply used.

USB

### What is the expected behavior?

No panic

### What is the actual behavior?

Have panic

### Steps to reproduce.

I was using http server to receive a file from a http client on an Android phone, sometimes it works well, sometimes it would panic on s_log_level_get_and_unlock

### Debug Logs.

```plain
Guru Meditation Error: Core  0 panic'ed (Load access fault). Exception was unhandled.

Stack dump detected
Core  0 register dump:
MEPC    : 0x40058ce6  RA      : 0x420c396e  SP      : 0x3fcd01e0  GP      : 0x3fc92200  
0x420c396e: get_uncached_log_level at /home/zhu/esp/esp-idf/components/log/log.c:278
 (inlined by) s_log_level_get_and_unlock at /home/zhu/esp/esp-idf/components/log/log.c:154

TP      : 0x3fc99f24  T0      : 0x4005890e  T1      : 0x4038bcda  T2      : 0xffffffff  
0x4038bcda: vTaskGenericNotifyGiveFromISR at /home/zhu/esp/esp-idf/components/freertos/tasks.c:5865

S0/FP   : 0x3fc96124  S1      : 0x0000001c  A0      : 0x3c0f2610  A1      : 0x5006005d  
A2      : 0x00000065  A3      : 0x00000041  A4      : 0x00000001  A5      : 0x3fc9b000  
A6      : 0x00280020  A7      : 0x3c0d0020  S2      : 0x50060058  S3      : 0x3c0f2610  
S4      : 0x3fc9b75c  S5      : 0x00000000  S6      : 0x3fc9ffc4  S7      : 0x0000ffff  
S8      : 0x3fcd042c  S9      : 0x3fcd02ec  S10     : 0x3c0db000  S11     : 0x3c0d0020  
T3      : 0x00000000  T4      : 0x2ac566f5  T5      : 0x6b206574  T6      : 0x79622d32  
MSTATUS : 0x00001881  MTVEC   : 0x40380001  MCAUSE  : 0x00000005  MTVAL   : 0x5006005d  
0x40380001: _vector_table at ??:?

MHARTID : 0x00000000  

Backtrace:

0x40058ce6 in ?? ()
#0  0x40058ce6 in ?? ()
#1  0x420c396e in get_uncached_log_level (level=<synthetic pointer>, tag=0x3c0f2610 "") at /home/zhu/esp/esp-idf/components/log/log.c:278
#2  s_log_level_get_and_unlock (tag=0x3c0f2610 "") at /home/zhu/esp/esp-idf/components/log/log.c:154
#3  0x00000000 in ?? ()
Backtrace stopped: frame did not save the PC
ELF file SHA256: e139a94d1698f590
```

### More Information.

I checked stack overflow of the tasks to make sure it's not the issue source, all tasks have more than 1000 bytes in stack, looks good to me.
I did try latest version of esp-idf, something is not compatible, so I gived up trying this on it.(but have to select "I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there. " or the "Submit new issue" button would be invalid)

KonstantinKondrashov, 2024-12-06:
Hi @zzzh!
Did it come from ISR? I think you can use  ESP_EARLY_LOG macro for logging from ISR. It calls` esp_rom_function()`, which consumes about 128 bytes of stack. The regular ESP_LOG macro which calls the `vprintf()` function consumes about 1168 bytes of stack. (I measured it recently for master branch).

> 0x420c396e: get_uncached_log_level at /home/zhu/esp/esp-idf/components/log/log.c:278

This point to [strcmp(tag, it->tag)](https://github.com/espressif/esp-idf/blob/v4.4.2/components/log/log.c#L278), the error is "Load access fault", bacause "it" is not a valid address, so dereferencing to take "tag" leads to the fail. Probably because it was corrupted by another task or so.

zzzh, 2024-12-06:
Hi, @KonstantinKondrashov!

Thanks for your quick respond.

It's in put_handler of the http server, should be an ISR, I tried using ESP_EARLY_LOG macro instead of ESP_LOG, it do decrease the stack usage size.

Though, the panic still happens as before..

Maybe I should use printf instead of ESP's log, I'll try it later.

KonstantinKondrashov, 2024-12-06:
If the panic is still happening it means the issue somewhere else. ESP_EARLY_LOG is just a wrapper over the ROM printf function. It is simplified version of the print from lib.

> Maybe I should use printf instead of ESP's log,

Does not matter for for your case. Using printf will be the same as using vprintf function which was used by ESP_LOG.

zzzh, 2024-12-06:
I thought it's because a change in log level in code (like use ESP_LOGI then ESP_LOGW) that lead to this bug, isn't it correct?

KonstantinKondrashov, 2024-12-06:
No, the log level change should not affect any issue.
I suspect that you have a memory corruption issue.
Consider using these methods to localize the issue.
Enable CONFIG_HEAP_POISONING_COMPREHENSIVE and CONFIG_FREERTOS_CHECK_STACKOVERFLOW_CANARY. More about in in the doc https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/system/heap_debug.html#heap-corruption-detection

zzzh, 2024-12-06:
Ok, I'll check it out.
Thanks for the help, it's really valuable.

zzzh, 2024-12-08:
> No, the log level change should not affect any issue. I suspect that you have a memory corruption issue. Consider using these methods to localize the issue. Enable [...]

Hi, @KonstantinKondrashov,
In my case, CONFIG_FREERTOS_CHECK_STACKOVERFLOW_CANARY was set already, so I enabled CONFIG_HEAP_POISONING_COMPREHENSIVE and look forward to see where it would say going wrong...
To my surprise, it never goes wrong again, the panic disappeared.
I don't know why, just be happy to pass it through and feed the result to you.
Thank you once more.

closed 2024-12-08 by zzzh