espressif/esp-idf #14985: ble connection parameter's .latency is sometimes ignored, even though there is no read, write, notify, indicate event.

ChulseungYoo, 2024-12-06:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

ESP-IDF Release v5.3.1

### Espressif SoC revision.

ESP32-H2

### Operating System used.

macOS

### How did you build your project?

VS Code IDE

### If you are using Windows, please specify command line type.

None

### Development Kit.

Custom Board Using ESP32-H2-MINI-1-N4

### Power Supply used.

Battery

### What is the expected behavior?

I wrote a little additional code based on the example of
Examples/bluetooth/nimble/power_save

`bleprph_gap_event(struct ble_gap_event *event, void *arg)
{
    struct ble_gap_conn_desc desc;
    int rc;

    switch (event->type) {
    case BLE_GAP_EVENT_CONNECT:
        /* A new connection was established or a connection attempt failed. */
        MODLOG_DFLT(INFO, "connection %s; status=%d ",
                    event->connect.status == 0 ? "established" : "failed",
                    event->connect.status);
        if (event->connect.status == 0) {
            rc = ble_gap_conn_find(event->connect.conn_handle, &desc);
            assert(rc == 0);
            bleprph_print_conn_desc(&desc);
                        /* Try to update connection parameters */
            struct ble_gap_upd_params params = {.itvl_min = 40,
                                                .itvl_max = 80,
                                                .latency = 10,
                                                .supervision_timeout = 1600,
                                                .max_ce_len = 300,
                                                .min_ce_len = 20};
            rc = ble_gap_update_params(event->connect.conn_handle, &params);
            if (rc != 0) {
                ESP_LOGE(
                    "ERR",
                    "failed to update connection parameters, error code: %d",
                    rc);
                return rc;
            }

        }
        MODLOG_DFLT(INFO, "\n");

        if (event->connect.status != 0) {
            /* Connection failed; resume advertising. */
#if CONFIG_EXAMPLE_EXTENDED_ADV
            ext_bleprph_advertise();
#else
            bleprph_advertise();
#endif
        }

`

I tried to achieve low power by adjusting the connection parameter even after connection established 
The connection interval of our product is about 100ms, when idle, there is no data transmission, and there is only data transmission in a certain mode, so I wanted to give a slave latency of 10 and operate at intervals of about 1 second when idle.

### What is the actual behavior?

I checked that it basically works, but as time goes by, it ignores the latency of the connection parameter, and the connection event occurs at the set connection interval.

This could be confirmed during the power consumption check of Nordic's PPK2.

The connection on the mobile device was carried out in the iOS lightblue app,

I didn't read or write the value separately after connecting.[image]
I set BT_NIMBLE_LOG_LEVEL to Debug logs and LOG_DEFAULT_LEVEL to Debug in SDK Config, but there was no additional log.

[image]
In addition, this image is a logging of power consumption for 4 minutes after connection. Usually, if some connection events are ignored by latency, it consumes an average current of about 200uA. However, the current sample consumes an average of 500uA.
비슷한 저희 기기 중 nrf52811 제품에서는 이러한 상황은 발생하지 않고 있습니다.

### Steps to reproduce.

to reproduce it. just use powersave example

`                        /* Try to update connection parameters */
            struct ble_gap_upd_params params = {.itvl_min = 40,
                                                .itvl_max = 80,
                                                .latency = 10,
                                                .supervision_timeout = 1600,
                                                .max_ce_len = 300,
                                                .min_ce_len = 20};
            rc = ble_gap_update_params(event->connect.conn_handle, &params);
            if (rc != 0) {
                ESP_LOGE(
                    "ERR",
                    "failed to update connection parameters, error code: %d",
                    rc);
                return rc;
            }
`
write this code in 
bleprph_gap_event function's BLE_GAP_EVENT_CONNECT case.

### Debug Logs.

```plain
I (11836) NimBLE: connection established; status=0 
I (11837) NimBLE: handle=0 our_ota_addr_type=0 our_ota_addr=
I (11889) NimBLE: 74:4d:bd:63:fe:88
I (11890) NimBLE:  our_id_addr_type=0 our_id_addr=
I (11890) NimBLE: 74:4d:bd:63:fe:88
I (11892) NimBLE:  peer_ota_addr_type=1 peer_ota_addr=
I (11898) NimBLE: 77:de:81:23:20:26
I (11902) NimBLE:  peer_id_addr_type=1 peer_id_addr=
I (11908) NimBLE: 77:de:81:23:20:26
I (11913) NimBLE:  conn_itvl=24 conn_latency=0 supervision_timeout=72 encrypted=0 authenticated=0 bonded=0

I (11923) NimBLE: GAP procedure initiated: 
I (11927) NimBLE: connection parameter update; conn_handle=0 itvl_min=40 itvl_max=80 latency=10 supervision_timeout=1600 min_ce_len=20 max_ce_len=300
I (11943) NimBLE: 

I (11946) NimBLE: 

I (12297) NimBLE: mtu update event; conn_handle=0 cid=4 mtu=256

I (12603) NimBLE: connection updated; status=0 
I (12605) NimBLE: handle=0 our_ota_addr_type=0 our_ota_addr=
I (12655) NimBLE: 74:4d:bd:63:fe:88
I (12655) NimBLE:  our_id_addr_type=0 our_id_addr=
I (12655) NimBLE: 74:4d:bd:63:fe:88
I (12658) NimBLE:  peer_ota_addr_type=1 peer_ota_addr=
I (12663) NimBLE: 77:de:81:23:20:26
I (12668) NimBLE:  peer_id_addr_type=1 peer_id_addr=
I (12673) NimBLE: 77:de:81:23:20:26
I (12677) NimBLE:  conn_itvl=72 conn_latency=10 supervision_timeout=1600 encrypted=0 authenticated=0 bonded=0

I (12688) NimBLE: 

I (13603) NimBLE: subscribe event; conn_handle=0 attr_handle=8 reason=1 prevn=0 curn=0 previ=0 curi=1

I (18764) NimBLE: Descriptor read; conn_handle=0 attr_handle=31
```

zhaoweiliang2021, 2025-04-01:
@ChulseungYoo  
Thank you for raising this issue. Please follow these steps:
1. Unzip the provided libble_app.zip file. Will get a libble_app.a file.
2. Replace the existing libble_app.a file in the specified directory(esp-idf/components/bt/controller/lib_esp32h2/esp32h2-bt-lib).
3. Rebuild your project and perform the test again.
If the issue persists, please provide further details.
We look forward to your updates.
Thanks.
[libble_app.zip](https://github.com/user-attachments/files/19549737/libble_app.zip)

ChulseungYoo, 2025-04-02:
@zhaoweiliang2021 
Thanks for your reply.

I tried to replace and rebuilt my project.
but in linking process. it has errors 
```
ls/riscv32-esp-elf/esp-14.2.0_20241119/riscv32-esp-elf/bin/../lib/gcc/riscv32-esp-elf/14.2.0/../../../../riscv32-esp-elf/bin/ld: esp-idf/bt/libbt.a(bt.c.obj): in function `sleep_modem_ble_mac_retention_init':
/Users/chulseungyoo/esp/v5.4/esp-idf/components/bt/controller/esp32h2/bt.c:575:(.text.sleep_modem_ble_mac_retention_init+0xc): undefined reference to `r_esp_ble_mac_retention_link_get'
/Users/chulseungyoo/.espressif/tools/riscv32-esp-elf/esp-14.2.0_20241119/riscv32-esp-elf/bin/../lib/gcc/riscv32-esp-elf/14.2.0/../../../../riscv32-esp-elf/bin/ld: esp-idf/bt/libbt.a(bt.c.obj): in function `controller_sleep_init':
/Users/chulseungyoo/esp/v5.4/esp-idf/components/bt/controller/esp32h2/bt.c:629:(.text.controller_sleep_init+0x6a): undefined reference to `r_esp_ble_stop_wakeup_timing'
/Users/chulseungyoo/.espressif/tools/riscv32-esp-elf/esp-14.2.0_20241119/riscv32-esp-elf/bin/../lib/gcc/riscv32-esp-elf/14.2.0/../../../../riscv32-esp-elf/bin/ld: /Users/chulseungyoo/esp/v5.4/esp-idf/components/bt/controller/esp32h2/bt.c:629:(.text.controller_sleep_init+0x74): undefined reference to `r_esp_ble_stop_wakeup_timing'
/Users/chulseungyoo/.espressif/tools/riscv32-esp-elf/esp-14.2.0_20241119/riscv32-esp-elf/bin/../lib/gcc/riscv32-esp-elf/14.2.0/../../../../riscv32-esp-elf/bin/ld: /Users/chulseungyoo/esp/v5.4/esp-idf/components/bt/controller/esp32h2/bt.c:645:(.text.controller_sleep_init+0xee): undefined reference to `r_esp_ble_stop_wakeup_timing'
/Users/chulseungyoo/.espressif/tools/riscv32-esp-elf/esp-14.2.0_20241119/riscv32-esp-elf/bin/../lib/gcc/riscv32-esp-elf/14.2.0/../../../../riscv32-esp-elf/bin/ld: esp-idf/bt/libbt.a(bt.c.obj): in function `controller_sleep_deinit':
/Users/chulseungyoo/esp/v5.4/esp-idf/components/bt/controller/esp32h2/bt.c:679:(.text.controller_sleep_deinit+0x52): undefined reference to `r_esp_ble_stop_wakeup_timing'
/Users/chulseungyoo/.espressif/tools/riscv32-esp-elf/esp-14.2.0_20241119/riscv32-esp-elf/bin/../lib/gcc/riscv32-esp-elf/14.2.0/../../../../riscv32-esp-elf/bin/ld: /Users/chulseungyoo/esp/v5.4/esp-idf/components/bt/controller/esp32h2/bt.c:679:(.text.controller_sleep_deinit+0x56): undefined reference to `r_esp_ble_stop_wakeup_timing'
collect2: error: ld returned 1 exit status
ninja: build stopped: subcommand failed.

```

zhaoweiliang2021, 2025-04-02:
@ChulseungYoo 
Which commit of idf used? is it commit c8fc5f643b7a7b0d3b182d3df610844e3dc9bd74 (HEAD, tag: v5.3.1)?

ChulseungYoo, 2025-04-02:
I use 5.4 release version.

zhaoweiliang2021, 2025-04-02:
@ChulseungYoo  The provided libble_app.a file only work with https://github.com/espressif/esp-idf/commit/c8fc5f643b7a7b0d3b182d3df610844e3dc9bd74 (HEAD, tag: v5.3.1). Can you switch back to that commit to verify? If it is not convenient to switch back, please provide the commit and sdkconfig files you are using.  The new libble_app.a file will be provied later.

ChulseungYoo, 2025-04-02:
Ok. I'll change the IDF version to 5.3.1 and try.

zhaoweiliang2021, 2025-04-02:
@ChulseungYoo  Great! looking forward to your reply.

ChulseungYoo, 2025-04-02:
@zhaoweiliang2021 
I Tried to rebuild it and the result is

[image]

The issue does not seem to be resolved yet.

This bug appears to occur somewhat randomly. Rather than being completely random, it tends to happen in clusters.

Looking at it more closely, it appears as follows:
The small increase in power consumption is due to the UI LED, so it can be ignored.[image]

This is  connection parameter I use.

        struct ble_gap_upd_params conn_params = {
            .itvl_min = 40,
            .itvl_max = 80,
            .latency = 5,
            .supervision_timeout = 1000,
        };
        
        esp_err_t err = ble_gap_update_params(conn_handle, &conn_params);
        if (err != ESP_OK) {
            ESP_LOGE(TAG, "ble_gap_update_params err: %s, %d", esp_err_to_name(err), err);
        }

zhaoweiliang2021, 2025-04-02:
@ChulseungYoo  Please tell me the model of your test phone. I want to further confirm what data the phone sent when the issue occurred. Thanks.

ChulseungYoo, 2025-04-02:
@zhaoweiliang2021 
wait a moment please.
previous version I Tried was built with 5.4
the setting was set to 5.3.1 but at the kit's console  version command returnd idf-5.4-dirty
I tried to set idf to 5.3.1 and this time. it is not happening for about 5 minutes.

[image]

[image]
In the zoomed-in image, it sometimes sends connection event messages consecutively, but this does not seem to happen continuously.

ChulseungYoo, 2025-04-02:
@zhaoweiliang2021 
I think this solution works.
Thanks for your help.

Now I am using esp-idf 5.4.
should I go back to use 5.3.1? 
or libble_app.a file for 5.4 version will be available soon?

zhaoweiliang2021, 2025-04-02:
@ChulseungYoo  This fix will be merged  as soon as possible after it has been fully tested internally. It may take some time to merge. You can provide me the 5.4 commit you are using, and I will make a lib based on your commit so that you can continue to develop. The official fix needs to wait until the next 5.4 version is released.

ChulseungYoo, 2025-04-02:
> [@ChulseungYoo](https://github.com/ChulseungYoo) This fix will be merged as soon as possible after it has been fully tested internally. It may take some time to merge. You can provide me the 5.4 [...]

I now use the 5.4 version
https://github.com/espressif/esp-idf/releases/tag/v5.4
the commit is https://github.com/espressif/esp-idf/commit/67c1de1eebe095d554d281952fde63c16ee2dca0

Thanks!

zhaoweiliang2021, 2025-04-02:
@ChulseungYoo  The lib file corresponding to v5.4 commit https://github.com/espressif/esp-idf/commit/67c1de1eebe095d554d281952fde63c16ee2dca0
Please follow these steps:

1. Unzip the provided libble_app.zip file. Will get a libble_app.a file.
2. Replace the existing libble_app.a file in the specified directory(esp-idf/components/bt/controller/lib_esp32h2/esp32h2-bt-lib).
3. Rebuild your project and perform the test again.
Thanks.
[libble_app.zip](https://github.com/user-attachments/files/19564987/libble_app.zip)

ChulseungYoo, 2025-04-10:
@zhaoweiliang2021 
I’ve tested the libble_app.zip, and the issue with ignoring slave latency has been resolved.

However, I’m still experiencing problems with BLE notifications — they are not being delivered reliably. There are noticeable delays and occasional data loss.

In my device, sensor data is sent to the central (iPhone) at 10 FPS, with each packet being 9 bytes (about 90 bytes per second). While that may be relatively high for a BLE device, the current performance isn’t stable enough.

That said, the original issue has been fixed, so this ticket can be closed.

I hope you can test this further and find an optimal solution for the notification delivery problem.

closed 2025-04-10 by ChulseungYoo