espressif/esp-idf #14986: gatt_server visible but Android cannot connect at all

letisoft, 2024-12-06:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.3.1

### Espressif SoC revision.

ESP32

### Operating System used.

Windows

### How did you build your project?

VS Code IDE

### If you are using Windows, please specify command line type.

None

### Development Kit.

ESP32 Wrover 

### Power Supply used.

USB

### What is the expected behavior?

gatt_server should be able to read charactersitics

### What is the actual behavior?

gatt_server example:
Android devices can see the device but cannot connect

### Steps to reproduce.

1. build gatt_server
2. run any app (I tried, my app, nRF Connect, BLE Scanner) on android
3. device is visible but cannot connect so cannot read char value

### Debug Logs.

```plain
(962) GATTS_DEMO: SERVICE_START_EVT, status 0, service_handle 40
I (962) GATTS_DEMO: ADD_CHAR_EVT, status 0,  attr_handle 42, service_handle 40
I (972) GATTS_DEMO: the gatts demo char length = 3
I (982) GATTS_DEMO: prf_char[0] =11
I (982) GATTS_DEMO: prf_char[1] =22
I (982) GATTS_DEMO: prf_char[2] =33
I (992) GATTS_DEMO: CREATE_SERVICE_EVT, status 0,  service_handle 44
I (1002) GATTS_DEMO: ADD_DESCR_EVT, status 0, attr_handle 43, service_handle 40
I (1002) GATTS_DEMO: SERVICE_START_EVT, status 0, service_handle 44
I (1012) GATTS_DEMO: ADD_CHAR_EVT, status 0,  attr_handle 46, service_handle 44
I (1022) GATTS_DEMO: ADD_DESCR_EVT, status 0, attr_handle 47, service_handle 44
```

esp-zhp, 2024-12-09:
Thank you for providing the log. Based on the information, the connection between the Android device and the GATT server example was not successfully established. Without a successful connection, it is not possible to proceed with further read or write operations.

Could you please let us know:
1. The type of the Android phone you are using?  
2. Have you tested with other **Android ** phones to see if the issue persists?  
3. What is the specific behavior on the phone? For example, does it show any error messages or simply fail to connect without any response?  
4. Do you have a sniffer device to capture the over-the-air communication for further analysis?  

This information will help us identify the root cause of the issue and provide a more targeted solution.

letisoft, 2024-12-09:
Hello,

Thank you for your answer!
I am afraid this was my fault:
1. I bought this (see attached) as "ESP-32 dev board" so this is what I selected in arduino - "ESP32 Dev Module"
2. It flashes no issues, it even runs :)
3. I boght a proper S3 dev board and was able to debug, it all worked, so I came back to the "ESP32 Dev Module" and found it works if set to "ESP32-WROOM-DA Module"?!?
[image]

Why I really have no idea!
I hope there can be better documentation that comes with these chip boards but I am not sure if you have to help

Best regards:
V

esp-zhp, 2024-12-09:
So, can you connect to ESP32 now?

letisoft, 2024-12-09:
Yes everything works as expect. It was me selecting the wrong target in Arduino IDE.

Though it is strange that is half working

esp-zhp, 2024-12-10:
If your issue has been resolved, you may close it for now.thanks.

closed 2024-12-16 by Alvin1Zhang