espressif/esp-idf #14992: [esp32-p4] cannot move location counter backwards

chegewara, 2024-12-06:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

9106c43accd9f5e75379f62f12597677213f5023

### Operating System used.

Linux

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

None

### What is the expected behavior?

Hi team,
i updated esp-idf master today and encountered few problems. Here is one i dont know how to deal with:
```
.espressif/tools/riscv32-esp-elf/esp-14.2.0_20240906/riscv32-esp-elf/bin/../lib/gcc/riscv32-esp-elf/14.2.0/../../../../riscv32-esp-elf/bin/ld:sections.ld:743 cannot move location counter backwards (from 480bc221 to 480b9468)
collect2: error: ld returned 1 exit status
ninja: build stopped: subcommand failed.
ninja failed with exit code 1
```

Thanks

### What is the actual behavior?

Cant build with error

### Steps to reproduce.

Here are options i changed in menuconfig
```
CONFIG_IDF_TARGET="esp32p4"
CONFIG_ESPTOOLPY_FLASHSIZE_16MB=y
CONFIG_PARTITION_TABLE_SINGLE_APP_LARGE=y
CONFIG_SPIRAM=y
CONFIG_SPIRAM_SPEED_200M=y
CONFIG_SPIRAM_XIP_FROM_PSRAM=y
CONFIG_SPIRAM_ALLOW_BSS_SEG_EXTERNAL_MEMORY=y
CONFIG_SPIRAM_ALLOW_NOINIT_SEG_EXTERNAL_MEMORY=y
CONFIG_ESP_HOST_WIFI_ENABLED=y
CONFIG_FATFS_LFN_HEAP=y
CONFIG_FREERTOS_HZ=1000
CONFIG_IDF_EXPERIMENTAL_FEATURES=y
```

chegewara, 2024-12-06:
I tracked down it to this command
```
    wifi_init_config_t cfg = WIFI_INIT_CONFIG_DEFAULT();
    ESP_RETURN_ON_ERROR(esp_wifi_init(&cfg), TAG, "failed to init wifi");
```

bryghtlabs-richard, 2024-12-09:
Seems like your program is too large for the available program size.

chegewara, 2025-01-01:
Since espressif have no interest to test and solve the issue i am closing it.
I found which sdkconfig option is responsible for that error message, so i dont need more assistance with it.

Thanks

closed 2025-01-01 by chegewara

nicklasb, 2025-02-19:
@chegewara 
> Since espressif have no interest to test and solve the issue i am closing it. I found which sdkconfig option is responsible for that error message, so i dont need more assistance with it.
> 
> Thanks

What was the option?

chegewara, 2025-02-19:
This is the problem
```
CONFIG_SPIRAM_XIP_FROM_PSRAM=y
```

nicklasb, 2025-02-19:
That was it! Thank you!