espressif/esp-idf #14993: Install v5.3.2 error

Temik007, 2024-12-06:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.3.2

### Operating System used.

Windows

### How did you build your project?

Other (please specify in More Information)

### If you are using Windows, please specify command line type.

CMD

### What is the expected behavior?

Install OK

### What is the actual behavior?

Install error

### Steps to reproduce.

1. Try install v5.3.2
2. Profit

### Build or installation Logs.

```plain
ERROR: Cannot install -r C:\Espressif\frameworks\esp-idf-v5.3.2\tools\requirements\requirements.core.txt (line 15) because these package versions have conflicting dependencies.
                          
                          The conflict is caused by:
                              esp-idf-kconfig 2.3.0 depends on windows-curses; platform_system == "Windows"
                              esp-idf-kconfig 2.2.0 depends on windows-curses; platform_system == "Windows"
                              esp-idf-kconfig 2.1.0 depends on windows-curses; platform_system == "Windows"
                              esp-idf-kconfig 2.0.2 depends on windows-curses; platform_system == "Windows"
                          
                          To fix this you could try to:
                          1. loosen the range of package versions you've specified
                          2. remove package versions to allow pip to attempt to solve the dependency conflict
                          
                          ERROR: ResolutionImpossible: for help visit https://pip.pypa.io/en/latest/topics/dependency-resolution/#dealing-with-dependency-conflicts
                          Traceback (most recent call last):
                            File "C:\Espressif\frameworks\esp-idf-v5.3.2\tools\idf_tools.py", line 3239, in <module>
                              main(sys.argv[1:])
                              ~~~~^^^^^^^^^^^^^^
                            File "C:\Espressif\frameworks\esp-idf-v5.3.2\tools\idf_tools.py", line 3231, in main
                              action_func(args)
                              ~~~~~~~~~~~^^^^^^
                            File "C:\Espressif\frameworks\esp-idf-v5.3.2\tools\idf_tools.py", line 2678, in action_install_python_env
                              subprocess.check_call(run_args, stdout=sys.stdout, stderr=sys.stderr, env=env_copy)
                              ~~~~~~~~~~~~~~~~~~~~~^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
                            File "C:\Program Files\Python313\Lib\subprocess.py", line 419, in check_call
                              raise CalledProcessError(retcode, cmd)
                          subprocess.CalledProcessError: Command '['C:\\Espressif\\python_env\\idf5.3_py3.13_env\\Scripts\\python.exe', '-m', 'pip', 'install', '--no-warn-script-location', '-r', 'C:\\Espressif\\frameworks\\esp-idf-v5.3.2\\tools\\requirements\\requirements.core.txt', '--upgrade', '--constraint', 'C:\\Espressif\\espidf.constraints.v5.3.txt', '--extra-index-url', 'https://dl.espressif.com/pypi']' returned non-zero exit status 1.
```

Temik007, 2024-12-06:
Sorry
remove /USEEMBEDDEDPYTHON=no and it's works

closed 2024-12-06 by Temik007

dehghanmr, 2024-12-08:
Where is /USEEMBEDDEDPYTHON=no?

Zelgoth, 2024-12-08:
Same issue with command line setup.  Could you please explain the fix in a little more detail @Temik007?

ceerqingtingml, 2024-12-13:
@Zelgoth Are you using python 3.13? Please use python 3.12 instead.

Zelgoth, 2024-12-15:
Thank you @ceerqingtingml.  This fixed the issue.  Now using 3.12.8 and it installed just fine.