espressif/esp-idf #14995: esp32 IDF 5.3.1 example SoftAP_STA can't work

MATACE, 2024-12-07:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

The SoftAP_STA can't work in the IDF 5.3.1, but it can work in the IDF 5.2.1，I've checked it.I was in the process of migrating my project from IDF5.2.1 to 5.3.1 and discovered this issue.

MATACE, 2024-12-09:
rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:7176
load:0x40078000,len:15564
ho 0 tail 12 room 4
load:0x40080400,len:4
0x40080400: _init at ??:?

load:0x40080404,len:3904
entry 0x40080640
I (31) boot: ESP-IDF HEAD-HASH-NOTFOUND 2nd stage bootloader
I (31) boot: compile time Dec  9 2024 09:35:25
I (32) boot: Multicore bootloader
I (36) boot: chip revision: v3.1
I (40) boot.esp32: SPI Speed      : 40MHz
I (45) boot.esp32: SPI Mode       : DIO
I (49) boot.esp32: SPI Flash Size : 2MB
I (54) boot: Enabling RNG early entropy source...
I (59) boot: Partition Table:
I (63) boot: ## Label            Usage          Type ST Offset   Length
I (70) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (78) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (85) boot:  2 factory          factory app      00 00 00010000 00100000
I (93) boot: End of partition table
I (97) esp_image: segment 0: paddr=00010020 vaddr=3f400020 size=1f784h (128900) map
I (150) esp_image: segment 1: paddr=0002f7ac vaddr=3ffb0000 size=0086ch (  2156) load
I (151) esp_image: segment 2: paddr=00030020 vaddr=400d0020 size=84348h (541512) map
I (341) esp_image: segment 3: paddr=000b4370 vaddr=3ffb086c size=03664h ( 13924) load
I (346) esp_image: segment 4: paddr=000b79dc vaddr=40080000 size=1734ch ( 95052) load
I (395) boot: Loaded app from partition at offset 0x10000
I (396) boot: Disabling RNG early entropy source...
I (408) cpu_start: Multicore app
I (416) cpu_start: Pro cpu start user code
I (416) cpu_start: cpu freq: 160000000 Hz
I (416) app_init: Application information:
I (419) app_init: Project name:     ap_sta
I (424) app_init: App version:      1
I (428) app_init: Compile time:     Dec  9 2024 09:35:02
I (434) app_init: ELF file SHA256:  8325646b5...
I (439) app_init: ESP-IDF:          HEAD-HASH-NOTFOUND
I (445) efuse_init: Min chip rev:     v0.0
I (450) efuse_init: Max chip rev:     v3.99 
I (455) efuse_init: Chip rev:         v3.1
I (460) heap_init: Initializing. RAM available for dynamic allocation:
I (467) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM
I (473) heap_init: At 3FFB8110 len 00027EF0 (159 KiB): DRAM
I (479) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (485) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (492) heap_init: At 4009734C len 00008CB4 (35 KiB): IRAM
I (500) spi_flash: detected chip: generic
I (503) spi_flash: flash io: dio
W (507) spi_flash: Detected size(4096k) larger than the size in the binary image header(2048k). Using the size in the binary image header.
I (521) main_task: Started on CPU0
I (531) main_task: Calling app_main()
I (581) wifi:wifi driver task: 3ffbfedc, prio:23, stack:6656, core=0
I (591) wifi:wifi firmware version: ccaebfa
I (591) wifi:wifi certification version: v7.0
I (591) wifi:config NVS flash: enabled
I (591) wifi:config nano formating: disabled
I (601) wifi:Init data frame dynamic rx buffer num: 32
I (601) wifi:Init static rx mgmt buffer num: 5
I (611) wifi:Init management short buffer num: 32
I (611) wifi:Init dynamic tx buffer num: 32
I (621) wifi:Init static rx buffer size: 1600
I (621) wifi:Init static rx buffer num: 10
I (621) wifi:Init dynamic rx buffer num: 32
I (631) wifi_init: rx ba win: 6
I (631) wifi_init: accept mbox: 6
[... 35 lines cut ...]
I (4721) esp_netif_handlers: sta ip: 192.168.1.142, mask: 255.255.255.0, gw: 192.168.1.10
I (4721) WiFi Sta: connected to ap SSID:TP-AMU-200 password:amu123456
I (4721) wifi:<ba-del>idx:0, tid:0
W (4731) esp_netif_lwip: napt disabled on esp_netif:0x3ffc5c3c
I (4731) wifi:<ba-add>idx:0 (ifx:0, 78:60:5b:b7:6f:7b), tid:0, ssn:2, winSize:64
I (4741) main_task: Returned from app_main()
I (15941) wifi:new:<6,2>, old:<6,0>, ap:<6,2>, sta:<6,0>, prof:11, snd_ch_cfg:0x0
I (15941) wifi:station: 62:8f:d9:26:f5:2e join, AID=1, bgn, 40D
I (16021) WiFi SoftAP: Station 62:8f:d9:26:f5:2e joined, AID=1
I (16041) wifi:<ba-add>idx:2 (ifx:1, 62:8f:d9:26:f5:2e), tid:0, ssn:0, winSize:64
I (16061) esp_netif_lwip: DHCP server assigned IP to a client, IP is: 192.168.4.2
I (17141) esp_netif_lwip: DHCP server assigned IP to a client, IP is: 192.168.4.2
I (18231) wifi:<ba-add>idx:3 (ifx:1, 62:8f:d9:26:f5:2e), tid:6, ssn:0, winSize:64
I (18281) wifi:<ba-add>idx:1 (ifx:0, 78:60:5b:b7:6f:7b), tid:3, ssn:0, winSize:64
I (20121) wifi:<ba-del>idx:1, tid:3
I (20151) wifi:<ba-add>idx:1 (ifx:0, 78:60:5b:b7:6f:7b), tid:3, ssn:1, winSize:64
I (20921) wifi:<ba-add>idx:4 (ifx:1, 62:8f:d9:26:f5:2e), tid:5, ssn:0, winSize:64
I (69241) wifi:station: 62:8f:d9:26:f5:2e leave, AID = 1, reason = 3, bss_flags is 756835, bss:0x3ffb9dbc
I (69241) wifi:new:<6,0>, old:<6,2>, ap:<6,2>, sta:<6,0>, prof:11, snd_ch_cfg:0x0
I (69251) wifi:<ba-del>idx:2, tid:0
I (69251) wifi:<ba-del>idx:4, tid:5
I (69251) wifi:<ba-del>idx:3, tid:6
I (69251) WiFi SoftAP: Station 62:8f:d9:26:f5:2e left, AID=1, reason:3

This is the log of softap_sta, showing that WiFi is not available

MATACE, 2024-12-09:
This softap_sta example does not work with vscode IDF plugin 5.3.1 and 5.2.3 esp32, showing that the network is not working properly, 5.2.1 works with platformio, my esp32 is esp32 wroom

MATACE, 2024-12-09:
I (11730) WiFi SoftAP: Station 62:8f:d9:26:f5:2e joined, AID=1
I (11760) wifi:<ba-add>idx:2 (ifx:1, 62:8f:d9:26:f5:2e), tid:0, ssn:0, winSize:64
I (11770) esp_netif_lwip: DHCP server assigned IP to a client, IP is: 192.168.4.2
I (11960) wifi:<ba-add>idx:3 (ifx:1, 62:8f:d9:26:f5:2e), tid:6, ssn:0, winSize:64
I (12280) wifi:[ADDBA]RX DELBA, reason:37, delete tid:0, initiator:0(recipient)
I (12320) wifi:[ADDBA]RX DELBA, reason:37, delete tid:0, initiator:0(recipient)
I (14060) wifi:<ba-add>idx:1 (ifx:0, 78:60:5b:b7:6f:7b), tid:3, ssn:0, winSize:64
I (16220) wifi:<ba-add>idx:4 (ifx:1, 62:8f:d9:26:f5:2e), tid:1, ssn:0, winSize:64
I (28920) wifi:[ADDBA]RX DELBA, reason:37, delete tid:5, initiator:0(recipient)
I (28930) wifi:[ADDBA]RX DELBA, reason:37, delete tid:5, initiator:0(recipient)
I (28950) wifi:[ADDBA]RX DELBA, reason:37, delete tid:5, initiator:0(recipient)
I (30580) wifi:[ADDBA]RX DELBA, reason:37, delete tid:5, initiator:0(recipient)
I (30590) wifi:[ADDBA]RX DELBA, reason:37, delete tid:5, initiator:0(recipient)
I (30600) wifi:[ADDBA]RX DELBA, reason:37, delete tid:5, initiator:0(recipient)
I (30700) wifi:[ADDBA]RX DELBA, reason:37, delete tid:5, initiator:0(recipient)
I (30700) wifi:[ADDBA]RX DELBA, reason:37, delete tid:5, initiator:0(recipient)
I (30710) wifi:[ADDBA]RX DELBA, reason:37, delete tid:5, initiator:0(recipient)

MATACE, 2024-12-09:
This can be executed on ESP32-U4WDH, but not on ESP32-D0WD-V3

nopnop2002, 2024-12-17:
What happens if you comment this out from sdkconfig?
```
CONFIG_ESP_WIFI_SOFTAP_SAE_SUPPORT=y
```

hansw123, 2025-02-12:
@MATACE 
sorry for late reply 
from the log, the station(62:8f:d9:26:f5:2e) which connected to the softap decide to leave . the reason may be the station ping fail or other reason.

hansw123, 2025-02-12:
@MATACE 
the root cause need more detail info ,could you provide the wifi wireshark capture?

AxelLin, 2025-02-24:
@MATACE Could you provide the wireshark log?

Alvin1Zhang, 2025-03-06:
@MATACE Thanks for reporting, can you please help provide more details? Thanks.

closed 2025-04-07 by Alvin1Zhang