espressif/esp-idf #14996: USB Host sometimes freezes randomly

alex-k8, 2024-12-07:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.3.1

### Espressif SoC revision.

ESP32-S3

### Operating System used.

Windows

### How did you build your project?

VS Code IDE

### If you are using Windows, please specify command line type.

PowerShell

### Development Kit.

ESP32-S3-DevKitC-1 & Custom Board

### Power Supply used.

USB

### What is the expected behavior?

I expected the USB Host functionality to work without randomly freezing.

### What is the actual behavior?

Randomly, either after 5 minutes or 5 hours, the USB Host does not receive any more data from the connected device. I press keys on my keyboard, or move my mouse, and I don't see any logs. I either have to replug the device, or restart the ESP board. This happens only on a few keyboards and mice. I suspect it has something to do with higher polling rate devices, since I never experienced it on my older keyboard.

### Steps to reproduce.

1. Build and upload the HID Host Example code
2. Connect a high polling rate mouse to the ESP
3. Move the mouse and see that you receive data properly
4. Move it for some time...
5. After some time the logs will not show any more output, it's like the device either stops sending data, or the library stops receiving data. However the "Disconnect - Device gone" callback is called if I remove the device.

### Debug Logs.

```plain
Nothing more is shown in the logs, no matter what log level I choose.
```

### More Information.

The only change I made to the code was that I use `hid_class_request_set_protocol(hid_device_handle, HID_REPORT_PROTOCOL_REPORT)` instead of `hid_class_request_set_protocol(hid_device_handle, HID_REPORT_PROTOCOL_BOOT)` in order to receive all the data that I need. This happens on a custom board as well as the ESP32-S3-DevKitC-1. It happens in both USB power and battery power. I've logged the available memory and there's no memory leak. Seems to be software related and not hardware.

roma-jam, 2024-12-07:
Hi @alex-k8, 

Could you please attach the `usb_host_lib` example output for all devices you experience problems with? 

Thanks.

alex-k8, 2024-12-07:
Hey @roma-jam 

I only have access to two of them at the moment, so I'll attach those.

CharaChorder Lite:
```
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x1 (POWERON),boot:0x8 (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce2810,len:0x178c
load:0x403c8700,len:0x4
load:0x403c8704,len:0xcb8
load:0x403cb700,len:0x2db0
entry 0xESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x1 (POWERON),boot:0x8 (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce2810,len:0x178c
load:0x403c8700,len:0x4
load:0x403c8704,len:0xcb8
load:0x403cb700,len:0x2db0
entry 0x403c8914
I (27) boot: ESP-IDF v5.3.1 2nd stage bootloader
I (27) boot: compile time Oct 24 2024 20:46:56
I (27) boot: Multicore bootloader
I (30) boot: chip revision: v0.1
I (34) boot.esp32s3: Boot SPI Speed : 80MHz
I (38) boot.esp32s3: SPI Mode       : DIO
I (43) boot.esp32s3: SPI Flash Size : 2MB
I (48) boot: Enabling RNG early entropy source...
I (53) boot: Partition Table:
I (57) boot: ## Label            Usage          Type ST Offset   Length
I (64) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (72) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (79) boot:  2 factory          factory app      00 00 00010000 00100000
I (87) boot: End of partition table
I (91) esp_image: segment 0: paddr=00010020 vaddr=3c020020 size=0b714h ( 46868) map
I (108) esp_image: segment 1: paddr=0001b73c vaddr=3fc92700 size=028e4h ( 10468) load
I (110) esp_image: segment 2: paddr=0001e028 vaddr=40374000 size=01ff0h (  8176) load
I (118) esp_image: segment 3: paddr=00020020 vaddr=42000020 size=1d6dch (120540) map
I (146) esp_image: segment 4: paddr=0003d704 vaddr=40375ff0 size=0c6c8h ( 50888) load
I (163) boot: Loaded app from partition at offset 0x10000
I (164) boot: Disabling RNG early entropy source...
I (176) cpu_start: Multicore app
I (185) cpu_start: Pro cpu start user code
I (185) cpu_start: cpu freq: 160000000 Hz
I (185) app_init: Application information:
I (188) app_init: Project name:     usb_host_lib_example
I (194) app_init: App version:      1
I (198) app_init: Compile time:     Dec  6 2024 19:35:39
I (204) app_init: ELF file SHA256:  c8e55ece3...
I (210) app_init: ESP-IDF:          v5.3.1
I (214) efuse_init: Min chip rev:     v0.0
I (219) efuse_init: Max chip rev:     v0.99 
I (224) efuse_init: Chip rev:         v0.1
I (229) heap_init: Initializing. RAM available for dynamic allocation:
I (236) heap_init: At 3FC95848 len 00053EC8 (335 KiB): RAM
I (242) heap_init: At 3FCE9710 len 00005724 (21 KiB): RAM
I (248) heap_init: At 3FCF0000 len 00008000 (32 KiB): DRAM
I (254) heap_init: At 600FE100 len 00001EE8 (7 KiB): RTCRAM
I (261) spi_flash: detected chip: generic
I (265) spi_flash: flash io: dio
W (269) spi_flash: Detected size(8192k) larger than the size in the binary image header(2048k). Using the size in the binary image header.
I (282) sleep: Configure to isolate all GPIO pins in sleep state
I (289) sleep: Enable automatic switching of GPIO sleep configuration
I (297) main_task: Started on CPU0
I (307) main_task: Calling app_main()
[... 229 lines cut ...]
        bInterfaceNumber 1
        bAlternateSetting 0
        bNumEndpoints 1
        bInterfaceClass 0x3
        bInterfaceSubClass 0x0
        bInterfaceProtocol 0x0
        iInterface 0
                *** Endpoint descriptor ***
                bLength 7
                bDescriptorType 5
                bEndpointAddress 0x82   EP 2 IN
                bmAttributes 0x3        INT
                wMaxPacketSize 20
                bInterval 1
I (857) CLASS: Getting Manufacturer string descriptor
Logitech
I (867) CLASS: Getting Product string descriptor
G403 HERO Gaming Mouse
I (867) CLASS: Getting Serial Number string descriptor
0C80387B3636
```

Let me know if you need anything further. I've been trying to troubleshoot this for months but unfortunately I couldn't figure it out.

Thank you!

roma-jam, 2024-12-07:
Hi @alex-k8 , 

Thanks for the prompt response. I‘ll try to find a way to reproduce the problem. 

Meanwhile, could you try to reproduce it with `esp-idf`, `release v5.1`  and help to confirm, that the problem is also present there? 

Thanks.

alex-k8, 2024-12-07:
Thank you @roma-jam
I'll downgrade and let you know. I usually reproduce it by spamming keys, but I'm not sure if that's actually what's causing it. I think I had let the esp run all night on its own, without pressing buttons, and the next morning it had not froze, so it's probably related to the callbacks that are called when receiving key input.

alex-k8, 2024-12-07:
Hey @roma-jam 

I managed to reproduce it on v5.1.2. I reproduced it twice in about 10 minutes with my mouse. I think it's running either at 500 or 1000hz. Mouse data is definitely faster than keyboard data so if my theory is correct, it makes sense because the keyboard cannot send data that fast compared to the mouse.

So if you have a gaming mouse or something laying around, I suspect you will be able to reproduce it. Unfortunately I don't have another high polling rate mouse to test it. I tested it with a cheap aliexpress mouse and I couldn't reproduce it, at least in the 30 minutes that I tried.

alex-k8, 2024-12-19:
Hi @roma-jam 
Did you by any chance manage to reproduce it? I got another report from a Keychron Q11 user that this issue is happening there as well. Thanks!

tore-espressif, 2024-12-19:
@alex-k8 We reproduced the problem. We plan to work on the fix in Q1/2025. Thank you for your patience, we will update this ticket once the fix is out

alex-k8, 2024-12-19:
Thank you @tore-espressif ! Awaiting eagerly your update. Let me know if you need anything further from me.

tore-espressif, 2025-02-13:
Hello @alex-k8 ,

so far it seems like a HW erratum.

You can try this hotfix:

[image]

This brings some additional latency and it needs more testing before we can release the fix in esp-idf

alex-k8, 2025-02-13:
Hey @tore-espressif , thank you very much! 1ms latency sounds like a fair tradeoff. Unfortunately, I could still reproduce it with my G403 mouse. it took around 20-30 minutes, not sure if it's random or not, but it still froze. I downloaded the github repo, made the change in the file you mentioned, and through VS Code's Configure extension screen of ESP-IDF, I added the repo as a new esp-idf version through the USE EXISTING SETUP, then selected that when building. Was the process correct?

tore-espressif, 2025-02-13:
@alex-k8 yes, the process sounds correct.

Nevertheless, the cheap fix did not do the trick... I'll give it few more days and the buy the G403 mouse...

alex-k8, 2025-02-14:
@tore-espressif I'm available for any test you want. The mouse specifically is the G403 Hero 16000 DPI. I see there is a 25000 DPI one now as well, I suppose they will both act the same. However I can test anything and save you the hustle of buying one, unless you believe it will be easier to troubleshoot on your own.

Shane-Wood-TL, 2025-02-15:
Hey @tore-espressif,
I am having the same issue as @alex-k8 although I am using a ps4 controller (so 250 Hz? polling rate). Same results after a period of time (I have had it working fine for over an hour) it eventually stops processing usb data. I also attempted to utilize the cheat to resolve the issue and it ended with the same results as before.

tore-espressif, 2025-02-17:
Hello all,
Could you please confirm that you have no logs in the RX callback and that the RX callback returns very fast (eg. just memcpy the data somewhere and return)?

I was able to run the test for 10+ hours without a problem when I disabled all logs in my tests...

alex-k8, 2025-02-18:
Hey @tore-espressif , seems promising! I've been testing it for the last hour. Unfortunately after 1 hour it froze again. @Shane-Wood-TL  could you also try and verify you see the same behavior?
I'll retry to see if it's reproducable.

Shane-Wood-TL, 2025-02-19:
No change for me, I saw a message(I cannot find it now?) about making the rx callback just call memcpy (or something just as fast). I will implement that change and edit this message to see if that changes anything as my current callback is doing some preliminary data processing + enqueuing data.

Update: 2 hours later still working when the rx callback was reduced. My current callback sets 4 atomic uint8_ts and gives a mutex( so nothing blocking). I will let it run overnight and update in the morning.

tore-espressif, 2025-02-19:
> making the rx callback just call memcpy (or something just as fast)

This fixed the problem for me too, but I was not happy with the fix. The USB Host Lib should allow longer executing callbacks too...

alex-k8, 2025-03-09:
Hey @tore-espressif ! Any luck or any other ideas? I got another report from a user which I think is the same issue. Their mouse is the Logitech Pro X.

@Shane-Wood-TL I think I missed the original message, could you tell me what exactly you changed? Which RX method is that?

Shane-Wood-TL, 2025-03-09:
@alex-k8 I modified the hid_host_generic_report_callback() function to be as quick as possible

[image]

The 4 bytes that are being modified are atomic (so no need for semaphores or similar). The semaphore is taken by a task that I have deemed ps4_data_processor, this task only ever acquires the semaphore, but does not release it. This ensures that the received data is only processed once. 

This appears to work without issues, I will continue to run tests, but it I have not had an issue since I switched to this (and moved away from a breadboard).

jakla123, 2025-10-24:
I'm experiencing the same issue, using Arduino along with the EspUsbHost library, as previously mentioned by @alex-k8 .
Has there been any update regarding this problem? Im testing it with ESP32-S3-DevKitC-1. 

Any guidance or progress would be greatly appreciated.

alex-k8, 2026-10-31:
Hello @tore-espressif ,

I'd like to kindly revisit this issue, as it doesn't seem that a definitive fix has been implemented yet. Unfortunately, this bug is still affecting my product, and it's becoming increasingly difficult to work around.

Has there been any further investigation or progress on this? Perhaps we could explore alternative approaches or investigate the underlying issue further.

I'd really appreciate any update or guidance. Thank you for your time!

tore-espressif, 2026-11-02:
@alex-k8 For simple HID host you can try this alternative USB Host Stack https://components.espressif.com/components/cherry-embedded/cherryusb

It uses Buffer DMA mode instead of Scatter-Gather DMA mode. The mode is simpler and might not suffer from the freezes that you reported.

alex-k8, 2026-11-03:
@tore-espressif on a minimal example that I did, so far it seems pretty solid! I will have to migrate my code to that to see it in action, I will keep you updated, thank you!