espressif/esp-idf #14998: Continuous ADC Sample Values are Decimated through USB

dustinlaroche, 2024-12-07:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

If I set the sample rate on the ESP32-C3 and run that for one minute, and then verify how many sample values there are, then have them outputted to serial port through USB, they do not match in value number size and/or length.

Icarus113, 2024-12-10:
Could you help give more info about the mismatch between the size of the results and the one when via USB?

dustinlaroche, 2024-12-10:
I have the sample rate set for 83333 Hz. It takes 160 seconds to acquire 2376800 samples. This does not equate to the correct sample rate. I have the serial baud rate set for 2,000,000.

Icarus113, 2024-12-11:
May I know how many ADC channels you captured? @dustinlaroche

dustinlaroche, 2024-12-11:
Only one channel.