espressif/esp-idf #14999: esp_mbedtls_server_session_create Stuck in infinite loop if WiFi Client drops during handshake

avrmp, 2024-12-07:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.4-dev-4375-gf420609c33-dirty

### Espressif SoC revision.

ESP32-S3 v0.2

### Operating System used.

Windows

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

PowerShell

### Development Kit.

Custom Board

### Power Supply used.

USB

### What is the expected behavior?

Using ASYNC HTTP Server example with TLS. When a WiFi Client sends http requests and abbruptly drops/closes socket during the handshake, the httpd_ssl_open function should return with an error and server should continiue normal oeration.

### What is the actual behavior?

When a WiFi Client sends http requests and abbruptly drops/closes socket during the handshake, the server gets stuck inside of the "esp_mbedtls_server_session_create" functions's loop and never returns, leading to the server being none-responsive until restart:

while ((ret = mbedtls_ssl_handshake(&tls->ssl)) != 0) {
        if (ret != ESP_TLS_ERR_SSL_WANT_READ && ret != ESP_TLS_ERR_SSL_WANT_WRITE) {
            ESP_LOGE(TAG, "mbedtls_ssl_handshake returned -0x%04X", -ret);
            mbedtls_print_error_msg(ret);
            ESP_INT_EVENT_TRACKER_CAPTURE(tls->error_handle, ESP_TLS_ERR_TYPE_MBEDTLS, -ret);
            ESP_INT_EVENT_TRACKER_CAPTURE(tls->error_handle, ESP_TLS_ERR_TYPE_ESP, ESP_ERR_MBEDTLS_SSL_HANDSHAKE_FAILED);
            tls->conn_state = ESP_TLS_FAIL;
            return ret;
        }
    }

### Steps to reproduce.

1. Download & setup version of esp-idf discussed above
2. Copy examples/protocols/http_server/async_handlers to your project directory
3. Modify start_webserver dunction so it uses tls similar to this: `httpd_ssl_config_t ssl_config = HTTPD_SSL_CONFIG_DEFAULT();
        ssl_config.httpd.server_port = settings.port;
        config.lru_purge_enable = true; 
        ssl_config.httpd.enable_so_linger = false;
        config.linger_timeout = 0;
        ssl_config.httpd.uri_match_fn = httpd_uri_match_wildcard;
        ssl_config.httpd.max_uri_handlers = MAX_NUM_HANDLERS;
        ssl_config.httpd.max_resp_headers = MAX_NUM_HANDLERS;
        ssl_config.httpd.max_open_sockets = MAX_NUM_CLIENTS;
        ssl_config.servercert = (uint8_t*)settings.server_cert_pem;
        ssl_config.servercert_len = settings.server_cert_len;
        ssl_config.cacert_pem = (uint8_t*)settings.ca_cert_pem;//Client Cert or it's authority
        ssl_config.cacert_len = settings.ca_cert_len;
        ssl_config.prvtkey_pem = (uint8_t*)settings.private_key_pem;
        ssl_config.prvtkey_len = settings.private_key_len;
        ssl_config.port_secure = settings.port;
        ssl_config.session_tickets = false;
         ret = httpd_ssl_start(&server, &ssl_config);`
5. Run the example
6. Send a POST request to the server from another esp32 client connected to the server device using softap
7. Unpower the client as soon as you see the server prints "performing session handshake"

### More Information.

Adding an "else" statement to the if statement inside the loop withing the "esp_mbedtls_server_session_create" function indicates the "mbedtls_ssl_handshake" keeps returning with "ESP_TLS_ERR_SSL_WANT_READ". indefenitely.

avrmp, 2024-12-25:
Just, bumping this up

AdityaHPatwardhan, 2025-01-20:
Hi @avrmp Thanks for the issue and apologies for the delayed response.
Yes, I think the issue here is that there is no fixed timeout defined for the server tls connection as available for the client connection.
I think adding a timeout value will fix this infinite loop issue.
I will add the necessary fix soon.
Thanks,
Aditya

closed 2025-02-24 by espressif-bot