espressif/esp-idf #15000: How to change buf_len in int httpd_req_recv(httpd_req_t *r, char *buf, size_t buf_len)

zzzh, 2024-12-08:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

I tried to set buf_len to 5120 or 10240, but it's useless, it's always 1024.

The esp-idf version I use is 4.4.2, can't use lastest version for compatible issue.

nileshkale123, 2024-12-09:
Hello @zzzh,

Where have you tried to change this value for buf_len? Could you please send the patch to reproduce the issue? 
Do you want to change the maximum HTTP request size or anything else?

nileshkale123, 2024-12-09:
Hello @zzzh,

Could you kindly provide more details about what you aim to achieve by modifying buf_len? Additionally, it would be very helpful if you could share the patch file for the changes you've made related to this issue.

Please also share any relevant debug logs and examples of your modifications so that we can replicate and analyze the issue on our side.

Looking forward to your input!

zzzh, 2024-12-10:
Hello, @nileshkale123,

I think increase the size of each transfer may speed up the whole upload progress, that's why.
I did not change anything and it works well except slow (need 1 minute to transfer 1MB), by changing the transfer size from 1k to 10k, I hope it can have 10 times speed.

nopnop2002, 2024-12-13:
@zzzh 

There is no limit to the size of data received by httpd_req_recv.

The return value of httpd_req_recv is the actual number of bytes received.

When using a 2048-byte buffer, processing is performed in 2048-byte units as shown below.

There may be a problem on your sending side.

```
W (36311) example: remaining=4576
W (36311) example: httpd_req_recv ret=2048
W (36311) example: httpd_req_recv ret=2048
W (36311) example: httpd_req_recv ret=480
```

zzzh, 2024-12-18:
> @zzzh
> 
> There is no limit to the size of data received by `httpd_req_recv`.
> 
> The return value of `httpd_req_recv` is the actual number of bytes received.
> 
> When using a 2048-byte buffer, processing is performed in 2048-byte units as shown below.
> 
> ```
> W (36311) example: remaining=4576
> W (36311) example: httpd_req_recv ret=2048
> W (36311) example: httpd_req_recv ret=2048
> W (36311) example: httpd_req_recv ret=480
> ```
> 
> When using a 4096-byte buffer, processing is performed in 4096-byte units as shown below.
> 
> ```
> W (13551) example: remaining=4576
> W (13551) example: httpd_req_recv ret=4096
> W (13551) example: httpd_req_recv ret=480
> ```
> 
> When using a 4096-byte buffer, a 1M byte transfer will complete in 1.5 second.
> 
> ```
> W (17061) example: received=1048576 elapsed time[ms]:1510
> ```
> 
> I used this to send.
> 
> ```
> curl -X POST --data-binary @1M.dummy  192.168.10.111/echo
> ```
> 
> There may be a problem on your sending side.

Hi, nopnop2002,

Good to know, what version of esp-idf did you test on?
I'm using v4.4.2, maybe too old.

nopnop2002, 2024-12-18:
>what version of esp-idf did you test on?

$ git describe --tags
v5.3.2-1-g9a6aba7401

zzzh, 2024-12-19:
ok, I'll test it on the latest version.

closed 2024-12-19 by zzzh