espressif/esp-idf #15002: Is there a way to parse HTTP headers following a esp_http_client_fetch_headers() call

Dazza0, 2024-12-09:
### Is your feature request related to a problem?

Hi, I'm currently trying to do a native HTTP request using the  `esp_http_client.h` API i.e., without calling the `esp_http_perform()` function (similar to the [`http_native_request()`](https://github.com/espressif/esp-idf/blob/v5.3.2/examples/protocols/esp_http_client/main/esp_http_client_example.c#L706) function in the esp_http_client example. The reason being that I want to manually fetch headers and parse them, before conditionally sending a GET request for the actual data payload.

However, after calling `esp_http_client_fetch_headers()` to fetch the headers, the `esp_http_client.h` API doesn't provide a way to access the buffer storing the headers.

Is there a recommended way to get access to those headers and parse them?

nopnop2002, 2024-12-10:
Is this no good?
https://github.com/espressif/esp-idf/tree/master/examples/protocols/http_request

You can get HTTP headers.
```
HTTP/1.0 200 OK
Accept-Ranges: bytes
Age: 372137
Cache-Control: max-age=604800
Content-Type: text/html
Date: Tue, 10 Dec 2024 02:05:49 GMT
Etag: "3147526947"
Expires: Tue, 17 Dec 2024 02:05:49 GMT
Last-Modified: Thu, 17 Oct 2019 07:18:26 GMT
Server: ECAcc (lac/5598)
Vary: Accept-Encoding
X-Cache: HIT
Content-Length: 1256
Connection: close
```

Dazza0, 2024-12-10:
@nopnop2002 

> Is this no good?
> https://github.com/espressif/esp-idf/tree/master/examples/protocols/http_request

That also works, but that would be operating at a lower level of abstraction. I was hoping to operate that the ESP HTTP Client layer where TLS is handled.

> You can still get the header with this.
> https://github.com/espressif/esp-idf/tree/v5.3.2/examples/protocols/esp_http_client
> ```
> I (9666) HTTP: HTTP_EVENT_ON_HEADER, key=Content-Type, value=application/rss+xml
> I (9666) HTTP: HTTP_EVENT_ON_HEADER, key=Server, value=Apache/2.4.37 (Rocky Linux) OpenSSL/1.1.1k
> I (9666) HTTP: HTTP_EVENT_ON_HEADER, key=ETag, value="b6752d053f69b6ed4c117feead10316d8937aad89bf27ee230b50b55a09f33a0"
> I (9686) HTTP: HTTP_EVENT_ON_HEADER, key=Access-Control-Allow-Origin, value=*
> I (9686) HTTP: HTTP_EVENT_ON_HEADER, key=expiry_extended_seconds, value=0
> I (9696) HTTP: HTTP_EVENT_ON_HEADER, key=Cache-Control, value=no-transform, max-age=120
> I (9706) HTTP: HTTP_EVENT_ON_HEADER, key=Date, value=Tue, 10 Dec 2024 02:34:07 GMT
> I (9716) HTTP: HTTP_EVENT_ON_HEADER, key=Content-Length, value=3545
> I (9716) HTTP: HTTP_EVENT_ON_HEADER, key=Connection, value=keep-alive
> ```

Is that done via the `esp_http_client_perform()` function?

nopnop2002, 2024-12-10:
>Is that done via the esp_http_client_perform() function?

Yes.

mahavirj, 2024-12-16:
@Dazza0 

> However, after calling esp_http_client_fetch_headers() to fetch the headers, the esp_http_client.h API doesn't provide a way to access the buffer storing the headers.

Once you call `esp_http_client_fetch_headers`, the API shall enable all headers to be available through the registered event handler. Please register event handler using:

https://github.com/espressif/esp-idf/blob/cb3ac7429c73b6e7c6d79ae05de480d4b6706770/components/esp_http_client/include/esp_http_client.h#L163

And then the header shall be available in the event handler through:

https://github.com/espressif/esp-idf/blob/cb3ac7429c73b6e7c6d79ae05de480d4b6706770/components/esp_http_client/include/esp_http_client.h#L57-L58

mahavirj, 2024-12-18:
Here is the quick patch for this on `esp_http_client` example:

https://gist.github.com/mahavirj/2b50cb1fdce832e75c664c318804666a

Build and run on `linux` target:

```
~/.../protocols/esp_http_client (master)$ ./build/esp_http_client_example.elf 
I (678383732) port: Starting scheduler.
I (678383741) HTTP_CLIENT: Connected to AP, begin http example
I (678384213) HTTP_CLIENT: fetching headers
I (678384712) HTTP_CLIENT: HTTP_EVENT_ON_HEADER, key=Date, value=Wed, 18 Dec 2024 02:44:14 GMT
I (678384712) HTTP_CLIENT: HTTP_EVENT_ON_HEADER, key=Content-Type, value=application/json
I (678384713) HTTP_CLIENT: HTTP_EVENT_ON_HEADER, key=Content-Length, value=242
I (678384713) HTTP_CLIENT: HTTP_EVENT_ON_HEADER, key=Connection, value=keep-alive
I (678384713) HTTP_CLIENT: HTTP_EVENT_ON_HEADER, key=Server, value=gunicorn/19.9.0
I (678384713) HTTP_CLIENT: HTTP_EVENT_ON_HEADER, key=Access-Control-Allow-Origin, value=*
I (678384713) HTTP_CLIENT: HTTP_EVENT_ON_HEADER, key=Access-Control-Allow-Credentials, value=true
I (678384713) HTTP_CLIENT: fetching headers done
I (678384713) HTTP_CLIENT: HTTP Stream reader Status = 200, content_length = 242
I (678384713) HTTP_CLIENT: HTTP_EVENT_DISCONNECTED
I (678384714) HTTP_CLIENT: HTTP_EVENT_DISCONNECTED
```

Hope this helps!

closed 2024-12-18 by mahavirj