espressif/esp-idf #15003: Crash when creating "main" task

ammaree, 2024-12-09:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

In order for our application to function weaned to assign a unique mask to each task, preferably at creating time.
This mask we store as an extra/reserved Local Storage Pointer in the FreeRTOS TCB. 
For our own application created tasks this can be done at any stage and works like a charm.
However, for external library/component created tasks, such as LVGL and others, this is a lot more difficult.

In order to try make the process transparent we have wrapped the xxTaskCreate?????() APIs to add functionality to allocate the masks on the fly and store it in the correct LSP slot immediately after creation.

This process works perfectly for the "ipc0/1" "esp_timer" "TmrSvc" "sys_event" "tiT" "wifi" and the app tasks.
The "idle0/1" tasks are created by FreeRTOS using internal APIs hence do not get maks, but that's not a problem.
The problem shows it when "main" is created using this mechanism, causing firmware to crash in vTaskAllocateMask() at the vTaskSetThreadLocalStoragePointer().

We have tried everything we can think of to fix, but no luck. All we have been able to see is that the stack pointer, when measured at __real_xTaskCreatePinnedToCore() is at a higher address for "main" than any other task.

We have tried increasing the various stack sizes accessible through menuconfig, primarily we hoped CONFIG_ESP_MINIMAL_SHARED_STACK_SIZE would make a difference, but no luck.

When we add the check for "main", as can be seen below, and add the extra vTaskSetThreadLocalStoragePointer() call at the start of app_main(), everything starts and functions perfectly. In case it may help, the SP at the start of the task creation process is as follows: 

'ipc0' SP=0x3ffe3bb0
'ipc1' SP=0x3ffe3bb0
'esp_timer' SP=0x3ffe3b80
'main' SP=0x3ffe3bc0

'sys_evt' SP=0x3ffc6690
'tiT' SP=0x3ffc6880
'wifi' SP=0x3ffc6830

Any suggestions?
```
static u32_t TaskTracker = 0xFF000000;					// reserve top 8 bits, 

void vTaskAllocateMask(TaskHandle_t xHandle) {
#if	(portNUM_PROCESSORS > 1)
	BaseType_t btSR = xRtosSemaphoreTake(&shTaskInfo, portMAX_DELAY);
#endif
	// Find next empty slot, mark as allocated, set as "LSP" in new task TCB
	u32_t Mask = 0x80000000 >> __builtin_clzl(~TaskTracker);
	TaskTracker |= Mask;
	vTaskSetThreadLocalStoragePointer(xHandle, buildFRTLSP_EVT_MASK, (void *)Mask);
#if	(portNUM_PROCESSORS > 1)
	if (btSR == pdTRUE) xRtosSemaphoreGive(&shTaskInfo);
#endif
}

BaseType_t __real_xTaskCreatePinnedToCore(TaskFunction_t, const char * const, const u32_t, void *, UBaseType_t, TaskHandle_t *, const BaseType_t);
BaseType_t __wrap_xTaskCreatePinnedToCore(TaskFunction_t pxTaskCode, const char * const pcName, const u32_t usStackDepth, void * pvParameters, UBaseType_t uxPriority, TaskHandle_t * pxCreatedTask, const BaseType_t xCoreID) {
	BaseType_t btRV = __real_xTaskCreatePinnedToCore(pxTaskCode, pcName, usStackDepth, pvParameters, uxPriority, pxCreatedTask, xCoreID);
	IF_myASSERT(debugTRACK, btRV == pdPASS);
	//Crash here ONLY with "main" (stack??) if vTaskSetThreadLocalStoragePointer() called in vTaskAllocateMask
	//vTaskAllocateMask(*pxCreatedTask);
	if (strcmp(pcName, "main") != 0) vTaskAllocateMask(*pxCreatedTask);
	return btRV;
}
```

sudeep-mohanty, 2024-12-10:
Hi @ammaree, Thanks for reporting the issue. Could you also share which target are you building for and your `sdkconfig` if possible?

Also, have you tried to increase the `CONFIG_MAIN_TASK_STACK_SIZE` in the menuconfig?

ammaree, 2024-12-10:
@sudeep-mohanty thanks for quick response

I am building for ESP32 and ESP32-S3.
I cannot see why CONFIG_MAIN_TASK_STACK_SIZE should play any role since it specifies the since of stack size for "main", the task we have just successfully created and are trying to set the LSP for, at which time things fail.
But, we have tried changing CONFIG_MAIN_TASK_STACK_SIZE from 3584 to 5120, as expected no difference.

SDKCONFIG is attached [sdkconfig.txt](https://github.com/user-attachments/files/18079866/sdkconfig.txt).

Crash dump output below:

```
entry 0x400805e0
0.390 (2) 0 boot esp_psram Virtual address not enough for PSRAM, map as much as we can. 4MB is mapped
[SP=0x3ffe3bb0 ipc0]
[SP=0x3ffe3bb0 ipc1]
[SP=0x3ffe3b80 esp_timer]
[SP=0x3ffe3bc0 main]
Guru Meditation Error: Core  0 panic'ed (LoadProhibited). Exception was unhandled.

Core  0 register dump:
PC      : 0x400db54c  PS      : 0x00060130  A0      : 0x801a8ccd  A1      : 0x3ffe3bc0  
--- 0x400db54c: __wrap_xTaskCreatePinnedToCore at /Users/andremaree/DevSpace/z-comp/rtos-support/FreeRTOS_Support.c:393

A2      : 0x401a8bb8  A3      : 0x3f4031dd  A4      : 0x00001000  A5      : 0x00000000  
--- 0x401a8bb8: main_task at /Users/andremaree/DevSpace/z-sdk/master/components/freertos/app_startup.c:159

A6      : 0x00000001  A7      : 0x00000000  A8      : 0x800db538  A9      : 0x3ffe3b80  
A10     : 0x00000001  A11     : 0x3f4031dd  A12     : 0x00001000  A13     : 0x7fffffff  
A14     : 0x3ffc5b18  A15     : 0x3ffafd80  SAR     : 0x00000020  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x00000000  LBEG    : 0x40094caa  LEND    : 0x40094cb5  LCOUNT  : 0x00000000  
--- 0x40094caa: memset at /builds/idf/crosstool-NG/.build/xtensa-esp-elf/src/newlib/newlib/libc/machine/xtensa/memset.S:150
0x40094cb5: memset at /builds/idf/crosstool-NG/.build/xtensa-esp-elf/src/newlib/newlib/libc/machine/xtensa/memset.S:160

Backtrace: 0x400db549:0x3ffe3bc0 0x401a8cca:0x3ffe3bf0 0x400d4d06:0x3ffe3c20 0x40081a5e:0x3ffe3c50 0x40079772:0x3ffe3c90 |<-CORRUPTED
--- 0x400db549: __wrap_xTaskCreatePinnedToCore at /Users/andremaree/DevSpace/z-comp/rtos-support/FreeRTOS_Support.c:389 (discriminator 3)
0x401a8cca: esp_startup_start_app at /Users/andremaree/DevSpace/z-sdk/master/components/freertos/app_startup.c:83
0x400d4d06: start_cpu0_default at /Users/andremaree/DevSpace/z-sdk/master/components/esp_system/startup.c:216
0x40081a5e: call_start_cpu0 at /Users/andremaree/DevSpace/z-sdk/master/components/esp_system/port/cpu_start.c:853 (discriminator 1)
```

sudeep-mohanty, 2024-12-11:
Hi @ammaree, The reason why the call to `vTaskSetThreadLocalStoragePointer()` fails is because the FreeRTOS scheduler has not started at the point the main task get's created. This results in an undefined behavior for the API call. I would suggest that you make an exception for the `main` task in your wrapper and store the required data in the TLSP region once `app_main` is called.

ammaree, 2024-12-11:
Morning @sudeep-mohanty

Thanks for the feedback, but maybe a couple of things need to be considered...
1. The results are 100% consistent and predictable.
2. The "ipc0" "ipc1" and "esp_timer" time tasks are ALL successfully created through the wrapped function BEFORE the scheduler has been started.
3. The problem 100% consistently shows up with the "main" task and the symptoms of the error are 100% consistent.

Surely the consistent and predictable situation above provides reasonable grounds to investigate why specifically the "main" task fails and not any of the preceding 3 tasks, all 4 being ESP IDF specific tasks.

sudeep-mohanty, 2024-12-12:
Hi @ammaree, Thanks for prompting me further on this issue. I looked in to it further and have a concrete answer for you as to why the `vTaskSetThreadLocalStoragePointer()` crashed for the ` main` task. The reason is that the main task is not created with a [task handle](https://github.com/espressif/esp-idf/blob/a2f420a36ecf8cd19b443adde1d75e5eaecc6309/components/freertos/app_startup.c#L85). As a result, the task handle is passed as `NULL` to `vTaskSetThreadLocalStoragePointer()`. The FreeRTOS kernel behavior in such a case is to pick the _current task_ handle. But since the scheduler has not been started until this point, the current task handle is undefined. This results in a crash when the you tries to access the TLSP region of an undefined TCB.

Now, there could be 2 ways to handle this -
1. As mentioned above, you could store the thread local variable once `app_main` is called to know that the scheduler is in a safe state for the TLSP operations.

2. As a workaround, you could add a task handle during the main task creation [here](https://github.com/espressif/esp-idf/blob/a2f420a36ecf8cd19b443adde1d75e5eaecc6309/components/freertos/app_startup.c#L85).

I would still suggest that it would be better if you find a way to store the TLSP data for all tasks only after the kernel is up and running, to avoid such issues.

ammaree, 2024-12-12:
@sudeep-mohanty 

Apologies, I should have picked up the hint from 'EXCVADDR: 0x00000000' and hunted for a NULL pointer.
Thanks for the 2 suggestions, I have however found a third and preferable (possibly temporary) solution.

```
	TaskHandle_t TempHandle;
	BaseType_t btRV = __real_xTaskCreatePinnedToCore(pxTaskCode, pcName, usStackDepth, pvParameters, uxPriority, &TempHandle, xCoreID);
	vTaskAllocateMask(TempHandle);
	if (pxCreatedTask) *pxCreatedTask = TempHandle;
```

Done this and the issue is solved.
BUT, for consistency alone maybe the change should be made at the location suggested in your option 2.

Thanks for the help.

André

sudeep-mohanty, 2024-12-12:
No worries! :) I'll close the ticket. Please feel free to re-open it or open a new one for further support.

closed 2024-12-12 by sudeep-mohanty