espressif/esp-idf #15006: 5.4 beta fails to install

dannybackx, 2024-12-09:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

5.4beta1 and 5.4beta2

### Operating System used.

Linux

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

None

### What is the expected behavior?

Running ". export.sh" as described fails : 
```
danny@victus:~/src/github/esp32/esp-idf-v5.4-beta2$ ./export.sh
Checking "python3" ...
Python 3.10.12
"python3" has been detected
error: Activation script failed
danny@victus:~/src/github/esp32/esp-idf-v5.4-beta2$ . export.sh
Checking "python3" ...
Python 3.10.12
"python3" has been detected
error: Activation script failed
```

### What is the actual behavior?

see above

### Steps to reproduce.

See above

### Build or installation Logs.

```plain
see above
```

### More Information.

works with all 5.3 versions

dannybackx, 2024-12-09:
Running install.sh from the 5.3.2 directory restored existing environment to working order.
Installation of the beta also killed that.

fhrbata, 2024-12-09:
Hello @dannybackx , can you please try to export `ESP_IDF_EXPORT_DEBUG` before running the `. ./export.sh` and post the output? It may help to see what's going on. 

```bash
export ESP_IDF_EXPORT_DEBUG=1
. ./export.sh
```

Thank you

dannybackx, 2024-12-09:
This doesn't change the output at all.

```
$ . ./export.sh
Could not detect IDF_PATH. Please set it before sourcing this script:
  export IDF_PATH=(add path here)
$ sh ./install.sh
Detecting the Python interpreter
Checking "python3" ...
Python 3.10.12
"python3" has been detected
Checking Python compatibility
Installing ESP-IDF tools
Updating /home/danny/.espressif/idf-env.json
Selected targets are: esp32c2, esp32c61, esp32c6, esp32c5, esp32p4, esp32s2, esp32h2, esp32s3, esp32, esp32c3
Current system platform: linux-amd64
Installing tools: xtensa-esp-elf-gdb, riscv32-esp-elf-gdb, xtensa-esp-elf, riscv32-esp-elf, esp32ulp-elf, openocd-esp32, esp-rom-elfs
Skipping xtensa-esp-elf-gdb@14.2_20240403 (already installed)
Skipping riscv32-esp-elf-gdb@14.2_20240403 (already installed)
Skipping xtensa-esp-elf@esp-14.2.0_20240906 (already installed)
Skipping riscv32-esp-elf@esp-14.2.0_20240906 (already installed)
Skipping esp32ulp-elf@2.38_20240113 (already installed)
Skipping openocd-esp32@v0.12.0-esp32-20241016 (already installed)
Skipping esp-rom-elfs@20241011 (already installed)
Installing Python environment and packages
Python 3.10.12
pip 24.3.1 from /home/danny/.espressif/python_env/idf5.0_py3.10_env/lib/python3.10/site-packages/pip (python 3.10)
WARNING: The following issue occurred while accessing the ESP-IDF version file in the Python environment: [Errno 2] No such file or directory: '/home/danny/.espressif/python_env/idf5.0_py3.10_env/idf_version.txt'. (Diagnostic information. It can be ignored.)
Skipping the download of /home/danny/.espressif/espidf.constraints.v5.4.txt because it was downloaded recently.
Upgrading pip and setuptools...
Requirement already satisfied: pip in /home/danny/.espressif/python_env/idf5.0_py3.10_env/lib/python3.10/site-packages (24.3.1)
Requirement already satisfied: setuptools in /home/danny/.espressif/python_env/idf5.0_py3.10_env/lib/python3.10/site-packages (71.0.0)
Installing Python packages
 Constraint file: /home/danny/.espressif/espidf.constraints.v5.4.txt
 Requirement files:
  - /home/danny/src/github/esp32/esp-idf-v5.4-beta2/tools/requirements/requirements.core.txt
Looking in indexes: https://pypi.org/simple, https://dl.espressif.com/pypi
Ignoring importlib_metadata: markers 'python_version < "3.8"' don't match your environment
Requirement already satisfied: setuptools in /home/danny/.espressif/python_env/idf5.0_py3.10_env/lib/python3.10/site-packages (from -r /home/danny/src/github/esp32/esp-idf-v5.4-beta2/tools/requirements/requirements.core.txt (line 7)) (71.0.0)
Requirement already satisfied: packaging in /home/danny/.espressif/python_env/idf5.0_py3.10_env/lib/python3.10/site-packages (from -r /home/danny/src/github/esp32/esp-idf-v5.4-beta2/tools/requirements/requirements.core.txt (line 8)) (24.2)
Requirement already satisfied: click in /home/danny/.espressif/python_env/idf5.0_py3.10_env/lib/python3.10/site-packages (from -r /home/danny/src/github/esp32/esp-idf-v5.4-beta2/tools/requirements/requirements.core.txt (line 11)) (8.1.7)
Requirement already satisfied: pyserial in /home/danny/.espressif/python_env/idf5.0_py3.10_env/lib/python3.10/site-packages (from -r /home/danny/src/github/esp32/esp-idf-v5.4-beta2/tools/requirements/requirements.core.txt (line 12)) (3.5)
Requirement already satisfied: cryptography in /home/danny/.espressif/python_env/idf5.0_py3.10_env/lib/python3.10/site-packages (from -r /home/danny/src/github/esp32/esp-idf-v5.4-beta2/tools/requirements/requirements.core.txt (line 13)) (43.0.3)
Requirement already satisfied: pyparsing in /home/danny/.espressif/python_env/idf5.0_py3.10_env/lib/python3.10/site-packages (from -r /home/danny/src/github/esp32/esp-idf-v5.4-beta2/tools/requirements/requirements.core.txt (line 14)) (3.2.0)
Requirement already satisfied: pyelftools in /home/danny/.espressif/python_env/idf5.0_py3.10_env/lib/python3.10/site-packages (from -r /home/danny/src/github/esp32/esp-idf-v5.4-beta2/tools/requirements/requirements.core.txt (line 15)) (0.31)
Requirement already satisfied: idf-component-manager in /home/danny/.espressif/python_env/idf5.0_py3.10_env/lib/python3.10/site-packages (from -r /home/danny/src/github/esp32/esp-idf-v5.4-beta2/tools/requirements/requirements.core.txt (line 16)) (2.1.1)
Requirement already satisfied: esp-coredump in /home/danny/.espressif/python_env/idf5.0_py3.10_env/lib/python3.10/site-packages (from -r /home/danny/src/github/esp32/esp-idf-v5.4-beta2/tools/requirements/requirements.core.txt (line 17)) (1.12.0)
Requirement already satisfied: esptool in /home/danny/.espressif/python_env/idf5.0_py3.10_env/lib/python3.10/site-packages (from -r /home/danny/src/github/esp32/esp-idf-v5.4-beta2/tools/requirements/requirements.core.txt (line 18)) (4.8.1)
Requirement already satisfied: esp-idf-kconfig in /home/danny/.espressif/python_env/idf5.0_py3.10_env/lib/python3.10/site-packages (from -r /home/danny/src/github/esp32/esp-idf-v5.4-beta2/tools/requirements/requirements.core.txt (line 19)) (2.3.0)
Requirement already satisfied: esp-idf-monitor in /home/danny/.espressif/python_env/idf5.0_py3.10_env/lib/python3.10/site-packages (from -r /home/danny/src/github/esp32/esp-idf-v5.4-beta2/tools/requirements/requirements.core.txt (line 20)) (1.5.0)
Requirement already satisfied: esp-idf-nvs-partition-gen in /home/danny/.espressif/python_env/idf5.0_py3.10_env/lib/python3.10/site-packages (from -r /home/danny/src/github/esp32/esp-idf-v5.4-beta2/tools/requirements/requirements.core.txt (line 21)) (0.1.6)
[... cut ...]
Requirement already satisfied: mdurl~=0.1 in /home/danny/.espressif/python_env/idf5.0_py3.10_env/lib/python3.10/site-packages (from markdown-it-py>=2.2.0->rich->esp-idf-size->-r /home/danny/src/github/esp32/esp-idf-v5.3.2/tools/requirements/requirements.core.txt (line 18)) (0.1.2)
Installing collected packages: cryptography
  Attempting uninstall: cryptography
    Found existing installation: cryptography 43.0.3
    Uninstalling cryptography-43.0.3:
      Successfully uninstalled cryptography-43.0.3
Successfully installed cryptography-42.0.8
All done! You can now run:

  . ./export.sh
```

fhrbata, 2024-12-09:
Hello @dannybackx,

thank you very much for the extra details. It seems you're executing the `install.sh` script for v5.4 while "v5.0 or v5.3" is already activated. This can be seen in the log you posted, which shows packages and python virtual environment being set up in `/home/danny/.espressif/python_env/idf5.0_py3.10_env/`, even for the v5.3 installation. I recommend starting the installation in a new terminal session without any  ESP-IDF environment active. I believe it should help to solve the problem.

The issue arises because the export(activate.py) script is looking for the python interpreter in the `.espressif/python_env/idf5.4_py3.12_env/bin/python` directory not honoring the already set `IDF_PYTHON_ENV_PATH` environmental variable. This might be a bug that we need to look into.

Thank you

dannybackx, 2024-12-10:
That seems to work.
Note that for me this requirement appears new, my installations used to work like this (with an existing IDF environment in place).

closed 2024-12-20 by espressif-bot