espressif/esp-idf #15008: 两台电脑ESP-IDF编译出来的程序bin文件不一样。

wangutian, 2024-12-10:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

两台电脑上的ESP-IDF版本一样，都是5.2，是用同一个离线下载器安装的。我已经把sdkconfig中编译时的时间戳去掉了，经过对比软件比较，两台电脑编译出来的程序bin文件有两处，分别在bin文件的开头和结尾。我还需要什么操作，才能让两台电脑编译出来的程序bin文件一样？

wangutian, 2024-12-10:
使用的是ESP32C3

dobairoland, 2024-12-10:
Hi @wangutian. You can read about this here: https://docs.espressif.com/projects/esp-idf/zh_CN/release-v5.2/esp32/api-guides/reproducible-builds.html Please try to enable reproducible build based on that page.

wangutian, 2024-12-10:
你好@dobairoland

> 您好。您可以在此处阅读相关内容：https://docs.espressif.com/projects/esp-idf/zh_CN/release-v5.2/esp32/api-guides/reproducible-builds.html 请尝试基于该页面启用可重现的构建。

您好，我试过了在ESP-IDF中将sdkconfig把可重现构建勾选上了，但是两台电脑编译出来的还是不一样，并且每一次都是bin文件的下边这个位置不同：
[image]

suda-morris, 2024-12-11:
是不是因为两台电脑上的toolchain路径不同？ @wangutian  这个问题应该已经在master上被修复了: 

https://github.com/espressif/esp-idf/commit/bef4cb05ff1cf8439b501ce99db75b468d31c273

dobairoland, 2024-12-11:
The diff of the BINs doesn't tell us too much. But you could diff the MAP files placed into the same directory with the BINs to see the difference. This could confirm the assumption of suda-morris.

closed 2025-01-14 by dobairoland