espressif/esp-idf #15009: esp32s2 USB CDC not working (without tinyusb)

jjoakim, 2024-12-10:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

I tried to make a new project using a esp32s2. I enabled USB CDC but it is not working. In my old project from two years ago, as far as I remember, this was working without problems.

- I copied the hello word example.
- Configured it for esp32s2.
- Since my esp32s2 only uses usb for console output I went ahead and enabled in menuconfig:
(Top) → Component config → ESP System Settings → Channel for console output → USB CDC

When I reboot the device it's no longer available to the system and thus I cannon use idf.py monitor. 

dmesg output gives me the following recurring messages.

> [32198.950492] usb 3-3: new full-speed USB device number 30 using xhci_hcd
> [32210.279669] usb 3-3: Device not responding to setup address.
> [32210.482356] usb 3-3: Device not responding to setup address.
> [32210.690001] usb 3-3: device not accepting address 30, error -71
> [32210.690042] usb 3-3: WARN: invalid context state for evaluate context command.
> [32210.690083] usb usb3-port3: attempt power cycle
> [32211.070353] usb 3-3: new full-speed USB device number 31 using xhci_hcd
> [32216.498284] xhci_hcd 0000:0c:00.3: Timeout while waiting for setup device command
> [32216.705943] usb 3-3: Device not responding to setup address.
> [32216.914262] usb 3-3: device not accepting address 31, error -71

I found a workaround for this problem by adding tinyusb to the project.

- idf.py add-dependency esp_tinyusb

I want to preserve as much battery power as possible so I am also curious if enabling tinyusb effects power consumption.

nopnop2002, 2024-12-11:
If you are using this, you can enable /dev/ttyACM0 by following the steps below.
[image]

Since it uses the chip's USB-OTG function, it will not be recognized as a USB device even if you simply connect it to the host with a USB cable.   
To write firmware, press ButtonRST while pressing Button0, then release ButtonRST and then Button0.   
Now the device will enter DFU mode, the USB device (/dev/ttyACM0) will appear, and you can upload the firmware.   
Once the firmware upload is complete, press ButtonRST again.   
The firmware will now start.   
If you press ButtonRST without pressing Button0, the USB device will disappear.   

To use STDOUT, you need to change Channel for console output to USB CDC in menuconfig.   
[image]

After changing Channel for console output to USB CDC in menuconfig, upload the firmware again.   
Once the firmware has been uploaded, the following display will appear.   
[image]

__Now press ButtonRST.__   
The firmware with USB CDC enabled will now start running.   
From now on, you will be able to write and monitor the firmware without pressing Button0 or ButtonRST.

jjoakim, 2024-12-11:
I did indeed follow this exact procedure and I can only reach as far as this point.

> The firmware with USB CDC enabled will now start running.
> From now on, you will be able to write and monitor the firmware without pressing Button0 or ButtonRST.

Since the system cannot recognize it as a proper usb device, as seen from the output of the dmesg command in my initial post, idf.py monitor will never work nor will flashing the device without entering DFU mode.

My device is similar to a Espressif ESP32-Saola-1 and I know USB CDC used to work previously with a version 4 of the sdk from a couple of years ago.

peter-marcisovsky, 2025-01-08:
Hi @jjoakim sorry for a late reply.
I managed to reproduce the issue you are facing.
Can I ask you, which IDF version are you using?

jjoakim, 2025-01-09:
Sure, I was using this version:

commit a2f420a36ecf8cd19b443adde1d75e5eaecc6309

peter-marcisovsky, 2025-01-10:
Perfect, thx for the info. 
I found out what is happening. As a workaround, please try to use some older IDF releases, as the "bug" is not present there.

fhrbata, 2025-01-10:
Hello @jjoakim ,

this is most probably related to the `minimal build` functionality recently introduced on the `master` branch. The commit  a2f420a has it enabled for the `hello_world` example. I think removing or commenting out the `idf_build_set_property(MINIMAL_BUILD ON)` line in the project's `CMakeLists.txt` should help. Or you can add the `esp_vfs_console` component to `PRIV_REQUIRES` dependencies in `main/CMakeLists.txt`. IIUC the USB CDC console is implemented through the `vfs` component(`esp_vfs_console` requires `vfs`), which is not added to the build for `hello_world` if `minimal build` is on. I guess maybe the USB CDC console should depends on VFS_SUPPORT_IO in kconfig, but this will need more investigation.

OT minimal build: If not enabled, the build system includes all discovered components in the build, leading to longer build times as many unused components are compiled unnecessarily. When the minimal build is enabled, only components explicitly defined as dependencies of the main component, along with their transitive dependencies, are included in the build.

Thank you very much for bringing this issue to our attention.