espressif/esp-idf #15011: Multiple I2c Addresses on same bus

panilsunil, 2024-12-10:
### Is your feature request related to a problem?

I am using ESP32-S3-Touch-LCD-4.3B. This device have 26 Address on the bus, which needs to add each and every address using i2c_master_bus_add_device. Which sums up 26 (i2c_master_dev_handle_t) Devs handles. Ref: https://www.waveshare.com/wiki/ESP32-S3-Touch-LCD-4.3B

### Describe the solution you'd like.

By adding these tow functions in i2c_master.c we can change the device address on the fly.
uint16_t i2c_master_get_device_address(i2c_master_dev_handle_t i2c_dev)
{
	ESP_RETURN_ON_FALSE(i2c_dev != NULL, ESP_ERR_INVALID_ARG, TAG, "i2c handle not initialized");
	return i2c_dev->device_address;
}

esp_err_t i2c_master_set_device_address(i2c_master_dev_handle_t i2c_dev, uint16_t device_address)
{
	ESP_RETURN_ON_FALSE(i2c_dev != NULL, ESP_ERR_INVALID_ARG, TAG, "i2c handle not initialized");
	i2c_dev->device_address = device_address;
	return ESP_OK;
}

### Describe alternatives you've considered.

If we can change the address of the (i2c_master_dev_handle_t) dynamically, then the task can be achieved using a small number of (i2c_master_dev_handle_t) dev Handles.,

### Additional context.

[image]

nopnop2002, 2024-12-10:
Legacy's i2c driver allows us to change the device address on the fly.
```
	i2c_cmd_handle_t cmd = i2c_cmd_link_create();

	i2c_master_start(cmd);
	i2c_master_write_byte(cmd, (dev->_address << 1) | I2C_MASTER_WRITE, true);
	i2c_master_stop(cmd);

	esp_err_t res = i2c_master_cmd_begin(dev->_i2c_num, cmd, I2C_TICKS_TO_WAIT);
	if (res == ESP_OK) {
		ESP_LOGI(TAG, "OLED configured successfully");
	} else {
		ESP_LOGE(TAG, "OLED configuration failed. code: 0x%.2X", res);
	}
	i2c_cmd_link_delete(cmd);
```

suda-morris, 2024-12-11:
The new i2c master driver give the "transmit" function to the "i2c device" instead of the "i2c bus".  This is by design. Because we can't guarantee all the devices will share the same configurations like "SCL frequency", "SCL wait time", "10bit address or not". These parameters belong to the I2c Device not the I2C bus.

But by adding a `i2c_master_device_set_address` may still be useful if e.g. I have 4 temperature they're the same just with different hw address. Then I only have to create one handle shared by all these 4 sensors :thinking: @mythbuster5

closed 2025-01-09 by espressif-bot