espressif/esp-idf #15012: i2c Conflict! i2c冲突

likestones, 2024-12-10:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.3.2

### Operating System used.

Windows

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

CMD

### What is the expected behavior?

He was supposed to burn and run normally

### What is the actual behavior?

Unlimited restart after burning and display through monitor says i2c:COnFLIcT! driver ng is not allowed to be used with this old driver□

### Steps to reproduce.

1.idf.py build
2.idf.py flash
3.idf.py monitor

### Build or installation Logs.

```plain
i2c:COFLIcT! driver ng is not allowed to be used with this old driver□[0m
12:56:25.773]
bort()was calledatP0x420ca4abon core 0
Backtrace: 0x40379eae:0x3foeb100 0x40383e49:0x3foeb1e0 0x4038bab5:0x3fceb200 0x420ea4ab:0x3fceb270 0x420038e3.0x
0x4200395a:0x3fceb2b0 0x40379a4a.0x3fceb2d0 0x403c0941:0x3fcb340 0x403c0037:0x3fceb380 0x40308959:0x3foeb4b0
0x40045001:0x3fceb570 0x40043ab6:0x3fceb6f0 0x40034045:0x3fceb710
```

suda-morris, 2024-12-11:
This error message could happen when you use two components, where component A is constructed based on the legacy i2c driver and component B is constructed based on the new i2c driver. In general, we don't expect the user to use both of them in the same application.

ammaree, 2024-12-11:
@suda-morris 
@mythbuster5 

I am having the same problem but with 1 specific device model only, the ESP32-S#-LCD-EV2 board using the latest BSP for this device.

As far as I can recall the BSP (used to?) use some mechanism for auto detecting the I2C driver/module being used. Is it possible that both drivers somehow get linked in here?

likestones, 2024-12-11:
Oh, I found it. I used an audio codec device. You need to close __idf.Py menuconifg_ ->Component config ->Audio Code Device Configuration_ ->**Enable backward compatibility for the I2c driver (force use of the old I2c_driver above v5.3)**

mythbuster5, 2024-12-16:
If no other topic, I will close this soon.

mythbuster5, 2024-12-31:
Seems there is no more update, I'm going to close it. Feel free to reopen if there is any follow up. Thanks!

closed 2024-12-31 by mythbuster5