espressif/esp-idf #15013: 5.4beta2 vs 5.3.2 difference

dannybackx, 2024-12-10:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

5.4 beta 2 compared to 5.3.2

### Espressif SoC revision.

Warning: ESP32-S3 has no Chip ID. Reading MAC instead. MAC: 48:27:e2:e7:4a:88

### Operating System used.

Linux

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

None

### Development Kit.

ZX3D95CD01S

### Power Supply used.

External 5V

### What is the expected behavior?

I expected the image to be the same

### What is the actual behavior?

Visual output shifted

### Steps to reproduce.

https://sourceforge.net/p/esp32-s3-86-box-project-base/code/HEAD/tree/trunk/alarm

Compile with the two versions mentioned.
[image]
[image]

Lzw655, 2024-12-19:
Hi @dannybackx, 

I'm sorry you encountered this issue and not sure what changes ESP-IDF 5.4 beta has compared to 5.3.2. But this is an RGB LCD [screen drift issue](https://docs.espressif.com/projects/esp-faq/en/latest/software-framework/peripherals/lcd.html#why-do-i-get-drift-overall-drift-of-the-display-when-esp32-s3-is-driving-an-rgb-lcd-screen) that occurs with a high probability when using WIFI or erasing the flash. I reviewed the source code you provided, and the issue can be resolved with the following modifications:
* Add new configuration options in sdkconfig to improve system performance:
  * CONFIG_ESP32S3_DATA_CACHE_LINE_64B=y
  * CONFIG_COMPILER_OPTIMIZATION_PERF=y
  * CONFIG_FREERTOS_HZ=1000
* Set `.bounce_buffer_size_px = 10 * 480,` in the GC9503 LCD initialization configuration `esp_lcd_rgb_panel_config_t` (if the issue persists, try increasing it to `20 * 480`).

dannybackx, 2024-12-21:
Thanks but these don't help

Lzw655, 2024-12-23:
Sorry, I missed something. Please also add the `CONFIG_SPIRAM_XIP_FROM_PSRAM=y` configuration.

dannybackx, 2024-12-23:
Please note that using the 5.4 (beta1, beta2, rc1) trees also causes a shift in a demo application (on the same hardware) that uses a completely different video software stack.
The one I referenced earlier uses esp-idf components for hardware drivers.

This one - see https://sourceforge.net/p/esp32-s3-86-box-project-base/code/HEAD/tree/trunk/minimal/ - uses the software stack provided by the hardware manufacturer. Same behaviour : with 5.3 the UI looks normal, with any of the 5.4 distributions, the screen is shifted.

ckirmse, 2025-01-13:
I am running into the same situation as dannybackx and can confirm it's there even in very simple apps--something important has changed in the esp-idf 5.4 release for rgb displays and so far I have not been able to work around it in any way.

dannybackx, 2025-01-20:
Same thing on a Spotpear module.

closed 2025-02-13 by espressif-bot

Jason2866, 2025-02-13:
@suda-morris The bug is not fixed. Branch master had this shift before as in branch IDF 5.4  Now the change is merged in branch master. Display in branch master shows an other bug now.
When enabling the option `CONFIG_LCD_RGB_RESTART_IN_VSYNC=y` is needed the display goes crazy fast moving left and right from the shift position to the correct position.
Totally unusable.

dannybackx, 2025-02-14:
The fix works for my primary targets, thanks.

dannybackx, 2025-02-14:
The fix wWorks for my primary targets, thanks.

Jason2866, 2025-02-14:
@dannybackx It does not work anymore when the option `CONFIG_LCD_RGB_RESTART_IN_VSYNC=y` is needed
Screen shifts crazy fast all the time