espressif/esp-idf #15015: esp_lcd set contrast function

mutatrum, 2024-12-10:
### Is your feature request related to a problem?

The SSD1306 and SH1106 displays have a contrast command (0x81), which is currently not exposed through esp_lcd_panel_ops. Other panels probably have similar contrast settings.

### Describe the solution you'd like.

Addition of a `esp_lcd_panel_contrast` function to set the contrast.

### Describe alternatives you've considered.

Directly executing the command:
```
#define SSD1306_CMD_CONTRAST 0x81
ESP_ERROR_CHECK_WITHOUT_ABORT(esp_lcd_panel_io_tx_param(io_handle, SSD1306_CMD_CONTRAST, (uint8_t[]) { 0x80 }, 1));
```
which is not very nice.

suda-morris, 2024-12-11:
Thanks @mutatrum for the suggestion, we may consider this feature.

WantClue, 2026-08-16:
@suda-morris are there any updates on this?