espressif/esp-idf #15016: 如何在同一个地址即实现http协议的数据请求又实现websocket的数据交互

cannnnnnnnnnnn, 2024-12-11:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

软件版本V5.3.1
我做出了一下修改

```c
    httpd_uri_t init_server = {
        .uri = "/web", // 192.168.0.1
        .method = HTTP_GET,
        .handler = http_server_handler,
        .user_ctx = NULL,
        .is_websocket = true};
    httpd_register_uri_handler(server, &init_server);
```

```c
static esp_err_t http_server_handler(httpd_req_t *req)
{
    if (req->method == HTTP_GET) {
        httpd_resp_send(req, (char *)WEB_HTML, HTTP_WEBPAGE_SIZE);
    }

    // 以下位websocket的数据解析
    ........
    return ESP_OK;
}
```

在正常的http数据请求是没有问题html页面可以正常请求到数据，但是在websocket总是链接失败数据无法正确回复，接收数据参考**ws_echo_server**

https://github.com/espressif/esp-idf/blob/c8fc5f643b7a7b0d3b182d3df610844e3dc9bd74/examples/protocols/http_server/ws_echo_server/main/ws_echo_server.c#L85

https://github.com/espressif/esp-idf/blob/c8fc5f643b7a7b0d3b182d3df610844e3dc9bd74/examples/protocols/http_server/ws_echo_server/main/ws_echo_server.c#L115

```
I (54926) http_server: frame len is 2
I (54926) http_server: Got packet with message: ��
I (54926) http_server: Packet type: 0
W (54926) httpd_txrx: httpd_sock_err: error in send : 104
W (54936) httpd_ws: httpd_ws_send_frame_async: Failed to send WS header
E (54946) http_server: httpd_ws_send_frame failed with -1
W (54946) httpd_uri: httpd_uri: uri handler execution failed
W (59836) httpd_txrx: httpd_sock_err: error in send : 104
```

cannnnnnnnnnnn, 2024-12-11:
[image]

nileshkale123, 2025-01-03:
Hello @cannnnnnnnnnnn ,

Does it work only with WebSocket?

Could you please provide detailed steps to reproduce the issue? Alternatively, sharing a small integrable patch to replicate the problem would be highly helpful so that we can investigate it further.