espressif/esp-idf #15019: ESP model occur Image hash failed when  after soft reboot

luacerbb, 2024-12-11:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

IDF_V5.1.4

### Espressif SoC revision.

ESP8684-WROOM-02C

### Operating System used.

Windows

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

None

### Development Kit.

custom board

### Power Supply used.

Battery

### What is the expected behavior?

Dev can running normally

### What is the actual behavior?

The device displays an abnormal prompt after performing a software restart, and then restarts repeatedly to prompt the same content until it is manually powered on。
[image]

### Steps to reproduce.

1. The device performs a long press button to trigger the software to restart
2. after restart , Device log prompt image hash failed-image is corrupt,Then reboot repeatedly
3. The device can only return to normal after powering off 
...

SoucheSouche, 2024-12-12:
Hi @luacerbb, Thanks for reporting the issue, we will take a look at it. Could you be more specific on how to reproduce the issue? I tried doing a software reset on the esp32c2 but I can't seem to reproduce your issue.

SoucheSouche, 2025-03-06:
@luacerbb, I am closing this ticket since you did not respond for 2 months.

closed 2025-03-06 by SoucheSouche