espressif/esp-idf #15021: esp hid device example won't work using mobile phone

Sagigamil, 2024-12-12:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

Hi all, 
I compiled and ran esp_hid_device_main.c example on esp32-c3. 
I managed to connect to the esp using my laptop, but nothing happen when I try to connect using my phone.
I also enabled the debug for all the  stack layers and yet, when I hit the device on the devices list on my phone (I own s23 ultra), nothing happens. Maybe my phone refuses to connect to the esp since it's unothorized vendor or something similar? 
Any idea?

esp-zhp, 2024-12-17:
@Sagigamil 
The version you’re using—I’ve fixed some HID-related issues in it, but are you sure your version includes these fixes?  

Here are some relevant commits (there may be more):  
- f7e174e9487cc218b92cffee9dc5953e962e0ff4  
- 5727802497b0504167216932af6d08a392845c6  

I recommend upgrading to the latest version to check if the issue still persists.

esp-zhp, 2024-12-17:
It should work on your phone.

esp-zhp, 2024-12-17:
We also have another HID example that you can refer to:  
`esp-idf/examples/bluetooth/bluedroid/ble/ble_hid_device_demo`.  

If you have any further questions, please let me know. I look forward to your feedback!

esp-zhp, 2024-12-24:
@Sagigamil any progress?

closed 2025-03-28 by Alvin1Zhang