espressif/esp-idf #15026: ESP-IDF v5.3.1 crashes during sendto() in pthread_local_storage

KaDw, 2024-12-13:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.3.1

### Espressif SoC revision.

ESP32-S3

### Operating System used.

Windows

### How did you build your project?

VS Code IDE

### If you are using Windows, please specify command line type.

CMD

### Development Kit.

Custom Board

### Power Supply used.

External 3.3V

### What is the expected behavior?

Send data over UDP to the server. 

### What is the actual behavior?

Crash during sendto()

### Steps to reproduce.

1. Create socket using socket()
2. Send UDP packet using sendto()
3. Wait for response using recv()
4. Start to transmit data using audio_element write callback function, sendto() is used with the same socket. Note that this is done in another thread than 2.

### Debug Logs.

```plain
[0;32mI (13254) ESIGuru Meditation Error: Core  0 panic'ed (LoadProhibited). Exception was unhandled.

Core  0 register dump:
PC      : 0x421a0f03  PS      : 0x00060230  A0      : 0x8200b6e4  A1      : 0x3c532fa0  
0x421a0f03: find_value at C:/Espressif/frameworks/esp-idf-v5.3.1/components/pthread/pthread_local_storage.c:175
A2      : 0xfffffff0  A3      : 0x00000003  A4      : 0x00060420  A5      : 0x00000000  
A6      : 0x0000004a  A7      : 0x3c532db0  A8      : 0x8038a033  A9      : 0x3c532f80  
A10     : 0x3fcee2fc  A11     : 0x000000b4  A12     : 0x00001400  A13     : 0x00060423  
A14     : 0x00000000  A15     : 0x3fcbeed8  SAR     : 0x0000001f  EXCCAUSE: 0x0000001c  
EXCVADDR: 0xfffffff0  LBEG    : 0x40056f5c  LEND    : 0x40056f72  LCOUNT  : 0x00000000  
0x40056f5c: memcpy in ROM
0x40056f72: memcpy in ROM

Backtrace: 0x421a0f00:0x3c532fa0 0x4200b6e1:0x3c532fc0 0x421bdb78:0x3c532fe0 0x4038ea1e:0x3c533000 0x4038ecad:0x3c533020 0x403908d2:0x3c533070 0x420461f6:0x3c5330d0 0x4204bc49:0x3c5330f0 0x420446f2:0x3c533120 0x4204c0ee:0x3c533140 0x4204c2b2:0x3c533170 0x40387a01:0x3c5331a0
0x421a0f00: find_value at C:/Espressif/frameworks/esp-idf-v5.3.1/components/pthread/pthread_local_storage.c:173
0x4200b6e1: pthread_getspecific at C:/Espressif/frameworks/esp-idf-v5.3.1/components/pthread/pthread_local_storage.c:190
0x421bdb78: sys_thread_sem_get at C:/Espressif/frameworks/esp-idf-v5.3.1/components/lwip/port/freertos/sys_arch.c:492
0x4038ea1e: netconn_apimsg at C:/Espressif/frameworks/esp-idf-v5.3.1/components/lwip/lwip/src/api/api_lib.c:128
0x4038ecad: netconn_send at C:/Espressif/frameworks/esp-idf-v5.3.1/components/lwip/lwip/src/api/api_lib.c:953
0x403908d2: lwip_sendto at C:/Espressif/frameworks/esp-idf-v5.3.1/components/lwip/lwip/src/api/sockets.c:1679
0x420461f6: sendto at C:/Espressif/frameworks/esp-idf-v5.3.1/components/lwip/include/lwip/sockets.h:46
 (inlined by) voip_sink_write at C:/proj/main/components/board/libs/my_sip/my_sip.c:330
0x4204bc49: audio_element_output at C:/Users/admin/.espressif/esp-adf/components/audio_pipeline/audio_element.c:426
0x420446f2: voip_sink_process at C:/proj/main/components/board/libs/my_audio/my_audio_pipe.c:360
0x4204c0ee: audio_element_process_running at C:/Users/admin/.espressif/esp-adf/components/audio_pipeline/audio_element.c:340
0x4204c2b2: audio_element_task at C:/Users/admin/.espressif/esp-adf/components/audio_pipeline/audio_element.c:487
0x40387a01: vPortTaskWrapper at C:/Espressif/frameworks/esp-idf-v5.3.1/components/freertos/FreeRTOS-Kernel/portable/xtensa/port.c:134
```

### More Information.

It worked wihtout any problems on IDF v5.1.2. I tried to find a difference in LWIP config that could cause this sort of problem but I didn't manage to.

Someone reported a similar problem here: https://github.com/espressif/esp-idf/issues/5482, I'm not using FreeRTOS TLSP in any of my application files but I wonder if the ADF pipeline_element is using them in some way .
I also tried to increase the CONFIG_FREERTOS_THREAD_LOCAL_STORAGE_POINTERS from 1 to 4, no luck.

Here are relevant SDK settings:

```
#
# Default configuration
# Espressif IoT Development Framework (ESP-IDF) Project Configuration
#

CONFIG_IDF_CMAKE=y
CONFIG_IDF_TARGET_ARCH_XTENSA=y
CONFIG_IDF_TARGET="esp32s3"
CONFIG_IDF_TARGET_ESP32S3=y
CONFIG_IDF_FIRMWARE_CHIP_ID=0x0009

#
# SDK tool configuration
#
CONFIG_SDK_TOOLPREFIX="xtensa-esp32s3-elf-"
# end of SDK tool configuration

#
# Audio HAL
#
CONFIG_AUDIO_BOARD_CUSTOM=y
# end of Audio HAL

#
# ADF Features
#
CONFIG_ESP_DISPATCHER_DELEGATE_TASK_CORE=1
CONFIG_ESP_DISPATCHER_DELEGATE_TASK_PRIO=10
CONFIG_ESP_DISPATCHER_DELEGATE_STACK_SIZE=4096
# end of ADF Features

#
# ESP32S3-Specific
#
CONFIG_ESP_DEFAULT_CPU_FREQ_MHZ_240=y
CONFIG_ESP_DEFAULT_CPU_FREQ_MHZ=240
CONFIG_SPIRAM=y

#
# Wi-Fi
#
CONFIG_ESP_WIFI_STATIC_RX_BUFFER_NUM=8
CONFIG_ESP_WIFI_DYNAMIC_RX_BUFFER_NUM=32
CONFIG_ESP_WIFI_CACHE_TX_BUFFER_NUM=32
CONFIG_ESP_WIFI_STATIC_TX_BUFFER_NUM=8
CONFIG_ESP_WIFI_TX_BA_WIN=15
CONFIG_ESP_WIFI_RX_BA_WIN=16
CONFIG_ESP_WIFI_MGMT_SBUF_NUM=15
CONFIG_ESP_WIFI_TASK_PINNED_TO_CORE_1=y
# CONFIG_ESP_WIFI_RX_IRAM_OPT is not set
# end of Wi-Fi

#
# LWIP
#
CONFIG_LWIP_LOCAL_HOSTNAME="myproj"
CONFIG_LWIP_IRAM_OPTIMIZATION=y
CONFIG_LWIP_TCPIP_RECVMBOX_SIZE=64

#
# TCP
#
CONFIG_LWIP_TCP_SND_BUF_DEFAULT=65535
CONFIG_LWIP_TCP_WND_DEFAULT=65535
CONFIG_LWIP_TCP_RECVMBOX_SIZE=64
CONFIG_LWIP_TCP_SACK_OUT=y
CONFIG_LWIP_TCP_RTO_TIME=2500
# end of TCP

#
# UDP
#
CONFIG_LWIP_UDP_RECVMBOX_SIZE=64
# end of UDP

CONFIG_LWIP_PPP_SUPPORT=y
CONFIG_LWIP_PPP_ENABLE_IPV6=y
# end of LWIP

#
# FreeRTOS
#
# CONFIG_FREERTOS_GENERATE_RUN_TIME_STATS is not set
# CONFIG_FREERTOS_VTASKLIST_INCLUDE_COREID is not set
CONFIG_FREERTOS_ENABLE_BACKWARD_COMPATIBILITY=y
CONFIG_FREERTOS_THREAD_LOCAL_STORAGE_POINTERS=4
# End of FreeRTOS

#
# Core dump
#
CONFIG_ESP_COREDUMP_ENABLE_TO_FLASH=y
# CONFIG_ESP_COREDUMP_ENABLE_TO_UART is not set
# CONFIG_ESP_COREDUMP_ENABLE_TO_NONE is not set
# CONFIG_ESP_COREDUMP_DATA_FORMAT_BIN is not set
CONFIG_ESP_COREDUMP_DATA_FORMAT_ELF=y
CONFIG_ESP_COREDUMP_CHECKSUM_CRC32=y
# CONFIG_ESP_COREDUMP_CHECK_BOOT is not set
CONFIG_ESP_COREDUMP_ENABLE=y
# CONFIG_ESP_COREDUMP_LOGS is not set
CONFIG_ESP_COREDUMP_MAX_TASKS_NUM=34
CONFIG_ESP_COREDUMP_USE_STACK_SIZE=y
CONFIG_ESP_COREDUMP_STACK_SIZE=1280
# end of Core dump

#
# Ultra Low Power (ULP) Co-processor
#
CONFIG_ULP_COPROC_ENABLED=y
# CONFIG_ULP_COPROC_TYPE_FSM is not set
CONFIG_ULP_COPROC_TYPE_RISCV=y
CONFIG_ULP_COPROC_RESERVE_MEM=8100

# Deprecated options for backward compatibility
CONFIG_MAIN_TASK_STACK_SIZE=8192
CONFIG_TASK_WDT_PANIC=y
CONFIG_TCPIP_RECVMBOX_SIZE=64
CONFIG_TCP_SND_BUF_DEFAULT=65535
CONFIG_TCP_WND_DEFAULT=65535
CONFIG_TCP_RECVMBOX_SIZE=64
CONFIG_UDP_RECVMBOX_SIZE=64
CONFIG_PPP_SUPPORT=y
# End of deprecated options
```

igrr, 2024-12-13:
Just a debugging suggestion, could you please try enabling CONFIG_FREERTOS_WATCHPOINT_END_OF_STACK to check that this isn't caused by a stack overflow in `audio_element_task` task?

KaDw, 2024-12-13:
@igrr I think I found the problem... I'm also using ESP-ADF and I didn't apply the correct patch, instead of [idf_v5.3_freertos.patch](https://github.com/espressif/esp-adf/blob/master/idf_patches/idf_v5.3_freertos.patch) I still ran with [idf_v5.1_freertos.patch](https://github.com/espressif/esp-adf/blob/master/idf_patches/idf_v5.1_freertos.patch). 

There is substantial difference since the 5.3 patch puts the `xTaskCreateRestrictedPinnedToCore` in configSUPPORT_STATIC_ALLOCATION section, and 5.1 in configSUPPORT_DYNAMIC_ALLOCATION

closed 2024-12-13 by KaDw