espressif/esp-idf #15028: ECM Driver For ESP32-S3

ParthaSarathiMishra, 2024-12-13:
### Is your feature request related to a problem?

Presently ESP32-S3 USB driver uses CDC ACM driver for acting as a host. It works as a virtual COM port. But issue is its very slow when doing network operations(like FTP upload) using a modem (like Quectel BG77, etc) over USB. The same modem when connected to PC with ECM driver makes the operation around 20 times faster. The AT command layer is adding overhead.

### Describe the solution you'd like.

It would be better if there is native ECM driver available for esp-idf which can take advantages of full bandwidth of the modem and reduce overhead and latency in network operations. Because many projects use LTE modem and not WiFi for networking.

ParthaSarathiMishra, 2024-12-16:
@sudeep-mohanty 
Any opinion?

tore-espressif, 2024-12-19:
@ParthaSarathiMishra Thank you for the issue.

Could you please share what performance do you expect?

The USB peripheral on ESP32-S3 is Full-Speed only, so practically it can give you 1MB/s at maximum.
The AT commands should not add any overhead after the modem was switched to 'data mode'.

ParthaSarathiMishra, 2024-12-19:
@tore-espressif 
Thanks for the response.

We have tested both using AT commands through USB and using the Quectel modem as a network module in PC using the same USB port. With AT commands what FTP upload takes 2mins through ESP using CDC ACM driver, in PC with ECM driver, it takes only 10sec.

That's why asking for any ECM driver for ESP.

tore-espressif, 2024-12-19:
> using the same USB port

What do you mean by 'the same USB port'. The USB port on your PC is at least High-Speed (480Mbit/s), while the port on esp32-s3 is Full-Speed (12Mbit/s)

ParthaSarathiMishra, 2024-12-19:
@tore-espressif 
Same USB port means the USB port of Quectel modem. Same USB is used for AT commands with usb while the same interface with PC with ECM driver to act as a network interface.

david-cermak, 2024-12-20:
Hi @ParthaSarathiMishra 

Are you actually using the CDC ACM driver with AT command? (e.g. sending Quectel specific commands for the FTP upload?),
Have you tried using data/PPP mode?
The PPPoS over CDC-ACM would be still theoretically slower because of the PPP protocol overhead and perhaps fragmentation, but the difference would be negligible IMO.

ParthaSarathiMishra, 2024-12-20:
david-cermak

We are using cdc_acm_host_data_tx_blocking() to manually send Quectel AT commands and parsing USB response which we receive through the handle_rx(.data_cb).

We haven't tried data/PPP mode.

david-cermak, 2024-12-20:
Please try the PPPoS mode, then. you'll get closer to the actual speed limit and probably to the same numbers as you'd get with the ECM.
The easiest way to start is with these components: `esp_modem` and `esp_modem_usb_dte`, or simply create a demo project using:
```
idf.py create-project-from-example "espressif/esp_modem=1.3.0:pppos_client"
```

ParthaSarathiMishra, 2024-12-20:
@david-cermak 
Will try and let you know.

lijunru-hub, 2025-01-15:
I also support developing a standard ECM & RNDIS driver, which will help us connect to devices based on standard protocols. This can eliminate the need for the AT dialing step.

SumeriLal, 2025-02-15:
Hi @ParthaSarathiMishra 

Can you please share the program for ESP32-S3 ? So, that we can review which kind of driver you are intializing.