espressif/esp-idf #15029: esp_wifi_start stops USB on ESP32-S5

JesperBllnbm, 2024-12-14:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.3.2.

### Espressif SoC revision.

ESP32 C6

### Operating System used.

macOS

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

None

### Development Kit.

ESP32 C6 mini u1

### Power Supply used.

USB

### What is the expected behavior?

I expect to be able to run the SOFT AP example from: https://github.com/espressif/esp-idf/tree/v5.3.2/examples/wifi/getting_started/softAP

and it should work as documented.

### What is the actual behavior?

I enter some kind of boot loop, whenever wifi is started usb connection is lost.

### Steps to reproduce.

1. Obtain hardware, I used: https://www.espressif.com/sites/default/files/documentation/esp32-c6-mini-1_mini-1u_datasheet_en.pdf in form of a LilyGo T-Knob
2. Flash the example *https://github.com/espressif/esp-idf/tree/v5.3.2/examples/wifi/getting_started/softAP* or anything that needs wifi.
3. Connect the monitor

### Debug Logs.

```plain
base) jesper@Mac softAP % '/Users/jesper/.espressif/python_env/idf5.3_py3.13_env/bin/python' '/Use
rs/jesper/esp/esp-idf/tools/idf_monitor.py' -p /dev/tty.usbmodem101 -b 115200 --toolchain-prefix ri
scv32-esp-elf- --target esp32c6 '/Users/jesper/esp/esp-idf/examples/wifi/getting_started/softAP/bui
ld/wifi_softAP.elf'
--- Warning: Serial ports accessed as /dev/tty.* will hang gdb if launched.
--- Using /dev/cu.usbmodem101 instead...
--- esp-idf-monitor 1.5.0 on /dev/cu.usbmodem101 115200
--- Quit: Ctrl+] | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H
ESP-ROM:esp32c6-20220919
Build:Sep 19 2022
rst:0x15 (USB_UART_HPSYS),boot:0xc (SPI_FAST_FLASH_BOOT)
Saved PC:0x4205b4a4
--- 0x4205b4a4: mpi_hal_clear_interrupt at /Users/jesper/esp/esp-idf/components/hal/mpi_hal.c:46

SPIWP:0xee
mode:DIO, clock div:2
load:0x40875720,len:0x18ec
[... 175 lines cut ...]
W (610) wifi:(TB)WDEV_PWR_TB_MCS5:19
W (610) wifi:(TB)WDEV_PWR_TB_MCS6:18
W (610) wifi:(TB)WDEV_PWR_TB_MCS7:18
W (620) wifi:(TB)WDEV_PWR_TB_MCS8:17
W (620) wifi:(TB)WDEV_PWR_TB_MCS9:15
W (620) wifi:(TB)WDEV_PWR_TB_MCS10:15
W (630) wifi:(TB)WDEV_PWR_TB_MCS11:15
I (630) wifi:11ax coex: WDEVAX_PTI0(0x55777555), WDEVAX_PTI1(0x00003377).

W (640) wifi:(he)deattach, ap, phymode:11bgn
I (640) wifi:mode : softAP (40:4c:ca:46:8f:45)
I (690) wifi:Total power save buffer number: 16
I (690) wifi:Init max length of beacon: 752/752
I (690) wifi:Init max length of beacon: 752/752
--- Error: read failed: [Errno 6] Device not configured
```

### More Information.

Maybe related to:

SOC_WIFI_PHY_NEEDS_USB_WORKAROUND[](https://docs.espressif.com/projects/esp-idf/en/stable/esp32s3/api-reference/system/soc_caps.html#c.SOC_WIFI_PHY_NEEDS_USB_WORKAROUND)
SoC has WiFi and USB PHYs interference, needs a workaround

or #8241

hansw123, 2024-12-17:
@JesperBllnbm 
sorry for late for reply,
could you reproduce this issue when you flash with linux pc

JesperBllnbm, 2024-12-17:
@hansw123 Thanks for responding! I don't have a linux computer. Is there anything else I can try?

JesperBllnbm, 2024-12-18:
Sadly the same:

```sh
/opt/homebrew/bin/cmake

/opt/homebrew/bin/ninja

/opt/homebrew/bin/cmake

/opt/homebrew/bin/ninja

[Build]
Project build complete. To flash, run:
ESP-IDF: Flash your project in the ESP-IDF Visual Studio Code Extension
or in a ESP-IDF Terminal:
idf.py flash
or
idf.py -p PORT flash
or
python -m esptool --chip esp32 -b 460800 --before default_reset --after hard_reset  --port /dev/ttyUSB1 write_flash --flash_mode dio --flash_size 2MB --flash_freq 40m 0x1000 bootloader/bootloader.bin 0x10000 wifi_softAP.bin 0x8000 partition_table/partition-table.bin 
or from the "/Users/jesper/Development/softAP/build" directory
python -m esptool --chip esp32 -b 460800 --before default_reset --after hard_reset write_flash "@flash_args"
Adding "set-target"'s dependency "fullclean" to list of commands with default set of options.
Executing action: fullclean
Executing action: set-target
Set Target to: esp32c6, new sdkconfig will be created.
Running cmake in directory /Users/jesper/Development/softAP/build
Executing "cmake -G Ninja -DPYTHON_DEPS_CHECKED=1 -DPYTHON=/Users/jesper/.espressif/python_env/idf5.3_py3.13_env/bin/python -DESP_PLATFORM=1 -DIDF_TARGET=esp32c6 -DCCACHE_ENABLE=0 /Users/jesper/Development/softAP"...
-- Existing sdkconfig '/Users/jesper/Development/softAP/sdkconfig' renamed to '/Users/jesper/Development/softAP/sdkconfig.old'.
-- Found Git: /usr/bin/git (found version "2.39.5 (Apple Git-154)")

-- The C compiler identification is GNU 13.2.0

-- The CXX compiler identification is GNU 13.2.0

-- The ASM compiler identification is GNU

-- Found assembler: /Users/jesper/.espressif/tools/riscv32-esp-elf/esp-13.2.0_20240530/riscv32-esp-elf/bin/riscv32-esp-elf-gcc

-- Detecting C compiler ABI info

-- Detecting C compiler ABI info - done

-- Check for working C compiler: /Users/jesper/.espressif/tools/riscv32-esp-elf/esp-13.2.0_20240530/riscv32-esp-elf/bin/riscv32-esp-elf-gcc - skipped

-- Detecting C compile features

-- Detecting C compile features - done

-- Detecting CXX compiler ABI info

-- Detecting CXX compiler ABI info - done

-- Check for working CXX compiler: /Users/jesper/.espressif/tools/riscv32-esp-elf/esp-13.2.0_20240530/riscv32-esp-elf/bin/riscv32-esp-elf-g++ - skipped

-- Detecting CXX compile features

-- Detecting CXX compile features - done

-- git rev-parse returned 'fatal: not a git repository (or any of the parent directories): .git'
-- Could not use 'git describe' to determine PROJECT_VER.

-- Building ESP-IDF components for target esp32c6

-- Project sdkconfig file /Users/jesper/Development/softAP/sdkconfig

-- Adding linker script /Users/jesper/esp/v5.3.2/esp-idf/components/riscv/ld/rom.api.ld

CMake Deprecation Warning at /Users/jesper/esp/v5.3.2/esp-idf/components/mbedtls/mbedtls/CMakeLists.txt:21 (cmake_minimum_required):
  Compatibility with CMake < 3.10 will be removed from a future version of
[... 521 lines cut ...]
I (531) phy_init: phy_version 320,348a293,Sep  3 2024,16:33:12
W (581) wifi:ACK_TAB0   :0x   90a0b, QAM16:0x9 (24Mbps), QPSK:0xa (12Mbps), BPSK:0xb (6Mbps)
W (581) wifi:CTS_TAB0   :0x   90a0b, QAM16:0x9 (24Mbps), QPSK:0xa (12Mbps), BPSK:0xb (6Mbps)
W (581) wifi:(agc)0x600a7128:0xd207c9ec, min.avgNF:0xce->0xd2(dB), RCalCount:0x7c, min.RRssi:0x9ec(-97.25)
W (591) wifi:(TB)WDEV_PWR_TB_MCS0:19
W (591) wifi:(TB)WDEV_PWR_TB_MCS1:19
W (601) wifi:(TB)WDEV_PWR_TB_MCS2:19
W (601) wifi:(TB)WDEV_PWR_TB_MCS3:19
W (601) wifi:(TB)WDEV_PWR_TB_MCS4:19
W (611) wifi:(TB)WDEV_PWR_TB_MCS5:19
W (611) wifi:(TB)WDEV_PWR_TB_MCS6:18
W (611) wifi:(TB)WDEV_PWR_TB_MCS7:18
W (621) wifi:(TB)WDEV_PWR_TB_MCS8:17
W (621) wifi:(TB)WDEV_PWR_TB_MCS9:15
W (621) wifi:(TB)WDEV_PWR_TB_MCS10:15
W (631) wifi:(TB)WDEV_PWR_TB_MCS11:15
I (631) wifi:11ax coex: WDEVAX_PTI0(0x55777555), WDEVAX_PTI1(0x00003377).

W (641) wifi:(he)deattach, ap, phymode:11bgn
I (641) wifi:mode : softAP (40:4c:ca:46:8f:45)
I (691) wifi:Total power save buffer number: 16
I (691) wifi:Init max length of beacon: 752/752
I (691) wifi:Init max length of beacon: 752/752
--- Error: read failed: [Errno 6] Device not configured
```

JesperBllnbm, 2024-12-18:
If I flash the ESP using a VS Code dev container it should be technically as if I flashed it from a Linux computer, right? So I can try that as well.

hansw123, 2024-12-18:
@JesperBllnbm 
we test c6 with idf.5.3.2  in macos,  everything is ok
from you log, your idf is dirty,  so could you please provide more diff  or what you changed info

JesperBllnbm, 2025-01-03:
@hansw123 Hmm that's weird. I did a clean install using the vs code extension and still get `v5.3.2-dirty`. But I didn't change anything. I'll try to ask the manufacturer of the board if they can reproduce the error, tbh I am bit out of ideas here.

hansw123, 2025-01-23:
@JesperBllnbm  maybe you can download the espidf again，or you can change the esp_idf version, we have well test, esp32c6 with macos is ok

JesperBllnbm, 2025-01-24:
@hansw123 Yeah, I tried that too. My guess is there's either something wrong with my board or this specific board just doesn’t support Wi-Fi at all. I’ve opened an issue on the manufacturer’s example repo [here](https://github.com/Xinyuan-LilyGO/T-Knob/issues/1).

JesperAtToni, 2025-02-08:
Just an update. Wifi works if I use the Arduino IDE to flash an example.

JesperBllnbm, 2025-02-08:
Just an update. Wifi works if I use the Arduino IDE to flash an example. And I tried IDF 5.4 with the same results. I really wonder what's going on.

hansw123, 2025-02-12:
@JesperBllnbm  
from the project, maybe the project you build were depend on the Arduino?, we have test the c6 in all idf version in macos.

closed 2025-02-14 by MaxwellAlan