espressif/esp-idf #15032: show graphic can't keep

learnerzcz, 2024-12-15:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

i use spi master demo in lcd display, the displayed "esp32" only keep a moment, how can i keep it displaying

` while (1) {
        // frame++;
        for (int y = 0; y < 240; y += PARALLEL_LINES) {
            // printf("y=%d\n",y);
            //Calculate a line.
            pretty_effect_calc_lines(lines[calc_line], y, frame, PARALLEL_LINES);
            //Finish up the sending process of the previous line, if any
            if (sending_line != -1) {
                send_line_finish(spi);
            }
            //Swap sending_line and calc_line
            sending_line = calc_line;
            calc_line = (calc_line == 1) ? 0 : 1;
            //Send the line we currently calculated.
            send_lines(spi, y, lines[sending_line]);
            //The line set is queued up for sending now; the actual sending happens in the
            //background. We can go on to calculate the next line set as long as we do not
            //touch line[sending_line]; the SPI sending process is still reading from that.
        }
        // vTaskDelay(pdMS_TO_TICKS(100));

     }`

nopnop2002, 2024-12-15:
>how can i keep it displaying

If you don't change any code, it will continue to display.

learnerzcz, 2024-12-15:
Yes，i found the problem ,is the backlight level is err i set.

closed 2024-12-15 by learnerzcz