espressif/esp-idf #15033: esp32S3 as OTBR, " OPENTHREAD: Failed to get LL address, error: Invalid If index"

githubworksonline, 2024-12-15:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

TAG: 5.3.2

### Espressif SoC revision.

Chip rev:  v0.1

### Operating System used.

Windows

### How did you build your project?

Command line with Make

### If you are using Windows, please specify command line type.

None

### Development Kit.

esp32s3 chip

### Power Supply used.

USB

### What is the expected behavior?

i do not understand openthread, it is my fist run this example. after 10minutes, nothing happened.

### What is the actual behavior?

i want it run as "https://docs.espressif.com/projects/esp-thread-br/en/latest/codelab/index.html"

### Steps to reproduce.

only idf5.3.2 otbr example on esp32S3 and ncp example on esp32H2.

### Debug Logs.

```plain
factoryreset

I (1017808) wifi:state: run -> init (0x0)
I (1017808) wifi:pm stop, total sleep time: 975447964 us / 1014050981 us

I (1017808) wifi:<ba-del>idx:0, tid:0
I (1017808) wifi:new:<8,0>, old:<8,1>, ap:<255,255>, sta:<8,1>, prof:1, snd_ch_cfg:0x0
 [0;32mI (1017828) example_connect: Wi-Fi disconnected 8, trying to reconnect... [0m
I (1017868) wifi:flush txq
I (1017868) wifi:stop sw txq
I (1017868) wifi:lmac stop hw txq
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0xc (RTC_SW_CPU_RST),boot:0x8 (SPI_FAST_FLASH_BOOT)
Saved PC:0x40375c30
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce2810,len:0x1870
load:0x403c8700,len:0x4
load:0x403c8704,len:0xce8
load:0x403cb700,len:0x2ed8
[... 106 lines cut ...]
 [0;32mI(6798) OPENTHREAD:[N] RoutingManager: BR ULA prefix: fd16:7835:7d60::/48 (generated) [0m
 [0;32mI(6798) OPENTHREAD:[N] RoutingManager: Local on-link prefix: fdde:ad00:beef:cafe::/64 [0m
 [0;31mE (6828) OPENTHREAD: Failed to leave multicast group [0m
 [0;32mI (6838) OPENTHREAD: Platform UDP bound to port 49153 [0m
 [0;32mI(6848) OPENTHREAD:[N] Mle-----------: Role disabled -> detached [0m
 [0;32mI (6858) OPENTHREAD: Platform UDP bound to port 61631 [0m
 [0;32mI (6858) OT_STATE: netif up [0m
I (6868) OPENTHREAD: NAT64 ready
 [0;32mI(7338) OPENTHREAD:[N] Mle-----------: Attach attempt 1, AnyPartition reattaching with Active Dataset [0m
 [0;32mI (7568) example_connect: Got IPv6 event: Interface "example_netif_sta" address: (hidded), type: ESP_IP6_ADDR_IS_GLOBAL [0m
 [0;32mI(8188) OPENTHREAD:[N] Mle-----------: RLOC16 fffe -> dc01 [0m
 [0;32mI(8198) OPENTHREAD:[N] Mle-----------: Role detached -> child [0m
 [0;32mI (8208) OPENTHREAD: Platform UDP bound to port 49154 [0m
 [0;32mI (8228) OPENTHREAD: Platform UDP bound to port 49155 [0m
I (8248) OPENTHREAD: NAT64 ready
I (8248) OPENTHREAD: NAT64 ready
 [0;31mE (8588) OPENTHREAD: Failed to get LL address, error: Invalid If index [0m
 [0;33mW(8848) OPENTHREAD:[W] DuaManager----: Failed to perform next registration: NotFound [0m
 [0;32mI (9618) OPENTHREAD: Platform UDP bound to port 53536 [0m
I (9618) OPENTHREAD: NAT64 ready
 [0;31mE (12608) OPENTHREAD: Failed to get LL address, error: Invalid If index [0m
 [0;31mE (16628) OPENTHREAD: Failed to get LL address, error: Invalid If index [0m
 [0;31mE (17658) OPENTHREAD: Failed to get LL address, error: Invalid If index [0m
 [0;31mE (35248) OPENTHREAD: Failed to get LL address, error: Invalid If index [0m
```

zwx1995esp, 2024-12-17:
Hi @githubworksonline ,Thanks for issuing this, I will investigate it later. For now, please use the recommanded code base shown in the docs: https://docs.espressif.com/projects/esp-thread-br/en/latest/dev-guide/build_and_run.html#set-up-the-repositories

Kris-Morrison, 2024-12-18:
> Hi @githubworksonline ,Thanks for issuing this, I will investigate it later. For now, please use the recommanded code base shown in the docs: https://docs.espressif.com/projects/esp-thread-br/en/latest/dev-guide/build_and_run.html#set-up-the-repositories

Yeah, I just followed the recommended code base - 5.3.2 and got same error "OPENTHREAD: Failed to get LL address, error: Invalid If index"

GASEN1216, 2024-12-19:
I had the same error, also using esp32S3 as OTBR

giangnv1993, 2024-12-25:
> I had the same error, also using esp32S3 as OTBR

Hi, I had the same issue. Did you can control from chip tool to end device ( ex: light, light_switch)?

zwx1995esp, 2024-12-25:
Hi @giangnv1993 @GASEN1216 @Kris-Morrison @githubworksonline  Thanks for issuing this, we have fixed this issue in our internal branch and will sync to github very soon, I will let you know if the fix becomes available in release/v5.3 branch on GitHub.

gparvind, 2024-12-25:
@zwx1995esp could you please share what was the issue and fix?

zwx1995esp, 2024-12-25:
> @zwx1995esp could you please share what was the issue and fix?

Sure, it is a software issue. Our [BR lib](https://github.com/espressif/esp-idf/tree/master/components/openthread) is a pre-built static binary library. Some structs used in BR lib is different between the branch v5.3.2 and the built branch master. This discrepancy leads to an ABI issue, as the data types are misaligned between the library and the code on branch v5.3.2.

gparvind, 2024-12-26:
@zwx1995esp any ETA on when the fix can be available? or possible to share the fix so we can apply it locally until 5.3/5.4 is updated?

zwx1995esp, 2024-12-26:
Hi @gparvind @giangnv1993 @githubworksonline @GASEN1216 The fixed lib has been published, you can access the lib via the commit [fde54d57185b33613f4b3f6d6aba151e31c33ba4](https://github.com/espressif/esp-idf/commit/fde54d57185b33613f4b3f6d6aba151e31c33ba4) or the latest branch of release/v5.3

closed 2025-01-03 by Alvin1Zhang