espressif/esp-idf #15034: SPI_Slave Example Code, print result messy

Bennyggcc, 2024-12-15:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

V3.5.2

### Espressif SoC revision.

ESP32-S3

### Operating System used.

Windows

### How did you build your project?

VS Code IDE

### If you are using Windows, please specify command line type.

None

### Development Kit.

Custom Board

### Power Supply used.

USB

### What is the expected behavior?

I run the SPI_Slave Example Code (esp-idf/examples/peripherals/spi_slave /(sender/receiver)), run 'sender code' on one esp32-s3 board as master, 'receiver code' on another esp32-s3 board as slaver. 

master code revised the SPI pin as follow:
[image]
slave code revised the SPI pin as follow:
[image]

I hope the host/slave can send data correctly and receive the response from the slave/host. 

But get error: --- Warning: Failed to decode multiple lines in a row. Try checking the baud rate and XTAL frequency setting in menuconfig.  The XTAL frequency  of ESP32-S3 just can be set as 40Mhz,  and I try to set the baud rate to 115200 still get the error. How can I solve these problems? 

Thank you!

### What is the actual behavior?

But the print content is the messy code as follow:
Master print:
[image]
Slave print:
[image]

### Steps to reproduce.

I just reproduce on SPI_slave example code, and change the correct SPI PIN, did not change other places.

### Debug Logs.

```plain
Master Device:
esptool.py v4.8.1
Serial port COM7
Connecting....
Chip is ESP32-S3 (QFN56) (revision v0.2)
Features: WiFi, BLE, Embedded PSRAM 8MB (AP_3v3)
Crystal is 40MHz
MAC: dc:da:0c:18:d4:60
Uploading stub...
Running stub...
Stub running...
Changing baud rate to 460800
Changed.
Configuring flash size...
Flash will be erased from 0x00000000 to 0x00005fff...
Flash will be erased from 0x00010000 to 0x0004efff...
Flash will be erased from 0x00008000 to 0x00008fff...
SHA digest in image updated
Compressed 21648 bytes to 13304...
Wrote 21648 bytes (13304 compressed) at 0x00000000 in 0.6 seconds (effective 269.9 kbit/s)...
Hash of data verified.
Compressed 256864 bytes to 129511...
Wrote 256864 bytes (129511 compressed) at 0x00010000 in 3.3 seconds (effective 627.0 kbit/s)...
Hash of data verified.
Compressed 3072 bytes to 103...
Wrote 3072 bytes (103 compressed) at 0x00008000 in 0.1 seconds (effective 363.6 kbit/s)...
Hash of data verified.

Slave Device:
esptool.py v4.8.1
Serial port COM5
Connecting....
Chip is ESP32-S3 (QFN56) (revision v0.1)
Features: WiFi, BLE, Embedded PSRAM 8MB (AP_3v3)
Crystal is 40MHz
MAC: 34:85:18:97:3f:28
Uploading stub...
Running stub...
Stub running...
Changing baud rate to 460800
Changed.
Configuring flash size...
Flash will be erased from 0x00000000 to 0x00005fff...
Flash will be erased from 0x00010000 to 0x0004bfff...
Flash will be erased from 0x00008000 to 0x00008fff...
SHA digest in image updated
Compressed 21648 bytes to 13304...
Wrote 21648 bytes (13304 compressed) at 0x00000000 in 0.6 seconds (effective 295.6 kbit/s)...
Hash of data verified.
Compressed 245728 bytes to 123510...
Wrote 245728 bytes (123510 compressed) at 0x00010000 in 3.0 seconds (effective 664.8 kbit/s)...
Hash of data verified.
Compressed 3072 bytes to 103...
Wrote 3072 bytes (103 compressed) at 0x00008000 in 0.1 seconds (effective 431.1 kbit/s)...
Hash of data verified.
```

mythbuster5, 2024-12-16:
@wanckl

wanckl, 2024-12-18:
@Bennyggcc  
Seems like slave can receive what master send, however master can not receive what slave send.

As you are using your Custom Board, can you check if any hardware issue, maybe on your GPIO12/13,

Or you can simply re-oder/change these pins and try again

Bennyggcc, 2024-12-23:
> @Bennyggcc Seems like slave can receive what master send, however master can not receive what slave send.
> 
> As you are using your Custom Board, can you check if any hardware issue, maybe on your GPIO12/13,
> 
> Or you can simply re-oder/change these pins and try again

@wanckl Thanks for your reply. But it seems not the hardware issue. I have succeeded to use this example code (https://github.com/hideakitai/ESP32DMASPI) based on the pins mentioned above to communicate between the Master and Slave device.