espressif/esp-idf #15035: file FILE([TO_CMAKE_PATH|TO_NATIVE_PATH] path result) must be called with   exactly three arguments.

SB211, 2024-12-15:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.3.2-176-g64e5b924f4

### Espressif SoC revision.

ESP32/ESP32-S3 to ESP32-C3/ESP32-C6

### Operating System used.

Linux

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

None

### Development Kit.

ESP32-S3 and ESP32-ROVER

### Power Supply used.

External 5V

### What is the expected behavior?

idf.py set-target esp32 is supposed to adjust the ESP framework according the used/defined ESP hardware.

### What is the actual behavior?

After the most recent update (git pull) of ESP-IDF 5.3.2, idf.py bails out as shown below. It seems the IDF_TOOL variable seems unset. Searching/grepping the folder IDF_PATH points to does not reveal any obvious and intuitive setting procedere of this variable.

[...]
Processing 1 dependencies:
[1/1] idf (5.3.2)
-- Project sdkconfig file /pool/home/ohartmann/Projects/datensammler/sdkconfig
-- Compiler supported targets: xtensa-esp-elf
CMake Deprecation Warning at /pool/home/ohartmann/Projects/esp-idf/components/mbedtls/mbedtls/CMakeLists.txt:21 (cmake_minimum_required):
  Compatibility with CMake < 3.10 will be removed from a future version of
  CMake.

  Update the VERSION argument <min> value.  Or, use the <min>...<max> syntax
  to tell CMake that the project requires at least <min> but has been updated
  to work with policies introduced by <max> or earlier.

-- Found Python3: /usr/local/bin/python (found version "3.11.11") found components: Interpreter
-- Performing Test CMAKE_HAVE_LIBC_PTHREAD
-- Performing Test CMAKE_HAVE_LIBC_PTHREAD - Success
-- Found Threads: TRUE
-- Performing Test C_COMPILER_SUPPORTS_WFORMAT_SIGNEDNESS
-- Performing Test C_COMPILER_SUPPORTS_WFORMAT_SIGNEDNESS - Success
-- App "datensammler" version: 0.0.1
-- Adding linker script /pool/home/ohartmann/Projects/datensammler/build/esp-idf/esp_system/ld/memory.ld
-- Adding linker script /pool/home/ohartmann/Projects/datensammler/build/esp-idf/esp_system/ld/sections.ld.in
-- Adding linker script /pool/home/ohartmann/Projects/esp-idf/components/esp_rom/esp32/ld/esp32.rom.ld
-- Adding linker script /pool/home/ohartmann/Projects/esp-idf/components/esp_rom/esp32/ld/esp32.rom.api.ld
-- Adding linker script /pool/home/ohartmann/Projects/esp-idf/components/esp_rom/esp32/ld/esp32.rom.libgcc.ld
-- Adding linker script /pool/home/ohartmann/Projects/esp-idf/components/esp_rom/esp32/ld/esp32.rom.newlib-data.ld
-- Adding linker script /pool/home/ohartmann/Projects/esp-idf/components/esp_rom/esp32/ld/esp32.rom.syscalls.ld
-- Adding linker script /pool/home/ohartmann/Projects/esp-idf/components/esp_rom/esp32/ld/esp32.rom.newlib-funcs.ld
-- Adding linker script /pool/home/ohartmann/Projects/esp-idf/components/soc/esp32/ld/esp32.peripherals.ld
Including esp_timer in color
CMake Error at /pool/home/ohartmann/Projects/esp-idf/tools/cmake/gdbinit.cmake:40 (file):
  file FILE([TO_CMAKE_PATH|TO_NATIVE_PATH] path result) must be called with
  exactly three arguments.
Call Stack (most recent call first):
  /home/ohartmann/Projects/esp-idf/tools/cmake/project.cmake:356 (__generate_gdbinit)
  /home/ohartmann/Projects/esp-idf/tools/cmake/project.cmake:923 (__project_info)
  CMakeLists.txt:11 (project)

-- Configuring incomplete, errors occurred!

### Steps to reproduce.

1. Step: update to most recent ESP-IDF 3.5.2 
2. Step: try rebuilding project via idf.py which worked fine with ESP-IDF 3.5.1
3. Step: be amazed ...
...

### Debug Logs.

```plain
[...]
Processing 1 dependencies:
[1/1] idf (5.3.2)
-- Project sdkconfig file /pool/home/ohartmann/Projects/datensammler/sdkconfig
-- Compiler supported targets: xtensa-esp-elf
CMake Deprecation Warning at /pool/home/ohartmann/Projects/esp-idf/components/mbedtls/mbedtls/CMakeLists.txt:21 (cmake_minimum_required):
  Compatibility with CMake < 3.10 will be removed from a future version of
  CMake.

  Update the VERSION argument <min> value.  Or, use the <min>...<max> syntax
  to tell CMake that the project requires at least <min> but has been updated
  to work with policies introduced by <max> or earlier.

-- Found Python3: /usr/local/bin/python (found version "3.11.11") found components: Interpreter
-- Performing Test CMAKE_HAVE_LIBC_PTHREAD
-- Performing Test CMAKE_HAVE_LIBC_PTHREAD - Success
-- Found Threads: TRUE
-- Performing Test C_COMPILER_SUPPORTS_WFORMAT_SIGNEDNESS
-- Performing Test C_COMPILER_SUPPORTS_WFORMAT_SIGNEDNESS - Success
-- App "datensammler" version: 0.0.1
-- Adding linker script /pool/home/ohartmann/Projects/datensammler/build/esp-idf/esp_system/ld/memory.ld
-- Adding linker script /pool/home/ohartmann/Projects/datensammler/build/esp-idf/esp_system/ld/sections.ld.in
-- Adding linker script /pool/home/ohartmann/Projects/esp-idf/components/esp_rom/esp32/ld/esp32.rom.ld
-- Adding linker script /pool/home/ohartmann/Projects/esp-idf/components/esp_rom/esp32/ld/esp32.rom.api.ld
-- Adding linker script /pool/home/ohartmann/Projects/esp-idf/components/esp_rom/esp32/ld/esp32.rom.libgcc.ld
-- Adding linker script /pool/home/ohartmann/Projects/esp-idf/components/esp_rom/esp32/ld/esp32.rom.newlib-data.ld
-- Adding linker script /pool/home/ohartmann/Projects/esp-idf/components/esp_rom/esp32/ld/esp32.rom.syscalls.ld
-- Adding linker script /pool/home/ohartmann/Projects/esp-idf/components/esp_rom/esp32/ld/esp32.rom.newlib-funcs.ld
-- Adding linker script /pool/home/ohartmann/Projects/esp-idf/components/soc/esp32/ld/esp32.peripherals.ld
Including esp_timer in color
CMake Error at /pool/home/ohartmann/Projects/esp-idf/tools/cmake/gdbinit.cmake:40 (file):
  file FILE([TO_CMAKE_PATH|TO_NATIVE_PATH] path result) must be called with
  exactly three arguments.
Call Stack (most recent call first):
  /home/ohartmann/Projects/esp-idf/tools/cmake/project.cmake:356 (__generate_gdbinit)
  /home/ohartmann/Projects/esp-idf/tools/cmake/project.cmake:923 (__project_info)
  CMakeLists.txt:11 (project)

-- Configuring incomplete, errors occurred!
```

igrr, 2024-12-15:
@SB211 Could you please mention, which version of CMake are you running (`cmake --version`)?

IDF v5.3.2 requires CMake >= 3.16. It seems, however, that gdbinit.cmake (which uses a more recent CMake syntax) gets included before the version check happens, hence the confusing error message.

I think upgrading CMake to a more recent version should solve the issue you are seeing.

SB211, 2024-12-15:
cmake version 3.31.2

SB211, 2024-12-15:
According to "git log", ESP-IDF v5.3.2 has despite being a release version undergone some changes in gdbinit.

The error occured recently  occured is related to "./tools/cmake/gdbinit.cmake",  line 40, which is

file(TO_CMAKE_PATH $ENV{ESP_ROM_ELF_DIR} ESP_ROM_ELF_DIR)

My first guess is/was that ESP_ROM_ELF_DIR is unknown/unset. My skills according to "cmake/ninja" are limited, usually I would check via "make -V ESP_ROM_ELF_DIR" the setting of that variable. So far.

Searching for ESP_ROM_ELF_DIR within ESP-IDF check-out folder via "grep -r ESP_ROM_ELF_DIR .", I found this lines:

[...]
./components/esp_rom/test_esp_rom.py:    rom_elfs_dir = os.getenv('ESP_ROM_ELF_DIR', '')
./components/esp_rom/gen_gdbinit.py:    rom_elfs_dir = os.getenv('ESP_ROM_ELF_DIR')
./components/esp_rom/gen_gdbinit.py:            'ESP_ROM_ELF_DIR environment variable is not defined. Please try to run IDF "install" and "export" scripts.')
./docs/en/api-guides/tools/idf-monitor.rst:The ROM ELF file is automatically loaded from a location based on the ``IDF_PATH`` and ``ESP_ROM_ELF_DIR`` environment variables. This can be overridden by calling ``esp_idf_monitor`` and providing a path to a specific ROM ELF file: ``python -m esp_idf_monitor --rom-elf-file [path to ROM ELF file]``.
./docs/zh_CN/api-guides/tools/idf-monitor.rst:ROM ELF 文件会根据 ``IDF_PATH`` 和 ``ESP_ROM_ELF_DIR`` 环境变量的路径自动加载。如需覆盖此行为，可以通过调用 ``esp_idf_monitor`` 并指定特定的 ROM ELF 文件路径：``python -m esp_idf_monitor --rom-elf-file [ROM ELF 文件的路径]``。
./tools/tools.json:        "ESP_ROM_ELF_DIR": "${TOOL_PATH}/"
./tools/cmake/gdbinit.cmake:    file(TO_CMAKE_PATH $ENV{ESP_ROM_ELF_DIR} ESP_ROM_ELF_DIR)
[...]

Please correct me if I'm wrong: ./tools/tools.json:        "ESP_ROM_ELF_DIR": "${TOOL_PATH}/" seems to be the crucial section. I guess TOOL_PATH is unset for unknown reasons.
Isn't the ROM file something being built according to selected ESP32 type (RISC-V oder Tensilica Core, Revision of the chip), is it? Calling "idf.py menuconfig" therefore can't know anything about ROM files - except according to the file system's hierachy (project's directory) the will-be-build-in location of the ROM would be known - but not set accordingly. Just a guess. I do not know what tool might set ESP_ROM_ELF_DIR or whether this variable is a new environment variable which has to be set by the develeoper.

SB211, 2024-12-15:
Reverting

commit 3af39a0ab0d0cb767c55c03222f6d3b85df0a0b2

solves the problem!

igrr, 2024-12-15:
> I guess TOOL_PATH is unset for unknown reasons.

You are right, that seems like the most likely cause.

Could you please run `. ./export.sh` in ESP-IDF directory again and then run `printenv` to check if `ESP_ROM_ELF_DIR` environment variable gets set by the export script?

eborchardt, 2025-01-13:
I ran into the same issue running on Windows. It does seem that the ESP_ROM_ELF_DIR variable is not getting set for some reason. I added ESP_ROM_ELF_DIR to the CMAKE settings in CLION and the error went away.

hathorr-hub, 2025-02-17:
I am also facing the same issue :
`
[build] CMake Error at /workspaces/nodes/build/esp32p4/_deps/esp_idf-src/tools/cmake/gdbinit.cmake:40 (file):
[build]   file FILE([TO_CMAKE_PATH|TO_NATIVE_PATH] path result) must be called with
[build]   exactly three arguments.
[build] Call Stack (most recent call first):
`
ESP-IDF: release/v5.4

How to fix this issue?

closed 2025-03-24 by espressif-bot

kevinhikaruevans, 2025-04-29:
If anybody is facing this issue using PlatformIO using the latest ESP IDF `platformio/framework-espidf@3.50401.0`, I applied this patch https://github.com/espressif/esp-idf/commit/b64ddb18939d06426607a04816e6de881524e87f in `~/.platformio/packages/framework-espidf@3.50301.0`

nightrider77, 2025-06-08:
esp-idf-v5.4.1

```
Executing action: flash
Running cmake in directory C:\Users\admin\workspace\hello\build
Executing "cmake -G Ninja -DPYTHON_DEPS_CHECKED=1 -DPYTHON=C:\Espressif\python_env\idf5.4_py3.11_env\Scripts\python.exe -DESP_PLATFORM=1 -DCCACHE_ENABLE=0 C:\Users\admin\workspace\hello"...
-- git rev-parse returned 'fatal: not a git repository (or any of the parent directories): .git'
-- Could not use 'git describe' to determine PROJECT_VER.
-- Building ESP-IDF components for target esp32
-- Project sdkconfig file C:/Users/admin/workspace/hello/sdkconfig
-- Compiler supported targets: xtensa-esp-elf
-- Found Python3: C:/Espressif/python_env/idf5.4_py3.11_env/Scripts/python.exe (found version "3.11.2") found components: Interpreter
-- App "app-template" version: 1
-- Adding linker script C:/Users/admin/workspace/hello/build/esp-idf/esp_system/ld/memory.ld
-- Adding linker script C:/Users/admin/workspace/hello/build/esp-idf/esp_system/ld/sections.ld.in
-- Adding linker script C:/Espressif/frameworks/esp-idf-v5.4.1/components/esp_rom/esp32/ld/esp32.rom.ld
-- Adding linker script C:/Espressif/frameworks/esp-idf-v5.4.1/components/esp_rom/esp32/ld/esp32.rom.api.ld
-- Adding linker script C:/Espressif/frameworks/esp-idf-v5.4.1/components/esp_rom/esp32/ld/esp32.rom.libgcc.ld
-- Adding linker script C:/Espressif/frameworks/esp-idf-v5.4.1/components/esp_rom/esp32/ld/esp32.rom.newlib-data.ld
-- Adding linker script C:/Espressif/frameworks/esp-idf-v5.4.1/components/esp_rom/esp32/ld/esp32.rom.syscalls.ld
-- Adding linker script C:/Espressif/frameworks/esp-idf-v5.4.1/components/esp_rom/esp32/ld/esp32.rom.newlib-funcs.ld
-- Adding linker script C:/Espressif/frameworks/esp-idf-v5.4.1/components/soc/esp32/ld/esp32.peripherals.ld
CMake Error at C:/Espressif/frameworks/esp-idf-v5.4.1/tools/cmake/gdbinit.cmake:40 (file):
  file FILE([TO_CMAKE_PATH|TO_NATIVE_PATH] path result) must be called with
  exactly three arguments.
Call Stack (most recent call first):
  C:/Espressif/frameworks/esp-idf-v5.4.1/tools/cmake/project.cmake:356 (__generate_gdbinit)
  C:/Espressif/frameworks/esp-idf-v5.4.1/tools/cmake/project.cmake:923 (__project_info)
  CMakeLists.txt:6 (project)

-- Configuring incomplete, errors occurred!
cmake failed with exit code 1, output of the command is in the C:\Users\admin\workspace\hello\build\log\idf_py_stderr_output_20096 and C:\Users\admin\workspace\hello\build\log\idf_py_stdout_output_20096
```

?!

Lapshin, 2025-06-08:
@nightrider77 , fix is already in branch [release/v5.4](https://github.com/espressif/esp-idf/commits/release/v5.4) (https://github.com/espressif/esp-idf/commit/f810a440c92ea451faadd00bdb80bed3ed956b11). It should be available in the upcoming v5.4.2