espressif/esp-idf #15039: with the ESP-IDF Windows Installer to [make -C mpy-cross, make submodules] for MicroPython

qqjyh, 2024-12-16:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

Is it possible to [make -C mpy-cross, make submodules] with the [ESP-IDF Windows Installer] for ESP32 or ESP8266 ? where to search related issue & solutions ?

igrr, 2024-12-16:
No, it's not possible. To use GNU Make on Windows you need either an MSYS or Cygwin environment, and IDF installer doesn't install those.

At the same time, ESP-IDF no longer officially supports MSYS.

I think you can try getting your development environment set up in WSL, that should work with both micropython and Espressif SDKs.

As for a place to ask this question, I would recommend either [Espressif](https://esp32.com) or [micropython](https://forum.micropython.org/) forums.

qqjyh, 2024-12-16:
Thank you very much for the right & rapid answer.

closed 2024-12-16 by Alvin1Zhang