espressif/esp-idf #15041: Missing event loop profile states

maedoc, 2024-12-16:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.3.1 & master cb3ac7429c7

### Espressif SoC revision.

c6 has no chip id, MAC: f0:f5:bd:ff:fe:01:98:30

### Operating System used.

macOS

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

None

### Development Kit.

esp32-c6-devkitc-1 v1.2

### Power Supply used.

USB

### What is the expected behavior?

`esp_event_dump` should provide the number of handler invocations and time spent per handler, as described in the docs. 

### What is the actual behavior?

The profile stats remain zero. 

### Steps to reproduce.

1. create project
2. add simple main.c, 
```c
#include "esp_event.h"
#include "esp_log.h"
#include "stdio.h"

ESP_EVENT_DECLARE_BASE(DBG);
ESP_EVENT_DEFINE_BASE(DBG);

void handler(void* arg, esp_event_base_t event_base, int32_t event_id, void* data) {
    ESP_LOGI("handler", "Event received: base=%s, id=%ld, data=%p", event_base, event_id, data);
}

void app_main(void)
{
    esp_log_level_set("event", ESP_LOG_DEBUG);
    ESP_ERROR_CHECK(esp_event_loop_create_default());
    ESP_ERROR_CHECK(esp_event_handler_register(DBG, ESP_EVENT_ANY_ID, handler, NULL));

    vTaskDelay(pdMS_TO_TICKS(1000));
    ESP_ERROR_CHECK(esp_event_post(DBG, 0, NULL, 0, pdMS_TO_TICKS(1000)));
    ESP_ERROR_CHECK(esp_event_post(DBG, 1, NULL, 0, pdMS_TO_TICKS(1000)));

    vTaskDelay(pdMS_TO_TICKS(1000));
    ESP_ERROR_CHECK(esp_event_dump(stdout));
}
```
4. set-target esp32c6
5. menuconfig set LOG_MAX to DEBUG and LOOP_PROFILING to enabled
6. build flash monitor

### Debug Logs.

```plain
I (243) main_task: Started on CPU0
I (243) main_task: Calling app_main()
D (243) event: running task for loop 0x4087cd14
D (243) event: created task for loop 0x4087cd14
D (253) event: created event loop 0x4087cd14
D (1253) event: running post DBG:0 with handler 0x4200c540 and context 0x4087dd24 on loop 0x4087cd14
--- 0x4200c540: handler at /try-event-loop/main/test_app_main.c:8

I (1253) handler: Event received: base=DBG, id=0, data=0x0
D (1253) event: running post DBG:1 with handler 0x4200c540 and context 0x4087dd24 on loop 0x4087cd14
--- 0x4200c540: handler at /try-event-loop/main/test_app_main.c:8

I (1263) handler: Event received: base=DBG, id=1, data=0x0
LOOP @0x4087cd14,sys_evt rx:2 dr:0
  HANDLER @0x4200c540 ev:DBG,ESP_EVENT_ANY_ID inv:0 time:0 us
--- 0x4200c540: handler at /try-event-loop/main/test_app_main.c:8

I (2263) main_task: Returned from app_main()
```
```

SoucheSouche, 2024-12-17:
Hi @maedoc, thanks for reporting the issue. I will take a look at it.

SoucheSouche, 2024-12-20:
Hi @maedoc, I found out what caused the problem and created a fix. It is under review right now.

The event handlers are stored in different lists based on what events trigger them. Before updating the handler profiling information, the function calling the handler makes sure it did not unregistered itself by going through the lists to find if the handler is still there. The problem is that the function was in fact not going through all lists of handlers. That is why for certain handlers, the "invoked" and "time" field where never updated.

maedoc, 2024-12-20:
hi, great news! thanks for having a look

SoucheSouche, 2025-01-30:
The fix was just merged to master internally.
I will see to it that it is backported to the affected versions.

SoucheSouche, 2025-02-06:
> > The fix was just merged to master internally. I will see to it that it is backported to the affected versions.
> 
> Sound Great. 👍
> 
> Do you know on which version this bug was introduced?

I don't know exactly but the bug exists from v5.2 at least it seems. I backported it back to 5.2.

closed 2025-02-13 by espressif-bot

maedoc, 2025-02-19:
will this be fixed in v5.4? or v5.3?

Alvin1Zhang, 2025-02-20:
Thanks for reporting and sharing the updates, the fix on release/5.4 is available at https://github.com/espressif/esp-idf/commit/f870a03223f001bdad43fc4faeaca91461dbc1ca and release/5.3 is available at https://github.com/espressif/esp-idf/commit/295f691e14085261cffca82f19fafa5dde20c2a6, they will be included in the coming v5.4.1 release and v5.3.3 release (you could refer to https://github.com/espressif/esp-idf/blob/master/ROADMAP.md#esp-idf-bugfix-releases for the rough release time)