espressif/esp-idf #15042: Pthread in C++ vs C

franz-ms-muc, 2024-12-17:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

Comparing 2 Examples: 

https://github.com/espressif/esp-idf/blob/master/examples/system/pthread/main/pthread_example.c

https://github.com/espressif/esp-idf/blob/master/examples/cxx/pthread/main/cpp_pthread.cpp

First Example gives 2 Options of setting e.g. Stacksize: 
1. pthread_attr_t
2. esp_pthread_cfg

Second Example does not give these 2 Options. 
it seems the way to do is over  std::thread::native_handle
C++ Developers had not given other portable ways to do this. 
see: https://en.cppreference.com/w/cpp/thread/thread/native_handle

will this work in ESP-IDF ?

franz-ms-muc, 2024-12-17:
found antother: 

[https://www.open-std.org/jtc1/sc22/wg21/docs/papers/2016/p0320r1.pdf](https://www.open-std.org/jtc1/sc22/wg21/docs/papers/2016/p0320r1.pdf)

```c++
std::thread::attributes attrs;
attrs.set_stack_size(4096*10);
std::thread deep_thought_2(attrs, find_the_question, 42);
```

franz-ms-muc, 2024-12-17:
this seems not to be supported in ESP-IDF

`error: 'attributes' is not a member of 'std::thread'`

KonstantinKondrashov, 2024-12-19:
Hi @franz-ms-muc!

>https://github.com/espressif/esp-idf/issues/14821
there is coming something in ESP-IDF 5.4

We did not plan to backport it 5.4. We do not backport features.

-------------------

> First Example gives 2 Options of setting e.g. Stacksize:
Second Example does not give these 2 Options.

**1.** Indeed the `pthread` in C allows set `stacksize` in two ways via `pthread_attr_t` and `esp_pthread_cfg_t`:
```c
    // 1. pthread_attr_t
    pthread_attr_t attr
    pthread_attr_init(&attr);
    pthread_attr_setstacksize(&attr, 16384);
    pthread_create(&thread2, &attr, example_thread, NULL);

    // 2. esp_pthread_cfg_t
    esp_pthread_cfg_t  esp_pthread_cfg = esp_pthread_get_default_config();
    esp_pthread_cfg.stack_size = 16384;
    esp_pthread_cfg.prio = 5;
    esp_pthread_set_cfg(&esp_pthread_cfg);
    pthread_create(&thread1, NULL, example_thread, NULL);
```

**2.** The `pthread` in C++ has a different implication, but it still allows set `stacksize` the same way using `esp_pthread_cfg_t `:
```c++

    // 1. no way to use attr way for now

    // 2. esp_pthread_cfg_t 
    esp_pthread_cfg_t esp_pthread_cfg = esp_pthread_get_default_config();
    esp_pthread_cfg.stack_size = 16384;
    esp_pthread_cfg.prio = 5;
    esp_pthread_set_cfg(&esp_pthread_cfg);
    std::thread thread_1(thread_func);
```

This is an overview that we currently have for pthread for both C and C++. It makes sense to have the attribute way to set it. I will check what we can do with that. Thank you for raising this issue.

KonstantinKondrashov, 2024-12-23:
Hi @franz-ms-muc!
Based on [this paper](https://www.open-std.org/jtc1/sc22/wg21/docs/papers/2016/p0320r1.pdf), this feature appears to be a straightforward implementation by wrapping the `std::thread` constructor. For reference, we have a dedicated repository, [esp-idf-cxx](https://github.com/espressif/esp-idf-cxx), which focuses on adding custom C++ features for ESP-IDF.

If you find an effective approach to achieve this, we encourage you to contribute to the repository. While we currently consider this feature minor, as the same functionality can already be achieved with the current APIs.
Thanks.

KonstantinKondrashov, 2025-01-02:
I am closing this as it is related to the [esp-idf-cxx](https://github.com/espressif/esp-idf-cxx). Feel free to open a PR there. Thanks.

closed 2025-01-02 by KonstantinKondrashov