espressif/esp-idf #15044: ESP-MQTT Configuration not present in menuconfig

MarcelloDelDino, 2024-12-17:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

Hi Team, 

I am trying to implement the esp-iot-aws reference example working for my existing project.
https://github.com/FreeRTOS/iot-reference-esp32
That project worked alredy fine on my Board.

Unfortunately, it is crashing with my existing project and the implemented blocks from the reference.
By comparing my sdkconfig file with the one from the reference example I saw that "ESP-MQTT Configuration" is missing.
The rest is identical (Except the Thing Name obviously).

That configuration will not be shown on idf.py menuconfig -> Component config.
How can I activate that?

Best regards,
Marcel

ESP-IDF v5.3.1-dirty
VSCode 1.96.0
MacBook Air M2

nopnop2002, 2024-12-18:
I build [this](https://github.com/espressif/esp-idf/tree/master/examples/protocols/mqtt/ws) on esp-idf v5.3.2-1.

Here is ESP-MQTT Configuration.
[image]

MarcelloDelDino, 2024-12-19:
I tried that already, but they will not be shown.
[image]

nopnop2002, 2024-12-19:
Do you have this file in your environment?
```
$ ls esp-idf/components/mqtt/esp-mqtt/Kconfig
esp-idf/components/mqtt/esp-mqtt/Kconfig
```

MarcelloDelDino, 2024-12-19:
Yes, I do have this in my environment.

`% ls esp/v5.3.2/esp-idf/components/mqtt/esp-mqtt/Kconfig
esp/v5.3.2/esp-idf/components/mqtt/esp-mqtt/Kconfig`

nopnop2002, 2024-12-19:
This seems like a vscode specific issue.

MarcelloDelDino, 2024-12-21:
On the shared screenshots I was doing that. 
I entered idf.py menuconfig on the terminal.

MarcelloDelDino, 2024-12-21:
I managed now to get the Parameters into my project by manually adding the path ${IDF_PATH}/components/mqtt/esp-mqtt into the root CMakeLists.txt as EXTRA_COMPONENTS_DIR.
[image]
But I do not understand why I needed to do this, because this should be included by default.

nopnop2002, 2024-12-21:
>But I do not understand why I needed to do this, because this should be included by default.

Yes, that's correct.

What version of esp-idf did you install?

Did you disable [.$HOME/esp-idf/export.sh] before installing?

MarcelloDelDino, 2024-12-30:
I did that now by restarting the computer, deleting the IDF-PATH and tools.
After that installed esp-idf.
Additionally I created a brand new project and inserted all my necessary project files to the new project.
But I did not add ${IDF_PATH}/components/mqtt/esp-mqtt into the root CMakeLists.txt as EXTRA_COMPONENTS_DIR.
Still the same issue.

I created another project as well, which has nothing to do with the other one, there I can see the ESP-MQTT Configuration on idf.py menunconfig. Even though it is not using MQTT at all.
SO it has to be something I am doing at my project.

Alvin1Zhang, 2025-03-14:
@euripedesrocha Thanks for reporting, would you please help share if any further updates? Thanks.

closed 2025-04-07 by euripedesrocha

HanielPinheiro, 2025-07-14:
> I managed now to get the Parameters into my project by manually adding the path ${IDF_PATH}/components/mqtt/esp-mqtt into the root CMakeLists.txt as EXTRA_COMPONENTS_DIR. <img alt="Bildschirmfoto 2024-12-21 um 07 20 32" width="1058" src="https://private-user-images.githubusercontent.com/155311265/397908980-ba23ccd3-829f-4c72-ba93-66edd6af696e.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3NTI1MjgxMTgsIm5iZiI6MTc1MjUyNzgxOCwicGF0aCI6Ii8xNTUzMTEyNjUvMzk3OTA4OTgwLWJhMjNjY2QzLTgyOWYtNGM3Mi1iYTkzLTY2ZWRkNmFmNjk2ZS5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjUwNzE0JTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI1MDcxNFQyMTE2NThaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT0yNTA1Mjg5ZThkN2VmMjYwOGRjZTYyMDA2YWMwYjY2NmFlOTlkMTc0NjQ5MmMwZWFlNjE1ZjY2OTFlOTE3Mzc2JlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCJ9.QrjnXRg0ojcSmOYcLEvFJjEEoURvyYF8DUbxYnAIJ90"> But I do not understand why I needed to do this, because this should be included by default.

Really thanks, this helps too much.

If you or someone else have some idea of why this happens, please, update!