espressif/esp-idf #15046: Bootloader hook does not seem to be able to turn on GPIOs

ftab, 2024-12-17:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.5-dev-813-ge44a712970

### Espressif SoC revision.

ESP32-P4 v0.1

### Operating System used.

Linux

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

None

### Development Kit.

Custom Board

### Power Supply used.

External 3.3V

### What is the expected behavior?

I can turn on GPIOs in the bootloader hook

### What is the actual behavior?

I can't turn on the GPIOs; only the application can turn them on

Observed via oscilloscope & probe

### Steps to reproduce.

Add bootloader hook:

```c
#include "hal/gpio_hal.h"

/* Function used to tell the linker to include this file
 * with all its symbols.
 */
void bootloader_hooks_include(void)
{
}

void bootloader_after_init(void)
{
    gpio_ll_output_enable(&GPIO, GPIO_NUM_4);
    gpio_ll_output_enable(&GPIO, GPIO_NUM_7);
    gpio_ll_set_level(&GPIO, GPIO_NUM_4, 1);
    gpio_ll_set_level(&GPIO, GPIO_NUM_7, 1);
    gpio_ll_hold_en(&GPIO, GPIO_NUM_4);
    gpio_ll_hold_en(&GPIO, GPIO_NUM_7);
}
```

In application code:
```c
    gpio_config_t io_conf = {
        .pin_bit_mask = (1ULL << GPIO_NUM_7),
        .mode = GPIO_MODE_OUTPUT,
        .pull_up_en = GPIO_PULLUP_DISABLE,
        .pull_down_en = GPIO_PULLDOWN_DISABLE,
        .intr_type = GPIO_INTR_DISABLE,
        .hys_ctrl_mode = GPIO_HYS_SOFT_DISABLE,
    };
    gpio_config(&io_conf);
    gpio_set_level(GPIO_NUM_7, 1);
```

Start up board.

Only GPIO 7 is high ~300ms after power on.

GPIO 4 is low the entire time.

songruo, 2024-12-18:
@ftab Please try to call
```
gpio_ll_func_sel(&GPIO, GPIO_NUM_4, PIN_FUNC_GPIO);
gpio_ll_func_sel(&GPIO, GPIO_NUM_7, PIN_FUNC_GPIO);
```
as well in the bootloader hook.

SoucheSouche, 2024-12-18:
Hello @songruo,

Thanks for reporting this issue. I tried running the example "bootloader_hooks" (https://github.com/espressif/esp-idf/tree/master/examples/custom_bootloader/bootloader_hooks) on esp32p4 and it the hook is called properly on the target.

Could you try double checking your project against the example to make sure you did not forget anything while linking your hook.c file ?

Also, if you still don't find anything after checking the example, could you send me a link to your project or the content of your CMakefiles, so I can take a look at it?

ftab, 2024-12-18:
@songruo that was the trick. Thank you! 🎉 

@SoucheSouche I was able to verify the bootloader hooks were installed by sending a log in bootloader_after_init(), but I was just missing gpio_ll_func_sel as @songruo pointed out. Thank you for jumping in and taking a look! 🙇

closed 2024-12-18 by ftab