espressif/esp-idf #15047: I2S audio demo - ESP32-P4-NANO

Ncouch64, 2024-12-17:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

Hello, I am creating an issue post about this because it seems the only related issues are long closed. I am attempting to test out the speaker function of the new P4 kit, but when running this example I am getting no output.

`
I (174110) i2s_es8311: [music] i2s music played, 640000 bytes are written.
I (185030) i2s_es8311: [music] i2s music played, 640000 bytes are written.
I (195930) i2s_es8311: [music] i2s music played, 640000 bytes are written.
I (206850) i2s_es8311: [music] i2s music played, 640000 bytes are written.
I (217770) i2s_es8311: [music] i2s music played, 640000 bytes are written.
I (228680) i2s_es8311: [music] i2s music played, 640000 bytes are written.
I (239600) i2s_es8311: [music] i2s music played, 640000 bytes are written.
I (250520) i2s_es8311: [music] i2s music played, 640000 bytes are written.
I (261420) i2s_es8311: [music] i2s music played, 640000 bytes are written.
I (272340) i2s_es8311: [music] i2s music played, 640000 bytes are written.
I (283260) i2s_es8311: [music] i2s music played, 640000 bytes are written.
`

suda-morris, 2024-12-18:
Have you set the correct GPIO according to your P4-Nano board? https://www.waveshare.net/w/upload/9/97/ESP32-P4-NANO.pdf

`ESP32P4 I2S_DO_IO` is connected `ES8311 PIN9-SDIN`, from the schematic it's GPIO9, so please set it correctly in the example_config.h.

Please note, esp-idf example doesn't contain any BSP package, there's no guarantee that it can run out of the box for all boards on the market. We expect the user will read the README before trying the example.

L-KAYA, 2024-12-18:
This issue probably caused by the PA (Power Amplifier) is not enabled. According to the schematic from the link above:

[image]

The `PA_CTRL` (which is connected to `GPIO53`) need to be at high level to enable the PA. You can either set `GPIO53` to high in the example, or just connect `GPIO53` to VCC with a jump wire.

BTW, P4-EV board also uses `GPIO53` as the `PA_CTRL`, PA is not enabled by default on these P4 board, please take care when using P4 dev kit~

Back to the example, the example can't ensure whether the PA on board is enabled by default or connect to a particular GPIO. So, the PA enable step won't be added into the example. But I think we can add some hints since this issue is pretty common for P4 boards

Ncouch64, 2024-12-18:
I appreciate the assistance here, still new to the ESP world so figuring things out as I go along. I will try what has been posted here and provide an update. Thank you!

closed 2024-12-20 by espressif-bot

Ncouch64, 2025-01-06:
Sorry for the much delayed response and re-opening this issue. I have been trying to get this working still following the recommendations here. I have added the code for setting PA_CTRL to the correct pin and also tried swapping the I2S_DO_IO and I2S_DI_IO pins. I still haven't been able to get this working, is it because the PA_CTRL must be explicitly set to "high"? If so how is this done? Once again apologize, trying to learn.

Ncouch64, 2025-01-06:
Hmm, I would say this has reached the point of being rather bizarre. I found the examples from the wiki with this board that has a revised audio example based on the i2s_es8311 example code that supposedly has the correct setup. Even when using this code I am still unable to get any audio from the board. I am really not sure what the issue could be.

`#elif CONFIG_IDF_TARGET_ESP32P4
#define I2C_SCL_IO      (GPIO_NUM_8)
#define I2C_SDA_IO      (GPIO_NUM_7)
#define GPIO_OUTPUT_PA  (GPIO_NUM_53)
#else
#define I2C_SCL_IO      (GPIO_NUM_6)
#define I2C_SDA_IO      (GPIO_NUM_7)
#endif

/* I2S port and GPIOs */
#define I2S_NUM         (0)
#if CONFIG_IDF_TARGET_ESP32P4
#define I2S_MCK_IO      (GPIO_NUM_13)
#define I2S_BCK_IO      (GPIO_NUM_12)
#define I2S_WS_IO       (GPIO_NUM_10)
#define I2S_DO_IO       (GPIO_NUM_9) // 9
#define I2S_DI_IO       (GPIO_NUM_11) // 11
#else`

L-KAYA, 2025-03-04:
If all the connections and configurations are correct (including the I2C pull-up), it looks more like a hardware issue. 

You can also try other audio examples like the [player examples in ADF](https://github.com/espressif/esp-adf/tree/master/examples/player) to check if only this IDF example can't work. If all player examples have no sound, it is probably a hardware issue then. You can check as follow:

Monitor the I2C signal to check if the configurations are all correct(sometimes I2C will be disturbed while using a mismatched pull-up resistance, see [the requirement of pull-up resistance](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/peripherals/i2c.html#introduction)), and then monitor the I2S signals on the line to check if they are output correctly. Besides, please also check the power of PA in case it is burnt/dead. Lastly use a new speaker to check if the speaker is broken.