espressif/esp-idf #15049: RMT5.1 Bug - Tearing down and re-instantiating

zackees, 2024-12-18:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.1

### Espressif SoC revision.

ESP-WROOM-32-DA

### Operating System used.

Windows

### How did you build your project?

Command line with Make

### If you are using Windows, please specify command line type.

None

### Development Kit.

ESP-WROOM-32-DA

### Power Supply used.

USB

### What is the expected behavior?

Hi there, I'm the acting main contributor for FastLED.

We recently upgraded to RMT5.

Our users are report first pixel is always green, some of the report blue. It depends don't he color order of the LED chipset.

The good news is that we found the issue, issued a work-around and are applying it in the next release. I want to bring this to your attention.

# Cause

In RMT 4.x, we had the ability to move RMT devices from one pin to another. This allows us to service lots of strips when the number of strips >= number of RMT devices.

In RMT5, this feature went away. We thought we could be clever by simply tearing down the RMT device every frame and then re-instantiating them when needed. This either happens on the next draw call or on the same draw call when the number of strips > number of RMT devices.

This worked great in tests so we shipped it.

However, a minority of our users are reporting that with the ESP-WROOM-32-DA, and related chipsets, the first led will always stay green.

Someone with an oscilloscope pointed out that there is an unexpected blip happening:

[image]

The fix on our end was to disable RMT recycling. Once they get allocated, that's it, until the strip itself is torn down, if ever.

This isn't a huge deal for us because RMT recycling looks awful, the leds update in a staggered fashion. So we are just going to disable it completely. However, you may be interested to know that there is a bug lurking somewhere in your firmware that might be manifesting on the first usage of RMT5.

https://github.com/FastLED/FastLED/issues/1774
https://github.com/FastLED/FastLED/issues/1761
https://github.com/FastLED/FastLED/issues/1640

### What is the actual behavior?

Green Light

### Steps to reproduce.

Keep in mind, i was NOT ever able to reproduce this myself.

Use FastLED 3.9.0 - 3.9.5 (this is when we switched to RMT5)

Use FireBeetle esp32-e

```
#define FASTLED_ALLOW_INTERRUPTS 0
#include <FastLED.h>

#define LED_PIN     15
#define BRIGHTNESS  60
#define LED_TYPE    WS2812B
#define COLOR_ORDER GRB
#define COLOR_ORDER GRB
#define NUM_LEDS 60
CRGBArray<NUM_LEDS> leds;

void setup()
{
  Serial.begin(115200); // Enable serial output
  FastLED.addLeds<LED_TYPE, LED_PIN, COLOR_ORDER>(leds, NUM_LEDS);
  FastLED.setBrightness(BRIGHTNESS);
  leds.fill_solid(0x000000); // Clear all leds
  FastLED.show();
}

void loop()
{
  delay(1000);
  for (uint8_t i = 0; i < NUM_LEDS; ++i)
  {
      leds[i].setRGB(0, 0, 255);
      if (i > 1)
      {
        leds[i-1].setRGB(0, 0, 0);
      }
      FastLED.show();
      delay(300);
  }

  leds.fill_solid(0x0000ff); // Clear all leds
  FastLED.show();
}
```

suda-morris, 2024-12-18:
Hi @zackees Thanks for the report. From your posted oscilloscope picture, it looks like after RMT driver destroy the channel object, the GPIO level goes to high, right?

zackees, 2024-12-18:
Yes, that appears to be the case.

We tear down then IMMEDIATELY re-instantiate.

We can do this because our version of the led-strip component has a pointer instead of an inlined array for the pixel data, so we just re-attach the new pixel buffer on re-instantiation.

suda-morris, 2024-12-18:
I think I can reproduce the "issue" and a simple "workaround" is to utilize the gpio pad's pull-up/down resistor.

```c
TEST_CASE("rmt bytes encoder", "[rmt]")
{
    rmt_tx_channel_config_t tx_channel_cfg = {
        .mem_block_symbols = SOC_RMT_MEM_WORDS_PER_CHANNEL,
        .clk_src = RMT_CLK_SRC_DEFAULT,
        .resolution_hz = 1000000, // 1MHz, 1 tick = 1us
        .trans_queue_depth = 4,
        .gpio_num = TEST_RMT_GPIO_NUM_A,
        .intr_priority = 3
    };
    printf("install tx channel\r\n");
    rmt_channel_handle_t tx_channel = NULL;
    TEST_ESP_OK(rmt_new_tx_channel(&tx_channel_cfg, &tx_channel));
    printf("install bytes encoder\r\n");
    rmt_encoder_handle_t bytes_encoder = NULL;
    rmt_bytes_encoder_config_t bytes_enc_config = {
        .bit0 = {
            .level0 = 1,
            .duration0 = 3, // 3us
            .level1 = 0,
            .duration1 = 9, // 9us
        },
        .bit1 = {
            .level0 = 1,
            .duration0 = 9, // 9us
            .level1 = 0,
            .duration1 = 3, // 3us
        },
    };
    TEST_ESP_OK(rmt_new_bytes_encoder(&bytes_enc_config, &bytes_encoder));
    printf("enable tx channel\r\n");
    TEST_ESP_OK(rmt_enable(tx_channel));

    printf("start transaction\r\n");
    rmt_transmit_config_t transmit_config = {
        .loop_count = 0, // no loop
    };
    TEST_ESP_OK(rmt_transmit(tx_channel, bytes_encoder, (uint8_t[]) {
        0x00, 0x01, 0x02, 0x03, 0x04, 0x05
    }, 6, &transmit_config));
    // adding extra delay here for visualizing
    vTaskDelay(pdMS_TO_TICKS(500));
    TEST_ESP_OK(rmt_transmit(tx_channel, bytes_encoder, (uint8_t[]) {
        0x00, 0x01, 0x02, 0x03, 0x04, 0x05, 0x06
    }, 7, &transmit_config));
    vTaskDelay(pdMS_TO_TICKS(500));
    TEST_ESP_OK(rmt_transmit(tx_channel, bytes_encoder, (uint8_t[]) {
        0x00, 0x01, 0x02, 0x03, 0x04
    }, 5, &transmit_config));
    vTaskDelay(pdMS_TO_TICKS(500));

    printf("disable tx channel\r\n");
    TEST_ESP_OK(rmt_disable(tx_channel));

    // If you want to keep the IO level after unintall the RMT channel, a workaround is to set the pull up/down register here
    // because when we uninstall the RMT channel, we will also disable the GPIO output
    gpio_pullup_dis(TEST_RMT_GPIO_NUM_A);
    gpio_pulldown_en(TEST_RMT_GPIO_NUM_A);

    printf("remove tx channel\r\n");
    TEST_ESP_OK(rmt_del_channel(tx_channel));

    vTaskDelay(pdMS_TO_TICKS(500));

    printf("install tx channel again\r\n");
    TEST_ESP_OK(rmt_new_tx_channel(&tx_channel_cfg, &tx_channel));
    printf("enable tx channel\r\n");
    TEST_ESP_OK(rmt_enable(tx_channel));

    printf("start transaction\r\n");
    TEST_ESP_OK(rmt_transmit(tx_channel, bytes_encoder, (uint8_t[]) {
        0x00, 0x01, 0x02, 0x03, 0x04, 0x05
    }, 6, &transmit_config));
    // adding extra delay here for visualizing
    vTaskDelay(pdMS_TO_TICKS(500));

    TEST_ESP_OK(rmt_disable(tx_channel));
    printf("remove tx channel and encoder\r\n");
    TEST_ESP_OK(rmt_del_channel(tx_channel));
    TEST_ESP_OK(rmt_del_encoder(bytes_encoder));
}
```

The reason is, when we uninstall a RMT channel ,we will also disable the corresponding GPIO output, so its level is undetermined. It can be affected by external pull-up/down resistor as well.

zackees, 2024-12-18:
Does this look like a floating pin?

[image]

tikiss, 2024-12-23:
Hi, I made the measurement with oscilloscope, let me clarify the measurement, the circuit has no external pull-up, I am using and [ESP32-C6-Zero](https://www.waveshare.com/wiki/ESP32-C6-Zero) board, simple connected the WS2812b LED string on the GP0 (tried with different GPIOs, same behaviour).
On the previous oscilloscope screenshot you see a normal LED refresh cycle, but during board power-up it looks like this:
[image]
As you can see, the very first LED data package is not starting with initial HIGH, but every other cycle is starting with 300us HIGH period. (This measurement was done with very quick refresh cycle to able to see two data package in one oscillator screen)

zackees, 2024-12-23:
Just FYI, we are trying a test with the RMT recycling re-enabled. Since I can't reproduce the issue locally, one of our users is going to try this tonight (EU timezone).

I've applied the suggested fix to enable input pulldown after rmt_disable.

I expect I'll have an update for you tomorrow.

zackees, 2024-12-23:
Hi there, my user got back to me and reported that setting the gpio to input pulldown does not work.

https://github.com/FastLED/FastLED/issues/1761#issuecomment-2560089124

You can ignore their comment about the rmt disable message popping up, that's our code issue trying to shoe horn the led-strip component into an async api.

suda-morris, 2024-12-24:
@zackees then could it be that there's an external pull-up resistor on that GPIO? because the pull-up/down resister internal the chip is weak.

Abasz, 2024-12-24:
https://wiki.dfrobot.com/FireBeetle_Board_ESP32_E_SKU_DFR0654 this is the board I use. It's the built in onboard RGB LED on the gpio 5.

I don't no if there is an external pull-up/down. But based on the schematic I was able to find I don't think so. Here I think gpio 5 is directly connected to the LED's data pin:

[image]

zackees, 2025-01-01:
It's happening to others too. Pulldown resistor fixes it.

https://www.reddit.com/r/FastLED/comments/1hq7b6m/data_pin_goes_high_impedance_before_transmit/