espressif/esp-idf #15050: need OTA single app mode for economize flash

shishirong, 2024-12-18:
### Is your feature request related to a problem?

2 app partition is too occupy flash space

### Describe the solution you'd like.

app request boot to bootloader
ota in bootloader 
if app data verify check success, set flag for jump to  app, otherwise stay in bootloader until success

igrr, 2024-12-19:
The exact mechanism you have described is probably difficult to realize. "Over the air" update typically requires Wi-Fi or Bluetooth stack to work. However, features such as Wi-Fi or BT don't work in the bootloader environment. If your "OTA" process is over some physical interface such as UART, then you can achieve it by implementing a customized bootloader.

Another approach which can help you reduce flash usage is differential OTA: https://github.com/espressif/idf-extra-components/tree/master/esp_delta_ota.

shishirong, 2024-12-19:
> The exact mechanism you have described is probably difficult to realize. "Over the air" update typically requires Wi-Fi or Bluetooth stack to work. However, features such as Wi-Fi or BT don't work in the bootloader environment.

Why can't add Wi-Fi or Bluetooth stack to bootloader?

igrr, 2024-12-19:
As it is now, Wi-Fi and BT stacks in IDF both have a number of dependencies: FreeRTOS, various other system components (esp_timer, esp_event, nvs_flash), wpa_supplicant and so on. Once you add together all those things, plus at least a minimal OTA implementation, the binary size is already above 500-700kB. Loading such a huge 2nd stage bootloader would take a long time for a regular (non-OTA) boot.

You can consider the following solution as well:
- Have an "OTA updater" app in the "factory" partition. This app in "ota0" partition would only have the OTA functionality.
- When the main app needs to perform an OTA, it sets boot partition to "factory" and triggers a reset.
- The 2nd stage bootloader boots into the "factory" app.
- The "factory" app performs the update of the "ota0" partition, sets the boot partition to "ota0" and reboots.

The downside of this approach is that your factory application doesn't get OTA-updated itself. This could make it vulnerable to newly discovered security issues in Wi-Fi, TLS, and so on. Therefore ideally you should also include OTA functionality into your main app, so that it can update the "factory" app when necessary. However, once you do this, it's not clear if there will be a significant amount of flash size savings.

shishirong, 2024-12-20:
Thank you for your patient reply.

closed 2024-12-20 by shishirong