espressif/esp-idf #15051: PN532 module panics

ImDevinC, 2024-12-18:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

5.3.1

### Espressif SoC revision.

ESP32-D0WD-V3 (revision v3.1)

### Operating System used.

Linux

### How did you build your project?

Other (please specify in More Information)

### If you are using Windows, please specify command line type.

None

### Development Kit.

DOIT ESP32 DEVKIT V1

### Power Supply used.

USB

### What is the expected behavior?

Module should startup correctly

### What is the actual behavior?

Module panics and prevents booting regularly

### Steps to reproduce.

Create the following yaml file for use with `esphome` and run `esphome run <file.yaml> --device <USB_DEVICE>`

```yaml
esphome:
  name: pn532-sample-esphome
esp32:
  board: esp32doit-devkit-v1
  framework:
    type: esp-idf
    version: 5.3.1
    platform_version: 6.9.0
spi:
  - clk_pin: GPIO14
    miso_pin: GPIO12
    mosi_pin: GPIO13
    interface: spi2
    id: SPI2
pn532_spi:
  cs_pin: GPIO16
  update_interval: 750ms
  spi_id: SPI2
  data_rate: 200kHz
logger:
  level: VERBOSE
```

[... 76 lines cut ...]
WARNING Decoded 0x4008b049: spi_ll_master_set_line_mode at /home/devin/.platformio/packages/framework-espidf/components/hal/esp32/include/hal/spi_ll.h:550
 (inlined by) spi_hal_setup_trans at /home/devin/.platformio/packages/framework-espidf/components/hal/spi_hal_iram.c:46
WARNING Decoded 0x400818bf: spi_new_trans at /home/devin/.platformio/packages/framework-espidf/components/esp_driver_spi/src/gpspi/spi_master.c:773
WARNING Decoded 0x40081c91: spi_device_polling_start at /home/devin/.platformio/packages/framework-espidf/components/esp_driver_spi/src/gpspi/spi_master.c:1376
WARNING Decoded 0x400d6885: esphome::spi::SPIDelegateHw::transfer(unsigned char const*, unsigned char*, unsigned int) at /home/devin/Projects/OpenSpool/firmware/.esphome/build/pn532-sample-esphome/src/esphome/comp
onents/spi/spi_esp_idf.cpp:76
WARNING Decoded 0x400f4fe7: esphome::spi::SPIDelegateHw::write_array(unsigned char const*, unsigned int) at /home/devin/Projects/OpenSpool/firmware/.esphome/build/pn532-sample-esphome/src/esphome/components/spi/sp
i_esp_idf.cpp:171
WARNING Decoded 0x400f4df5: esphome::spi::SPIDevice<(esphome::spi::SPIBitOrder)0, (esphome::spi::SPIClockPolarity)0, (esphome::spi::SPIClockPhase)0, (esphome::spi::SPIDataRate)1000000>::write_byte(unsigned char) a
t /home/devin/Projects/OpenSpool/firmware/.esphome/build/pn532-sample-esphome/src/esphome/components/spi/spi.h:440
WARNING Decoded 0x400d5d09: esphome::pn532_spi::PN532Spi::write_data(std::vector<unsigned char, std::allocator<unsigned char> > const&) at /home/devin/Projects/OpenSpool/firmware/.esphome/build/pn532-sample-esphom
e/src/esphome/components/pn532_spi/pn532_spi.cpp:36
WARNING Decoded 0x400d4389: esphome::pn532::PN532::write_command_(std::vector<unsigned char, std::allocator<unsigned char> > const&) at /home/devin/Projects/OpenSpool/firmware/.esphome/build/pn532-sample-esphome/s
rc/esphome/components/pn532/pn532.cpp:278
WARNING Decoded 0x400d45b1: esphome::pn532::PN532::setup() at /home/devin/Projects/OpenSpool/firmware/.esphome/build/pn532-sample-esphome/src/esphome/components/pn532/pn532.cpp:21 (discriminator 2)
WARNING Decoded 0x400d601d: esphome::pn532_spi::PN532Spi::setup() at /home/devin/Projects/OpenSpool/firmware/.esphome/build/pn532-sample-esphome/src/esphome/components/pn532_spi/pn532_spi.cpp:21
WARNING Decoded 0x400d7da8: esphome::PollingComponent::call_setup() at /home/devin/Projects/OpenSpool/firmware/.esphome/build/pn532-sample-esphome/src/esphome/core/component.cpp:212
WARNING Decoded 0x400f511f: esphome::Component::call() at /home/devin/Projects/OpenSpool/firmware/.esphome/build/pn532-sample-esphome/src/esphome/core/component.cpp:104
WARNING Decoded 0x400d79fe: esphome::Application::setup() at /home/devin/Projects/OpenSpool/firmware/.esphome/build/pn532-sample-esphome/src/esphome/core/application.cpp:38
```

### More Information.

Building this as part of esphome, wasn't sure if this should be filed here, with platformio, or with esphome.
However I can say that changing the framework from `esp-idf` to `arduino` in the esphome config allows the pn532 module to load properly.

wanckl, 2024-12-18:
@ImDevinC  
Not sure if issue from share interrupt somewhere in esp_home, but I have a patch for share interrupt, you can have a try with patch or replace your `esp-idf\components\esp_driver_spi\src\gpspi\spi_master.c` file in esp-idf repo with patched one.

[spi_master_share_intr.patch](https://github.com/user-attachments/files/18181901/spi_master_share_intr.patch)
[spi_master.c.txt](https://github.com/user-attachments/files/18182015/spi_master.c.txt)

Waitting for your reply even if fixed or not, thank you.

ImDevinC, 2024-12-18:
It looks like platform.io framework `6.9.0` doesn't support `5.3.2` or above yet as I get a build error with this change (same error if I try to change to using `5.3.2` without your patch). I'll see if I can figure out the best way to test this module using raw esp-idf without esphome and report back. Appreciate the quick reply!

Alvin1Zhang, 2025-01-16:
@ImDevinC Thanks for reporting, will you please help share if any updates for the issue? Thanks.

ImDevinC, 2025-01-17:
Can do, unfortunately I've been busy on other projects and got distracted. I'll try to revisit this again sometime in the next week

Alvin1Zhang, 2025-03-06:
@ImDevinC Reminder of the updates, thanks.

ginkgm, 2025-09-24:
Could you provide the code you use to reproduce the issue? @mcpat-it