espressif/esp-idf #15052: Some devices disconnects from the chip AP when scanning for networks while BLE is enabled

lucacresp, 2024-12-18:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.0.7 and v5.3.2 

### Espressif SoC revision.

ESP32-S3 (QFN56) (revision v0.2)

### Operating System used.

Windows

### How did you build your project?

VS Code IDE

### If you are using Windows, please specify command line type.

PowerShell

### Development Kit.

ESP32-S3-DevKitC-1 v1.0 and also on a Custom Board

### Power Supply used.

USB

### What is the expected behavior?

I expected it to scan the avaiable networks without disconnecting the device connected to the chip's AP while BLE is enabled.

### What is the actual behavior?

Instead it disconnects certain devices when calling "esp_wifi_scan_start()", in my case it does this to:

- tp-link AC600 Mini Wireless USB Adapter Archer T2U V3

- tp-link AC1300 High Gain Wireless MU-MIMO USB Adaprter Archer T4U v3.20

It work as intended when connecting to the chip's AP with different android mobile devices.

### Steps to reproduce.

1. start the wifi in AP
2. start BLE
3. connect device to the AP
4. periodically scan for avaiable networks

### Debug Logs.

```plain
I (180053) wi-fi: device connected to AP
I (180073) esp_netif_lwip: DHCP server assigned IP to a client, IP is: 192.168.4.2
I (185293) wi-fi: Found 2 networks
I (188023) wi-fi: device disconnected to AP
I (188023) wifi:starting SA query procedure with STA(5c:62:8b:4b:a2:8f)
I (188033) wifi:Send SA Query req with transaction id ec14
E (188033) wifi:CCMP replay detected: A1=48:27:e2:ca:cc:a1 A2=5c:62:8b:4b:a2:8f PN=0, RSC=7 seq=9280
I (188223) wi-fi: device disconnected to AP
I (188233) wifi:Send SA Query req with transaction id ec15
E (188233) wifi:CCMP replay detected: A1=48:27:e2:ca:cc:a1 A2=5c:62:8b:4b:a2:8f PN=0, RSC=7 seq=9328
I (188333) wi-fi: device disconnected to AP
I (188433) wifi:Send SA Query req with transaction id ec16
E (188433) wifi:CCMP replay detected: A1=48:27:e2:ca:cc:a1 A2=5c:62:8b:4b:a2:8f PN=0, RSC=7 seq=9376
I (188433) wi-fi: device disconnected to AP
I (188633) wifi:Send SA Query req with transaction id ec17
I (188833) wifi:Send SA Query req with transaction id ec18
I (189033) wifi:Send SA Query req with transaction id ec19
I (189053) wifi:STA not responded to 6 SA Query attempts, Reset connection sending disassoc
I (189053) wifi:station: 5c:62:8b:4b:a2:8f leave, AID = 1, reason = 209, bss_flags is 756835, bss:0x3fcc9b2c
I (189063) wifi:new:<1,0>, old:<1,1>, ap:<1,1>, sta:<0,0>, prof:1, snd_ch_cfg:0x0
I (189063) wi-fi: device disconnected to AP
```

### More Information.

This occurs on both Windows 11 and Windows 10.
I use VS Code with the ESP-IDF extension, which i used to install my ESP-IDF versions.
In both the ESP-IDF versions i listed the debug logs are similar but not always identical, the general behavior also seems to vary sometimes: sometimes device fails to connected to the AP, other times one scan goes well but the second one disconnects the device, etc....
Also i remember it working fine on ESP-IDF v5.0.4 but i can really confirm this.

Espressif-liuuuu, 2025-08-06:
Hi @lucacresp , thanks for your report and sorry for late reply.  
Refering to https://docs.espressif.com/projects/esp-idf/en/latest/esp32s3/api-guides/coexist.html, the performance of softap mode with coexistence is unstable, especially with the scenarios BLE works much time.
I will try to reproduce your issue but here there are two questions at first:

1) What did BLE work in your case, BLE adv/scan or just controller enabled?
2) Did ESP32S3 do the periodically scan or the tplink network adapter?

lucacresp, 2025-08-06:
Hi, my BLE only advertise itself, the ESP32S3 periodically scans for avaiable APs while advertising itself as an available AP to other devices; when a device connects to the ESP32S3's AP, the moment the periodic scan for APs is triggered said device is disconected.