espressif/esp-idf #15054: SDK 5.3.2 Bug in RMT while LOG is disable cause DESYNC and weird behavior on use

filzek, 2024-12-18:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

RMT driver can be real problematic if the LOG is disable in SDKCONFIG and will not work as intent.
Only after turn the log on in SDKCONFIG and se it to none in APP_MAIN it get back to work correct.

RMT legacy driver in SDK 5.3.2.

commit 12938e511e1f19cdb29e8490f2c9e30358c840bb (HEAD -> release/v5.3, origin/release/v5.3)
Merge: de65ee55a7 6d18437d56
Author: David ─îerm├ík <david.cermak@espressif.com>
Date:   Mon Dec 16 22:20:59 2024 +0800

    Merge branch 'lwip/dns_clear_cache_fix_v5.3' into 'release/v5.3'

    fix(lwip): Fixed active DNS entries before clearing cache(v5.3)

    See merge request espressif/esp-idf!35003
    
    

So, we need to solve the problem as this could be a very problematic in field.

suda-morris, 2024-12-19:
@filzek could you please provide a minimal project (only contain the RMT specific code) that can reproduce the "The timming amond spaces in RMT act completely odd and also sometime the RMT even do not transmit anything."? And  your sdkconfig. Thanks.

suda-morris, 2025-01-14:
After I turn off the log in menuconfig [image]
Run the rmt test app: https://github.com/espressif/esp-idf/tree/master/components/esp_driver_rmt/test_apps/rmt

The result is just as expected. No issue found.

closed 2025-01-14 by suda-morris