espressif/esp-idf #15055: 测试自己画的ESP32S3板子，读取到的wifi信号值是-4、-5，这样，感觉信号很好，但传输丢包严重

beautifulpeer, 2024-12-19:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

1. 测试自己画的ESP32S3板子，读取到的wifi信号值是-4、-5，这样，感觉信号很好，但传输丢包严重
2. 接收端使用的是开发板，之前画的测试板没有问题
3. 我们是根据这个rssi来判断信号质量，但根据读取回来的信号值很强，但丢包严重，这是为什么？
@esp-zhp

esp-zhp, 2024-12-19:
@beautifulpeer 这看起来是 WI-FI 的问题，将有其他更专业的同事为您解决。

beautifulpeer, 2024-12-19:
收到，请帮我即可转接，感谢！！！

MaxwellAlan, 2024-12-19:
@beautifulpeer 

你好，IDF GitHub 上主要关注软件问题，从你的描述中看起来更像是硬件电路相关的问题，
与硬件相关的问题可以在官网的 硬件设计 审阅 中寻求帮助 https://www.espressif.com/zh-hans/support/services#hardware-rf

另外：由于 WIFI 的收发包存在不同的调制方式以及不同的速率，RSSI 好只能说明 1M 速率的接收看起来还不错，但并不代表其他速率能够正常接收，即通信可能存在丢包。如果想要更详细的分析，可以提供 WIFI 的空口抓包我们看下。

谢谢。

beautifulpeer, 2024-12-19:
但是我们的速率也就在0.4Mbit/s，还没有到达1M

MaxwellAlan, 2024-12-19:
1M 是 物理层 的速率，对应到应用层的吞吐 差不多就是 0.5M上限

closed 2024-12-24 by Alvin1Zhang