espressif/esp-idf #15056: console Linux target does not work for 5.3.2, but work in 5.3.1

joyhope, 2024-12-19:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.3.2

### Espressif SoC revision.

linux

### Operating System used.

Linux

### How did you build your project?

Command line with Make

### If you are using Windows, please specify command line type.

None

### Development Kit.

linux

### Power Supply used.

USB

### What is the expected behavior?

I make a little change for system/console/basic/ for support linux  target.  The same application, using v5.3.1 SDK, the console works.  waiting for user input, but with 5.3.2 SDK, the console constant work, output "Unrecognized command
". It seem console not block for user input. 

### What is the actual behavior?

Expect: 
``
Type 'help' to get the list of commands.
Use UP/DOWN arrows to navigate through command history.
Press TAB when typing command name to auto-complete.
linux> help
free 
  Get the current size of free heap memory

heap 
  Get minimum size of free heap memory that was available during program
  execution
```

The problem v5.3.2 behavior

auto execute output. 
```
linux> 
Unrecognized command
```

### Steps to reproduce.

system/console/basic 

root cmake file

```
if("${IDF_TARGET}" STREQUAL "linux")
    set(COMPONENTS main)
endif()
```

cmakefile in main.
```
set(requires "")
idf_build_get_property(target IDF_TARGET)

if(${target} STREQUAL "linux")
    list(APPEND requires esp_stubs esp_event esp-tls esp_http_client console nvs_flash)
endif()

idf_component_register(SRCS "console_example_main.c"
                    INCLUDE_DIRS "."
                    REQUIRES ${requires})
```

Some minor compile problem in console/advanced/ , easy to process. 

### More Information.

It is good to support linux target.

ESP-Marius, 2024-12-19:
The console implementation on linux has a known issue, the `read()` call can be interrupted by the signals, e.g. what the freertos implementation uses for sheduling different threads.

I don't think there is anything different between v5.3.1 and v5.3.2 in this regard, the fact that it worked is probably related to another difference in your project.

Right now it cannot be reliably used on linux, finding a workaround for this issue is in our backlog.

joyhope, 2024-12-19:
This is a strange result.  I compare the console, esp_vfs_console with two
versions.  Nothing changed.  But in my test case, it works perfectly on
v5.3.1.

 The read is not blocked.

joyhope, 2024-12-20:
I attached the problem project.  If the target is Linux, there is a simple
compile error in the components/xxx_cmd. It is easy to process.   I am very
curious which kind of bug will cause this problem.

joyhope, 2025-03-30:
the problem is not solved, 5.4 is still the same problem.  

The default idf.py --preview  set-target linx will cause config error, and there are many compile related issues.

joyhope, 2025-03-31:
Is it possible let whole samples pass linux target compile?   otherwise, if user want to fix, there are too much work for fixing the compile problems.

joyhope, 2026-03-04:
From the v5.3.2 to 5.5.3, the default console example could not compile and run.  Hope the problem could be solve in the standard release.