espressif/esp-idf #15058: fish: unable to run `export.fish` due to python error

xobs, 2024-12-19:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.5-dev-847-gcb3ac7429c

### Operating System used.

macOS

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

None

### What is the expected behavior?

I should be able to run `. export.fish`

### What is the actual behavior?

```fish
❯ . export.fish
Checking "python3" ...
Python 3.13.0
python3 has been detected
env: python: No such file or directory
❯
```

### Steps to reproduce.

1. Run `install.fish`
2. Run `. export.fish`

xobs, 2024-12-19:
This patch fixes the issue:

```patch
diff --git a/export.fish b/export.fish
index e2d9600021..5114f33f59 100644
--- a/export.fish
+++ b/export.fish
@@ -25,5 +25,5 @@ end

 source "$idf_path"/tools/detect_python.fish

-eval ("$idf_path"/tools/activate.py --export)
+eval ("$ESP_PYTHON" "$idf_path"/tools/activate.py --export)
 set -e idf_path
```

mfialaf, 2024-12-19:
Hi @xobs,
thank you for noticing this. I'll take a look and follow up soon.

mickeyl, 2025-01-05:
It's a bit unfortunate that this hasn't been included in the 5.4 release…

mfialaf, 2025-01-06:
Hi @mickeyl, this bugfix was backported and should be in branch `release/v5.4` soon. Or in release 5.4.1.

closed 2025-01-09 by espressif-bot