espressif/esp-idf #15060: How to solve the problem of insufficient interrupt resource conflicts

travisbit, 2024-12-19:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v3.2-dev-306-gbeb3611ca

### Espressif SoC revision.

ESP32S3

### Operating System used.

Windows

### How did you build your project?

Eclipse IDE

### If you are using Windows, please specify command line type.

CMD

### Development Kit.

ESP IDF

### Power Supply used.

USB

### What is the expected behavior?

The program prompts that interrupt resources are running low and need to be freed up.
But from the interrupt resources of printing, it seems that all interrupts that have already been allocated must be allocated interrupts, otherwise the system may not run. Please provide a solution

### What is the actual behavior?

prompt the following info
No free interrupt inputs for SPI3 interrupt

### Steps to reproduce.

I have printed out the resources allocated for interrupts, and the allocation is as follows. I still need 2 interrupt resources for other functions. How can I adjust the following resources to free up 2 interrupt resources

        intr_alloc: Connected src 39 to int 2 (cpu 0)		ETS_RTC_CORE_INTR_SOURCE
	intr_alloc: Connected src 59 to int 3 (cpu 0)		ETS_SYSTIMER_TARGET2_INTR_SOURCE
	intr_alloc: Connected src 79 to int 9 (cpu 0)		ETS_FROM_CPU_INTR0_SOURCE
	intr_alloc: Connected src 57 to int 12 (cpu 0)		ETS_SYSTIMER_TARGET0_INTR_SOURCE
	intr_alloc: Connected src 80 to int 2 (cpu 1)		ETS_FROM_CPU_INTR1_SOURCE
	intr_alloc: Connected src 58 to int 3 (cpu 1)		ETS_SYSTIMER_TARGET1_INTR_SOURCE
	intr_alloc: Connected src 52 to int 13 (cpu 0)		ETS_TG0_WDT_LEVEL_INTR_SOURCE
	intr_alloc: Connected src 27 to int 17 (cpu 0)		ETS_UART0_INTR_SOURCE
	intr_alloc: Connected src 29 to int 18 (cpu 0)		ETS_UART2_INTR_SOURCE
	intr_alloc: Connected src 28 to int 19 (cpu 0)		ETS_UART1_INTR_SOURCE
	intr_alloc: Connected src 16 to int 20 (cpu 0)		ETS_GPIO_INTR_SOURCE
	intr_alloc: Connected src 21 to int 21 (cpu 0)		ETS_SPI2_INTR_SOURCE
	intr_alloc: Connected src 38 to int 23 (cpu 0)		ETS_USB_INTR_SOURCE

### Debug Logs.

```plain
19: I (1236) intr_alloc: Connected src 39 to int 2 (cpu 0)
22: I (1253) intr_alloc: Connected src 59 to int 3 (cpu 0)
27: I (1284) intr_alloc: Connected src 79 to int 9 (cpu 0)
28: I (1290) intr_alloc: Connected src 57 to int 12 (cpu 0)
29: I (1290) intr_alloc: Connected src 80 to int 2 (cpu 1)
30: I (1300) intr_alloc: Connected src 58 to int 3 (cpu 1)
33: I (1310) intr_alloc: Connected src 52 to int 13 (cpu 0)
37: I (1350) intr_alloc: Connected src 27 to int 17 (cpu 0)
41: I (1440) intr_alloc: Connected src 29 to int 18 (cpu 0)
42: I (1440) intr_alloc: Connected src 28 to int 19 (cpu 0)
59: I (1530) intr_alloc: Connected src 16 to int 20 (cpu 0)
60: I (1530) intr_alloc: Connected src 21 to int 21 (cpu 0)
81: I (2920) intr_alloc: Connected src 38 to int 23 (cpu 0)

I (9146) MQTT: rcv get_version cmd
I (9146) OTDR_IAP: OTDR version is: 0260
I (10136) example: Install panel IO
**E (10136) intr_alloc: No free interrupt inputs for SPI3 interrupt (flags 0x80E)**
E (10136) lcd_panel.io.spi: esp_lcd_new_panel_io_spi(87): adding spi device to bus failed
I (10146) gpio: GPIO[36]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 
ESP_ERROR_CHECK failed: esp_err_t 0x105 (ESP_ERR_NOT_FOUND) at 0x4200e507
```

travisbit, 2024-12-19:
The following interrupts are all connected to hardware devices and are necessary.

ETS_UART0_INTR_SOURCE
ETS_UART2_INTR_SOURCE
ETS_UART1_INTR_SOURCE
ETS_GPIO_INTR_SOURCE
ETS_SPI2_INTR_SOURCE
ETS_USB_INTR_SOURCE

igrr, 2024-12-19:
Please take a look at the second part of [this documentation section](https://docs.espressif.com/projects/esp-idf/en/latest/esp32s3/api-reference/system/intr_alloc.html#troubleshooting-interrupt-allocation), suggestions 2 and 3 seem relevant to your case.

travisbit, 2024-12-19:
Multiple CPUs make significant changes to my solution, which is not very practical. Can the shared solution be applied to serial ports

Additionally, there are many interruptions that are reserved here. Why are they reserved and how can they be enabled

CPU 0 interrupt status:
 Int  Level  Type   Status
  0     1    Level  Reserved
  1     1    Level  Reserved
  2     1    Level  Used: RTC_CORE
  3     1    Level  Used: SYSTIMER_TARGET2
  4     1    Level  Reserved
  5     1    Level  Reserved
  6     1    Level  CPU-internal
  7     1    Level  CPU-internal
  8     1    Level  Reserved
  9     1    Level  Used: FROM_CPU_INTR0
 10     1    Edge   Free (not general-use)
 11     3    Level  CPU-internal
 12     1    Level  Used: SYSTIMER_TARGET0
 13     1    Level  Used: TG0_WDT_LEVEL
 14     7    Level  Reserved
 15     3    Level  CPU-internal
 16     5    Level  CPU-internal
 17     1    Level  Used: UART0
 18     1    Level  Used: UART2
 19     2    Level  Used: UART1
 20     2    Level  Used: GPIO
 21     2    Level  Used: SPI2
 22     3    Edge   Reserved
 23     3    Level  Used: USB
 24     4    Level  Reserved
 25     4    Level  Reserved
 26     5    Level  Free (not general-use)
 27     3    Level  Reserved
 28     4    Edge   Free (not general-use)
 29     3    Level  CPU-internal
 30     4    Edge   Reserved
 31     5    Level  Reserved

ESP-Marius, 2024-12-25:
Without allocating interrupts to both CPUs you will face difficulty fully utilizing everything as S3 has a lot of interrupt sources.

Most of the peripheral drivers should support allocating [shared interrupts](https://docs.espressif.com/projects/esp-idf/en/latest/esp32s3/api-reference/system/intr_alloc.html#multiple-handlers-sharing-a-source), I suggest you give that approach a try if using both CPUs is not possible.

> Why are they reserved and how can they be enabled

They are reserved for misc system stuff or e.g. wifi, so enabling them wont necessary be possible. The last year we merged a79c6f7f67741fcba4ac0bc2d98749062b90c837 to give greater flexibility and free up some interrupts, so you could try upgrading to a version that includes this one and see if it helps you.

travisbit, 2024-12-25:
ok,thanks,   I'll upgrade to the new version and check it.

ESP-Marius, 2025-01-09:
Marking this one as closed then, if you still think there is anything we havent resolved then please feel free to reopen

closed 2025-01-09 by ESP-Marius