espressif/esp-idf #15063: Sleeping modes combination confusion

pavel808, 2024-12-19:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

This is more of a general inquiry.

I have an esp32-c6 which I am using on a battery-powered device, which is set up as a Matter Thread end device. Low power consumption is important. 

I know that there are different esp-idf sleeping modes, light and deep sleep as described here : 
https://docs.espressif.com/projects/esp-idf/en/v5.1/esp32c6/api-reference/system/sleep_modes.html#:~:text=ESP32%2DC6%20contains%20the%20following,their%20supply%20voltage%20is%20reduced.

On top of that I have the option to configure the device as a Thread Sleepy End Device. 

I'm confused as to what sleeping modes to use overall for the lowest power consumption. For example, can I combine the deep sleep option with the SED option from Thread. Or should I just stick to the Thread Sleepy device option and nothing more? Thanks in advance.

chshu, 2024-12-20:
Light sleep is recommended for Thread use cases because the network connection is maintained during the sleep phase. Sleepy devices only need to wake periodically to send keep-alive messages.

In contrast, waking from deep sleep is similar to a software reset, the network connection is lost, requiring the device to reattach to the network. This process consumes more resources and involves additional message transmissions.

You can find the Matter-over-Thread ICD (sleepy device) example here:
https://github.com/espressif/esp-matter/tree/main/examples/icd_app

pavel808, 2024-12-20:
> Light sleep is recommended for Thread use cases because the network connection is maintained during the sleep phase. Sleepy devices only need to wake periodically to send keep-alive messages.
>
> In [...]

Thank you

closed 2025-01-04 by pavel808