espressif/esp-idf #15064: toolchain difference between Mac & Linux builds of 5.3.2 when enabling C2X/C23.

davidzuhn, 2024-12-20:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.3.2

### Operating System used.

Linux

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

None

### What is the expected behavior?

I have a project I've been building on a Mac host for some time now, and that's working just fine.  

I recently added an aarch64 Linux box for doing CI work, and when I using the same ESP-IDF release (v5.3.2, or the tip of the 5.3 branch), I expect to get a clean compilation when I have a clean compilation of identical source files on the Mac:

### What is the actual behavior?

Instead I get an error compiling various files within esp-idf itself:

get a compilation error within ESP-IDF:
```
[3/1046] Building C object esp-idf/esp_hw_support/CMakeFiles/__idf_esp_hw_support.dir/spi_share_hw_ctrl.c.obj
FAILED: esp-idf/esp_hw_support/CMakeFiles/__idf_esp_hw_support.dir/spi_share_hw_ctrl.c.obj 
/home/ubuntu/.espressif/tools/xtensa-esp-elf/esp-13.2.0_20240530/xtensa-esp-elf/bin/xtensa-esp32s3-elf-gcc -DESP_PLATFORM -DIDF_VER=\"v5.3.2\" -DSOC_MMU_PAGE_SIZE=CONFIG_MMU_PAGE_SIZE -DSOC_XTAL_FREQ_MHZ=CONFIG_XTAL_FREQ -D_GLIBCXX_HAVE_POSIX_SEMAPHORE -D_GLIBCXX_USE_POSIX_SEMAPHORE -D_GNU_SOURCE -D_POSIX_READER_WRITER_LOCKS -I/home/ubuntu/esp32/programs/blue-knobby-throttle-firmware/build/config -I/home/ubuntu/esp32/esp-idf-v5.3.2/components/esp_hw_support/include -I/home/ubuntu/esp32/esp-idf-v5.3.2/components/esp_hw_support/include/soc -I/home/ubuntu/esp32/esp-idf-v5.3.2/components/esp_hw_support/include/soc/esp32s3 -I/home/ubuntu/esp32/esp-idf-v5.3.2/components/esp_hw_support/dma/include -I/home/ubuntu/esp32/esp-idf-v5.3.2/components/esp_hw_support/ldo/include -I/home/ubuntu/esp32/esp-idf-v5.3.2/components/esp_hw_support/port/include -I/home/ubuntu/esp32/esp-idf-v5.3.2/components/esp_hw_support/include/esp_private -I/home/ubuntu/esp32/esp-idf-v5.3.2/components/esp_hw_support/port/esp32s3/. -I/home/ubuntu/esp32/esp-idf-v5.3.2/components/esp_hw_support/port/esp32s3/include -I/home/ubuntu/esp32/esp-idf-v5.3.2/components/newlib/platform_include -I/home/ubuntu/esp32/esp-idf-v5.3.2/components/freertos/config/include -I/home/ubuntu/esp32/esp-idf-v5.3.2/components/freertos/config/include/freertos -I/home/ubuntu/esp32/esp-idf-v5.3.2/components/freertos/config/xtensa/include -I/home/ubuntu/esp32/esp-idf-v5.3.2/components/freertos/FreeRTOS-Kernel/include -I/home/ubuntu/esp32/esp-idf-v5.3.2/components/freertos/FreeRTOS-Kernel/portable/xtensa/include -I/home/ubuntu/esp32/esp-idf-v5.3.2/components/freertos/FreeRTOS-Kernel/portable/xtensa/include/freertos -I/home/ubuntu/esp32/esp-idf-v5.3.2/components/freertos/esp_additions/include -I/home/ubuntu/esp32/esp-idf-v5.3.2/components/heap/include -I/home/ubuntu/esp32/esp-idf-v5.3.2/components/log/include -I/home/ubuntu/esp32/esp-idf-v5.3.2/components/soc/include -I/home/ubuntu/esp32/esp-idf-v5.3.2/components/soc/esp32s3 -I/home/ubuntu/esp32/esp-idf-v5.3.2/components/soc/esp32s3/include -I/home/ubuntu/esp32/esp-idf-v5.3.2/components/hal/platform_port/include -I/home/ubuntu/esp32/esp-idf-v5.3.2/components/hal/esp32s3/include -I/home/ubuntu/esp32/esp-idf-v5.3.2/components/hal/include -I/home/ubuntu/esp32/esp-idf-v5.3.2/components/esp_rom/include -I/home/ubuntu/esp32/esp-idf-v5.3.2/components/esp_rom/include/esp32s3 -I/home/ubuntu/esp32/esp-idf-v5.3.2/components/esp_rom/esp32s3 -I/home/ubuntu/esp32/esp-idf-v5.3.2/components/esp_common/include -I/home/ubuntu/esp32/esp-idf-v5.3.2/components/esp_system/include -I/home/ubuntu/esp32/esp-idf-v5.3.2/components/esp_system/port/soc -I/home/ubuntu/esp32/esp-idf-v5.3.2/components/esp_system/port/include/private -I/home/ubuntu/esp32/esp-idf-v5.3.2/components/xtensa/esp32s3/include -I/home/ubuntu/esp32/esp-idf-v5.3.2/components/xtensa/include -I/home/ubuntu/esp32/esp-idf-v5.3.2/components/xtensa/deprecated_include -I/home/ubuntu/esp32/esp-idf-v5.3.2/components/esp_timer/include -I/home/ubuntu/esp32/esp-idf-v5.3.2/components/lwip/include -I/home/ubuntu/esp32/esp-idf-v5.3.2/components/lwip/include/apps -I/home/ubuntu/esp32/esp-idf-v5.3.2/components/lwip/include/apps/sntp -I/home/ubuntu/esp32/esp-idf-v5.3.2/components/lwip/lwip/src/include -I/home/ubuntu/esp32/esp-idf-v5.3.2/components/lwip/port/include -I/home/ubuntu/esp32/esp-idf-v5.3.2/components/lwip/port/freertos/include -I/home/ubuntu/esp32/esp-idf-v5.3.2/components/lwip/port/esp32xx/include -I/home/ubuntu/esp32/esp-idf-v5.3.2/components/lwip/port/esp32xx/include/arch -I/home/ubuntu/esp32/esp-idf-v5.3.2/components/lwip/port/esp32xx/include/sys -I/home/ubuntu/esp32/esp-idf-v5.3.2/components/efuse/include -I/home/ubuntu/esp32/esp-idf-v5.3.2/components/efuse/esp32s3/include -I/home/ubuntu/esp32/esp-idf-v5.3.2/components/spi_flash/include -I/home/ubuntu/esp32/esp-idf-v5.3.2/components/bootloader_support/include -I/home/ubuntu/esp32/esp-idf-v5.3.2/components/bootloader_support/bootloader_flash/include -I/home/ubuntu/esp32/esp-idf-v5.3.2/components/esp_driver_gpio/include -I/home/ubuntu/esp32/esp-idf-v5.3.2/components/esp_pm/include -I/home/ubuntu/esp32/esp-idf-v5.3.2/components/esp_mm/include -I/home/ubuntu/esp32/esp-idf-v5.3.2/components/esp_psram/include -mlongcalls  -fno-builtin-memcpy -fno-builtin-memset -fno-builtin-bzero -fno-builtin-stpcpy -fno-builtin-strncpy -ffunction-sections -fdata-sections -Wall -Werror=all -Wno-error=unused-function -Wno-error=unused-variable -Wno-error=unused-but-set-variable -Wno-error=deprecated-declarations -Wextra -Wno-unused-parameter -Wno-sign-compare -Wno-enum-conversion -gdwarf-4 -ggdb -mdisable-hardware-atomics -Og -fno-shrink-wrap -fstack-protector-all -fmacro-prefix-map=/home/ubuntu/esp32/programs/blue-knobby-throttle-firmware=. -fmacro-prefix-map=/home/ubuntu/esp32/esp-idf-v5.3.2=/IDF -fstrict-volatile-bitfields -fno-jump-tables -fno-tree-switch-conversion -std=gnu17 -Wno-old-style-declaration -std=gnu2x -MD -MT esp-idf/esp_hw_support/CMakeFiles/__idf_esp_hw_support.dir/spi_share_hw_ctrl.c.obj -MF esp-idf/esp_hw_support/CMakeFiles/__idf_esp_hw_support.dir/spi_share_hw_ctrl.c.obj.d -o esp-idf/esp_hw_support/CMakeFiles/__idf_esp_hw_support.dir/spi_share_hw_ctrl.c.obj -c /home/ubuntu/esp32/esp-idf-v5.3.2/components/esp_hw_support/spi_share_hw_ctrl.c
/home/ubuntu/esp32/esp-idf-v5.3.2/components/esp_hw_support/spi_share_hw_ctrl.c:18:63: error: implicit declaration of function 'ATOMIC_VAR_INIT'; did you mean 'ATOMIC_FLAG_INIT'? [-Werror=implicit-function-declaration]
   18 | static atomic_bool spi_periph_claimed[SOC_SPI_PERIPH_NUM] = { ATOMIC_VAR_INIT(true), ATOMIC_VAR_INIT(false),
      |                                                               ^~~~~~~~~~~~~~~
      |                                                               ATOMIC_FLAG_INIT
/home/ubuntu/esp32/esp-idf-v5.3.2/components/esp_hw_support/spi_share_hw_ctrl.c:18:63: error: initializer element is not constant
/home/ubuntu/esp32/esp-idf-v5.3.2/components/esp_hw_support/spi_share_hw_ctrl.c:18:63: note: (near initialization for 'spi_periph_claimed[0]')
/home/ubuntu/esp32/esp-idf-v5.3.2/components/esp_hw_support/spi_share_hw_ctrl.c:18:86: error: initializer element is not constant
   18 | static atomic_bool spi_periph_claimed[SOC_SPI_PERIPH_NUM] = { ATOMIC_VAR_INIT(true), ATOMIC_VAR_INIT(false),
      |                                                                                      ^~~~~~~~~~~~~~~
/home/ubuntu/esp32/esp-idf-v5.3.2/components/esp_hw_support/spi_share_hw_ctrl.c:18:86: note: (near initialization for 'spi_periph_claimed[1]')
/home/ubuntu/esp32/esp-idf-v5.3.2/components/esp_hw_support/spi_share_hw_ctrl.c:20:63: error: initializer element is not constant
   20 |                                                               ATOMIC_VAR_INIT(false),
      |                                                               ^~~~~~~~~~~~~~~
/home/ubuntu/esp32/esp-idf-v5.3.2/components/esp_hw_support/spi_share_hw_ctrl.c:20:63: note: (near initialization for 'spi_periph_claimed[2]')
```

[... cut ...]
ninja failed with exit code 1, output of the command is in the /home/ubuntu/esp32/esp-idf-v5.3.2/examples/get-started/hello_world/build/log/idf_py_stderr_output_82081 and /home/ubuntu/esp32/esp-idf-v5.3.2/examples/get-started/hello_world/build/log/idf_py_stdout_output_82081
ubuntu@turingpi-n4:~/esp32/esp-idf-v5.3.2/examples/get-started/hello_world$
```

### More Information.

If I set the C standard version to 17, I can compile without trouble.   

I was getting aggressive in C standard versions.   The concern is not that the compiler doesn't support C2X/C23, but that the two different builds of what is supposed to be the same compiler behave differently when run on Mac vs. aarch64.

Lapshin, 2024-12-20:
Hi @davidzuhn !

Unfortunately, I can't reproduce it with the steps you described. Because when you add a line `set(CMAKE_C_STANDARD 23)` to project's `CMakeLists.txt` the compile command line become to look like this - `xtensa-esp32s3-elf-gcc ... -std=gnu2x .... -std=gnu17 ...`. So, the `c17` standard will be applied because the first `-std=` option was overridden.

To test compilation with `c23` I applied this patch to IDF:

```
--- a/tools/cmake/build.cmake
+++ b/tools/cmake/build.cmake
@@ -131,7 +131,7 @@ function(__build_set_lang_version)
         # Use latest supported versions.
         # Please update docs/en/api-guides/c.rst, docs/en/api-guides/cplusplus.rst and
         # tools/test_apps/system/cxx_build_test/main/test_cxx_standard.cpp when changing this.
-        set(c_std gnu17)
+        set(c_std gnu2x)
         set(cxx_std gnu++2b)
     else()
         enable_language(C CXX)
```

After that, I can observe the issue.

I believe that for your MacOS builds std option just was not applied as you expected because the error message is valid for `c23` standard. `ATOMIC_VAR_INIT` was removed in `c23`. In `c17` it was deprecated but still works.

Lapshin, 2024-12-20:
Another question is why did you get different behavior

Maybe it is related to different `cmake` versions? Could you please tell me what cmake versions are installed on both machines?

davidzuhn, 2024-12-20:
On the Mac I've got Cmake 3.31.1 (via Homebrew) and on the aarch64 it's 3.22.1

Lapshin, 2024-12-23:
@davidzuhn , I suggest changing project's properties using [IDF's API](https://docs.espressif.com/projects/esp-idf/en/stable/esp32/api-guides/build-system.html#esp-idf-cmake-build-system-api). For your particular example, you can use lines in `main/CMakeLists.txt` to change the standard (if you really want this to apply to the whole project including IDF's files):

```cmake
idf_build_set_property(C_COMPILE_OPTIONS "--std=gnu99" APPEND)
idf_build_set_property(CXX_COMPILE_OPTIONS "--std=gnu++20" APPEND)
```

I suspect that in your case, applying only to the user's top-level files will be sufficient:

```cmake
target_compile_options(${COMPONENT_LIB} PRIVATE --std=gnu99)
```

The problem is that cmake after the v3.26 (change https://github.com/Kitware/CMake/commit/914571a04249ee561a30309197d8e1c1cacbe18b) places std option specified via `CMAKE_C_STANDARD` to the beginning of command line. So, IDF will override it.
On the other hand, we can't use set std via `CMAKE_C_STANDARD` in IDF because the minimal supported version of cmake is 3.16 which does not support the C17 standard (support added in version 3.21)

closed 2025-03-07 by Alvin1Zhang