espressif/esp-idf #15067: Add "out-of-the-box" working example of FreeRTOS and other components Mocking

Fo-Zi, 2024-12-20:
### Is your feature request related to a problem?

The "problem" I see is that while mocking seems to be integrated (At least for some components) on esp-idf, I haven't been able to find a **working example** on how to use these mocking capabilities. I have read the documentation, on both esp-idf and CMock, the problem is how to go from that, to include tests that rely on mocking to an existent project. Esp-idf uses CMake, but not in the "traditional" way I have seen in other projects, so how to do this doesn't seem trivial.

### Describe the solution you'd like.

I'd like to see a working project example, like many on the examples folder of esp-idf and were of GREAT HELP. But defining tests that use mocking. FreeRTOS mocking alone would be a huge added value, since you can test through software alone task interaction logic for complex applications.

### Describe alternatives you've considered.

Since I have been unable to integrate mocking natively, I have been considering on using Ceedling as a nested isolated project, and doing the mocking myself. But I am not so familiar with Ceedling so I'd need more time to achieve this.

ESP-Marius, 2024-12-25:
Hi, did you take a look at [this](https://docs.espressif.com/projects/esp-idf/en/v5.3.2/esp32/api-guides/unit-tests.html#mocks) part of the docs? 

At the end it also links to some examples, including [esp-event tests](https://github.com/espressif/esp-idf/blob/v5.3.2/components/esp_event/host_test/esp_event_unit_test/main/esp_event_test.cpp) which uses mocks for the freertos calls.

Fo-Zi, 2024-12-30:
Hi @ESP-Marius ! Sorry for the late reply. Yes, I've read the section of the docs regarding mocks, I didn't try this particular example tho, by using the esp-event tests project I was able to compile the tests and run them by hand, thank you! 
However, I still have some doubts, regarding how to integrate this onto a full project. If I can get this working I could make a pull request to add it as an example if it helps.

My doubts are:

- When I work with idf, I'll be ofc targeting an esp32 microcontroller based board. What I think people would expect is to be able to test their code on the host through some command, and if they want to compile their project for the target, then use some other command instead. How can I do this with IDF? The examples overwrite the "**_sdkconfig.defaults_**" to target Linux if I am not mistaken
- All the examples use like an isolated project to define the test runner executable. In a real project I believe one would have many components, and would like to test each component individually in the host. The best way to do this is to define one nested project per component? Or is there a native way of doing this through the SDK?

ESP-Marius, 2025-01-02:
There is more information about running on host [here](https://docs.espressif.com/projects/esp-idf/en/stable/esp32s2/api-guides/host-apps.html#running-esp-idf-applications-on-host)

Basically, to run on the host you would just compile for the linux target (instead of e.g. esp32) with `idf.py --preview set-target linux`. (overridding it in sdkconfig.defaults has the same result) .

The linux target already provides implementation capable of running on linux/macos for many of the ESP-IDF components. Mocking with cmock is something you can then add on-top of this, if you need to mock any specific components/interfaces.

> The best way to do this is to define one nested project per component?

Yes, this is the way we do it in IDF, every component will have its own nested `test_apps`-folder containing the different test-projects. We also have several tools to assist with all of this: 

* [IDF build apps](https://github.com/espressif/idf-build-apps) to help you automate the process when you have to build a large amount of test-apps
* [pytest-embedded](https://github.com/espressif/pytest-embedded) for running tests

For overall layout, with pre-defined github workflows etc you can also take a look at https://github.com/espressif/gh-esp-test-template, which should give a nice introduction on how to structure everything. If there is anything missing there or anything is unclear, then any feedback in that repo would be welcomed.

ESP-Marius, 2025-01-08:
Considering this one as closed. I've added a link to our unit-test docs to make the template project more visible as well, should be available the next time the code-base syncs to Github.

Feel free to re-open if you still think our docs/examples are lacking in this area.

closed 2025-01-08 by ESP-Marius