espressif/esp-idf #15069: [esp-idf][OpenThread]: When using WiFi DNS and Openthread DNS64 in the same code, an LwIP DNS Hook fail may happen

SuGlider, 2024-12-20:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

@wqx6 - 

The issue is related to commit https://github.com/espressif/esp-idf/commit/f62628d33422edcc4602d539560ce32e8e174f43#diff-c64cd8bea4c6aa81e0c356a5f4e5dd0f4f8d9507fb7efee6a11d47f3dfcb3d76R58

Lines 58 and 64 have been added to use `switching_lock` Mutex.
Whenever the application tries to resolve DNS using, for instance through WiFi, **and OpenThread has not been initialized or has been deinitialized**, the MUTEX will be NULL, therefore, `lwip_hook_dns_external_resolve()` that is called for any DNS query, will fail with  these messages:

```
E (36035) OPENTHREAD: esp_openthread_task_switching_lock_acquire(38): Failed to acquire the lock because the mutex is not ready
E (36035) OPENTHREAD: esp_openthread_task_switching_lock_release(46): Failed to release the lock because the mutex is not ready
E (36046) OT_DNS64: Cannot find NAT64 prefix
```

This is impacting ESP 32 Arduino Framework:
https://github.com/espressif/arduino-esp32/issues/10754

The error messages are caused by `esp_openthread_task_switching_lock_acquire(portMAX_DELAY);` and `esp_openthread_task_switching_lock_release();` calls  in `esp_openthread_get_nat64_prefix()`, which is called from `lwip_hook_dns_external_resolve()`

--------
**A proposed solution would be to first check if OpenThread stack is Active before performing any operation in the `lwip_hook_dns_external_resolve()` routine.**

KenFellows, 2024-12-24:
When will this issue get fixed?

zwx1995esp, 2024-12-25:
Hi @KenFellows We have a internal PR to fix this issue. This PR is now under code reviewing and will be merged to our internal branch soon.

SuGlider, 2024-12-25:
> When will this issue get fixed?

I have tested the internal PR and it works with Arduino + WiFi + DNS Client
It shall be merged soon to IDF and ESP32 Arduino shall build a new Version/Release with the fix.

Alvin1Zhang, 2025-01-09:
Thanks for reporting, the fix is available at https://github.com/espressif/esp-idf/commit/5a261da8805932c6816e00ff9d2d3d0efe5be353, and has been back ported until release/5.1, feel free to reopen if the issue still happens.

closed 2025-01-09 by Alvin1Zhang

SuGlider, 2025-01-13:
Thanks @Alvin1Zhang !!

Sobhan-Nami, 2025-03-25:
hi.
this problem still exist in esp-idf v5.4.
is there any way to fix this?