espressif/esp-idf #15071: Is WPA3_PSK supported for the ESP32-S3

Jaffa-Cakes, 2024-12-21:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

Hey everyone, I'm unsure where else I should ask this question so here we are. I have also posted this question on the esp-idf-svc repo since I am using rust, though this shouldn't be a Rust related issue: https://github.com/esp-rs/esp-idf-svc/issues/530

My router[^1] supports WPA2-PSK and WPA3-PSK and I normally have it running in WPA3-PSK only mode. When attempting to connect to my network in WPA3 Personal mode, I experience an auth issue. When allowing hybrid WPA2/WPA3 on my router and attempting to connect with WPA2 Personal it works.

Doing research online leads me to believe that the ESP32-S3 [^2] should be able to handle WPA3. I'm a bit confused as to why. ChatGPT leads me to believe it may have something to do with WPA3-SAE?

I would also like to note, that unless I explicitly set WPA2 as the preferred connection method in my software that it will not connect - meaning if I don't provide a preference or do preference WPA3 then it wont work at all.

## Code

```rust
use esp_idf_svc::{
    eventloop::EspSystemEventLoop,
    hal::modem::WifiModemPeripheral,
    nvs::EspDefaultNvsPartition,
    wifi::{AuthMethod, BlockingWifi, ClientConfiguration, Configuration, EspWifi},
};

const WIFI_SSID: &str = env!("WIFI_SSID");
const WIFI_PASSWORD: &str = env!("WIFI_PASSWORD");
const WIFI_AUTH_METHOD: &str = env!("WIFI_AUTH_METHOD");

pub struct Wifi<'a> {
    wifi: BlockingWifi<EspWifi<'a>>,
}

impl<'a> Wifi<'a> {
    pub fn init<M>(
        modem: M,
        sys_loop: EspSystemEventLoop,
        nvs: EspDefaultNvsPartition,
    ) -> anyhow::Result<Self>
    where
        M: WifiModemPeripheral + 'a,
    {
        let mut wifi =
            BlockingWifi::wrap(EspWifi::new(modem, sys_loop.clone(), Some(nvs))?, sys_loop)?;

        let auth_method = match WIFI_AUTH_METHOD {
            "WPA2_PSK" => AuthMethod::WPA2Personal,
            "WPA3_PSK" => AuthMethod::WPA3Personal,
            _ => return Err(anyhow::anyhow!("Invalid WIFI_AUTH_METHOD")),
        };

        let wifi_configuration = Configuration::Client(ClientConfiguration {
            ssid: WIFI_SSID
                .try_into()
                .map_err(|_| anyhow::anyhow!("Invalid SSID Format"))?,
            bssid: None,
            auth_method, // This is commented out if I want to simulate having no preference
            password: WIFI_PASSWORD
                .try_into()
                .map_err(|_| anyhow::anyhow!("Invalid Password Format"))?,
            channel: None,
            ..Default::default()
        });
        wifi.set_configuration(&wifi_configuration)?;

        Ok(Self { wifi })
    }

    pub fn is_up(&mut self) -> anyhow::Result<bool> {
        self.wifi
            .is_up()
            .map_err(|_| anyhow::anyhow!("Error checking WiFi status"))
    }

    pub fn connect(&mut self, retries: usize) -> anyhow::Result<()> {
        log::info!("Config:\n{:#?}", self.wifi.get_configuration()?);

        for attempt in 0..retries {
            if !self
                .wifi
                .is_connected()
                .map_err(|_| anyhow::anyhow!("Error checking WiFi status"))?
            {
                self.wifi.start()?;
                match self.wifi.connect() {
                    Ok(_) => {
                        log::info!("Connected to WiFi!");
                        self.wifi.wait_netif_up()?;
                        return Ok(());
                    }
                    Err(e) => {
                        log::error!("Failed to connect to WiFi: {:#?}", e);
                        if e.code() == esp_idf_svc::sys::ESP_ERR_TIMEOUT {
                            log::error!("Failed to connect to WiFi: {}", e);
                        } else {
                            return Err(e.into());
                        }
                    }
                }
            }
        }

        Err(anyhow::anyhow!("Failed to connect to WiFi"))
    }

    pub fn disconnect(&mut self) -> anyhow::Result<()> {
        self.wifi.disconnect()?;

        Ok(())
    }
}
```

## Security Options

[image]

## WPA3-PSK Only Router Mode Logs

https://pastebin.com/sdpCQxt5

## WPA2-PSK/WPA3-PSK Router Mode with Preferred WPA2 Personal Logs

https://pastebin.com/XnUvFHQp

[^1]: Telstra Smart Modem 3
[^2]: ESP32-S3-WROOM-1U-N8R8 Development Board

kapilkedawat, 2024-12-23:
WPA3_PSK is supported, could you please enable supplicant logs and share.

Jaffa-Cakes, 2025-01-16:
It was a me problem, I forgot to attach an antenna to the 1U model. 😅

closed 2025-01-16 by Jaffa-Cakes