espressif/esp-idf #15072: Wi-Fi PSK routine could be optimized further

ivq, 2024-12-21:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

Wi-Fi PSK is vital for WPA/WPA2 Personal networks, and its calculation is a heavy load.
Currently, we have fast-pbkdf2, but the performance of it is far from optimal.
Optimizing PSK routine brings many benefit. For one thing, it reduces time spent to join an AP. For another, it improves
responsiveness of the Wi-Fi stack as the Wi-Fi task calls the routine.

# Proposed improvement
I'm working on an optimized PSK implementation and it is showing significant improvement.

kapilkedawat, 2025-01-08:
Hi @ivq, how did you take the measurements? I ran the numbers, and they are higher for both fastpbkdf2 and fastpsk. The results were obtained by generating 100 random SSIDs and passphrases, then averaging the timings.

|           | MbedTLS | Fast_PBKDF2 | Fast_PSK |
| --------- | ------  | ----------- | ------- |
|  ESP32C2  |    602     |     258        |   180     |
|  ESP32C3  |    639     |     282        |   111      |
|  ESP32C5  |    589     |     261        |   149      |
|  ESP32C6  |    672     |     304        |   172      |
|  ESP32C61 |    799     |     349        |   171      |
|  ESP32S2  |    944     |     264        |   102      |
|  ESP32S3  |    992     |     450        |    91      |

kapilkedawat, 2025-03-05:
@ivq sorry for delayed response, this change is merged internally and will be available in github shortly.

kapilkedawat, 2025-03-07:
Changes are merged in master https://github.com/espressif/esp-idf/commit/03311c6867e77cf303752959542c38db12eed29a

closed 2025-03-07 by Alvin1Zhang